moveit/moveit #138: Capability to remove waypoints from robot_trajectory

mikeferguson, 2016-08-24:
Ported from https://github.com/ros-planning/moveit_core/issues/229:

I'm guessing it's a concious design decision, but I just discovered that there are various ways of adding waypoint to trajectories, but apparently (?) no way to remove them (looking at https://github.com/ros-planning/moveit_core/blob/indigo-devel/robot_trajectory/include/moveit/robot_trajectory/robot_trajectory.h).

What I want to do is essentially removing the first x waypoints in a trajectory. The way I'd do it right now is creating a new trajectory and iterating over the indices I want to have in my reduced trajectory and inserting them into the new one. Would it be reasonable to add a method for removing waypoints, or is the method I outlined the recommended one?

de-vri-es, 2016-08-25:
It seems like a needless omission to me. Internally an `std::deque` is used which has good performance for deleting at both ends.

I'd even say it would be nice to expose an API compatible with the Container concept (or possibly a meaningful subset). That would enable easy usage of generic algorithms. But looking at the current interface that does not seem easy without breaking backwards compatibility.

Still, +1 for adding functions to remove waypoints.

v4hn, 2016-08-26:
I know this is more than a year ago, but @skohlbr patches are still welcome :)

+1 for adding simple functions to remove the first/last `n` waypoints.
If it is compatible with the current API, also +1 to become compatible with C++ containers!

mikeferguson, 2024-12-08:
https://github.com/moveit/moveit2/commit/00f6c6fa6aad0000f279f25901660ccbf82482d1

closed 2024-12-08 by mikeferguson