moveit/moveit #140: "Returning dirty link transformation" warning when planning with joint constraints

mikeferguson, 2016-08-24:
Problem: when planning with joint constraints, warning message "Returning dirty link transformation" keeps popping up.

Problem location: I have tracked the problem down to the following function in moveit_core/constraint_samplers/src/default_constraint_samplers.cpp:

```
bool constraint_samplers::JointConstraintSampler::sample(robot_state::RobotState &state,
                                                         const robot_state::RobotState & /* reference_state */,
                                                         unsigned int /* max_attempts */)
{
  ...
  state.setJointGroupPositions(jmg_, values_);
  return true;
}
```
The problem is that after `state.setJointGroupPositions(jmg_, values_)` the state is not updated. Therefore when followed by a call to an IKConstraintSampler, the robot state is actually dirty, thus causing the warning message.

Fix: Just by adding `state.update()` after the `state.setJointGroupPositions(jmg_, values_)` in above code solves the problem.

Ported from https://github.com/ros-planning/moveit_core/issues/279

mikeferguson, 2016-08-24:
Can't confirm the above, but the reasoning seems correct, so marking as bug

scottpaulin, 2016-08-30:
Thanks for the really detailed issue @johnnieleung

You basically nailed the issue on the head here:
`The problem is that after state.setJointGroupPositions(jmg_, values_) the state is not updated. Therefore when followed by a call to an IKConstraintSampler, the robot state is actually dirty, thus causing the warning message.`

You are right that RobotState::update (or RobotState::updateLinkTransforms) fixes the issue. The tricky part is figuring out where to call this method, since it is expensive. 

This issue will hopefully be fixed as a result of pull requests #186 and #188.

v4hn, 2016-09-24:
both requests have been merged.
Thus, this problem should be solved.

closed 2016-09-24 by v4hn