moveit/moveit #141: ROS Buildfarm Dev Job Timeout

davetcoleman, 2016-08-24:
We're starting to get occasional timeouts on the ROS buildfarm for moveit due to its size - the limit is 120 minutes and you can see the history of the builds [here](http://build.ros.org:8080/job/Jdev__moveit__ubuntu_trusty_amd64/buildTimeTrend). After talking with @tfoote I think the best course of action is to disable our usage of the the ROS buildfarm for testing our source tree - i.e. removing jobs from Jdev__moveit__ubuntu_trusty_amd64. We'll still have Travis CI for all our testing. In reality I don't think many of us really look at the buildfarm because it frequently has false negatives when other packages break - I know I filter all its emails to a folder I never look at.

For implementation, I think it just means we remove the entries from [rosdistro](https://github.com/ros/rosdistro/blob/master/indigo/distribution.yaml#L6543) that say:

```
    source:
      test_commits: false
```

130s, 2016-08-24:
Related but side-issue: @tfoote could you answer http://answers.ros.org/question/236406/default-bool-value-for-test_commits-and-test_pull_requests-in-rosdistro-format/ ?

130s, 2016-08-24:
Not urgent. I'm just curious about the decision made here. Dave would you elaborate a bit more?

Because most of developers and users of MoveIt! will be using debs, and ROS buildfarm is the place where debs are born, IMO it's good and ideal to test building the code there (commit-by-commit test may not be necessary though). Also I'm fan of relying on ros.org resource as much as possible.

130s, 2016-08-24:
> Not urgent. I'm...(cut)

I meant "my request for more explanation is not urgent". I think this timeout issue is more urgent since it's occurring every few hours.

davetcoleman, 2016-08-24:
Yes, it would be ideal if we could continue using the build farm for testing our source, but it seems there isn't a good way to go about this. One alternative is if someone who knew a lot about CMake spent time optimizing our build process to be faster. I don't think I have the skill to do this.

After further discussing things today with @wjwwood I came to the viewpoint that the buildfarm has been built by OSRF to test lower-level packages in ROS like roscpp, but not much focus has been given to the higher level packages like moveit and its different approach. The problem I see with the buildfarm setup right now is that it makes no assumptions about the dependencies its building on top of, opting for a very clean environment where everything is installed in Docker for each job. Because MoveIt! has so many dependencies, it spends ~23min just setting up that environment. Its also, by requirement, required to be very generic, so we can't add various optimizations to speed up the build. I think we can go a lot further with Travis.

rhaschke, 2016-08-24:
I agree with @davetcoleman, that it suffices to have Travis checking in place for development.

tfoote, 2016-08-24:
Indeed the ros buildfarm jobs are designed to prioritize accuracy and correctness over speed. We do have caching so repeated builds on the same executor can use cached environments. For this case a slightly customized version running on travis with optimizations that use a preinstalled environment can definitely speed things up by avoiding lower level package installation. However that comes at the cost that it will mask errors like missing dependencies which are things that can only be caught by building in a clean environment. This has been a problem in the past that people would make a release and it would fail because of a missing dependency but the CI jobs were passing due to having extra dependencies injected by the old buildfarm for tooling. Now our goal is that if a devel job passes we expect the sourcedeb and binarydeb jobs to pass as well. 

If you need to optimize that more I think using travis makes sense. There's definitely a bunch of work that could be done to improve the environment caching. @davetcoleman you've seen how a 20 minute environment setup can go to a 2 minute setup when the cache is hit. But caching across a distributed cluster is not well supported in docker yet. https://github.com/docker/docker/issues/20316

dirk-thomas, 2016-08-24:
@davetcoleman You mention above that "it frequently has false negatives". As far as we are aware the number of false positives has been greatly reduced with the new build farm which was deployed over half a year ago. If you would point to specific builds we could take a closer look and see if they are actually false positives and if there is anything we can do about it to be classified correctly.

One reason why building the Docker image takes so long is because we install every dependency one-by-one. While that enables reusing previously created images it also increases the time. It is a pretty straight forward trade off. Currently we only fold multiple packages into a single install call if otherwise the number of Docker steps gets too high for Docker to be handled. But it should be no effort to modify that logic to e.g. group the dependencies in a single or a few install calls. That should reduce the build time (and make reuse less likely). We could try that for the current devel job of `moveit` and see how much it helps.

dirk-thomas, 2016-08-24:
The separation between "build-and-install" and "build-and-test" is described in the build farm docs: https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/jobs/devel_jobs.rst

One reason is to detect wrong invocations of tests when the users didn't want tests to be configured. One example of that is in this CMake code: https://github.com/ros-planning/moveit/blob/f254737b8c1ef83ad4fce9501da12c46b7eaf88d/moveit_ros/perception/mesh_filter/CMakeLists.txt#L15 You can search the [console output](http://build.ros.org:8080/job/Jdev__moveit__ubuntu_trusty_amd64/10/consoleFull) for "CMake Warning" to find the warning.

dirk-thomas, 2016-08-25:
I just tried a test build with all dependency installations folded into a single apt call (https://github.com/ros-infrastructure/ros_buildfarm/commit/53a581fc1149f165d3ecfe6003738f264a967bda). The time to install all dependencies shrank to 6 minutes. Overall the devel job finished in 1.5h (http://build.ros.org:8080/job/__test_Jdev__moveit__ubuntu_trusty_amd64/2/).

davetcoleman, 2016-08-25:
> you've seen how a 20 minute environment setup can go to a 2 minute setup when the cache is hit

Could we get a longer build time if no cache is available?

> As far as we are aware the number of false positives has been greatly reduced with the new build farm which was deployed over half a year ago.

This could easily be the case, I was speaking from 5 years of bombarding emails and I have since learned to ignore those emails. I don't know if its still an issue anymore, thank you for working on it and sorry if I'm making outdated negative claims.

> The time to install all dependencies shrank to 6 minutes. Overall the devel job finished in 1.5h 

That sounds really promising! How can we make that a permanent change?

tfoote, 2016-08-25:
On Thu, Aug 25, 2016 at 12:35 AM, Dave Coleman <notifications@github.com>
wrote:

> you've seen how a 20 minute environment setup can go to a 2 minute setup
> when the cache is hit
>
> Could we get a longer build time if no cache is available?
>
There's no way to know if the cache is available until you ask for the
content.

> As far as we are aware the number of false positives has been greatly
> reduced with the new build farm which was deployed over half a year ago.
>
> This could easily be the case, I was speaking from 5 [...]
Please do look at them. We've entirely refactored it and are have added a
lot of infrastructure to detect infrastructure failures, suppress the
emails, and retrigger automatically. If you do get false failures please
ticket them for us to track.

> The time to install all dependencies shrank to 6 minutes. Overall the
> devel job finished in 1.5h
>
> That sounds really promising! How can we make that a permanent change?
>

It was just merged:
https://github.com/ros-infrastructure/ros_buildfarm/pull/334

>

dirk-thomas, 2016-08-25:
The referenced PR does not yet fold all installations into a single step. It only provides the infrastructure for folding. It currently only reduces the number of steps to 80 (due to limits within Docker).

To consider folding everything into one step (as it was done for the test build) I have to think a bit more about the pros and cons. Maybe there is some heuristic to at least leverage some of the reuse will still reducing the install time. I am sure we can get some improvement to this merged soon.

tfoote, 2016-08-25:
Sorry I didn't realize that you'd not just collapsed the installs within the fold, but also reduced it to one step. 

Reducing it to one step will likely reduce the time for any individual job to build it's cache from scratch, but it will prevent sharing of caches between jobs. Which means that it will cause more cache building and require more storage for the overall system.

davetcoleman, 2016-10-27:
buildfarm timeouts seem to no longer be an issue, perhaps because of optimizations upstream. closing this issue.

closed 2017-01-13 by v4hn