moveit/moveit #159: Collision checking ignores some collada meshes

mikeferguson, 2016-08-25:
Ported from https://github.com/ros-planning/moveit_core/issues/202

I have worked around this for the time being by converting to .stl collision meshes, but when I use collision checking with the UR5 model from http://github.com/ros-industrial/universal_robot, it would see no problem with poking the arm through a table, as long as the elbow was sticking out far enough past the far side:

```
\ |
 \|
  |
  |\
  | O
  |/
  |
 /|
/ |
```

I think there may be a problem with scales or orientations or something, as it would sometimes detect collisions where there did not appear to be any.

de-vri-es, 2016-08-25:
I think the issue may be in Assimp, which enforces Y_UP convention for collada models. Basically, if a collada model uses anything other than Y_UP it is rotated until Y does point up [1].

rviz works around this by clearing the root transformation of the imported model [2]. I'm not sure if that won't break other situations where there *should* be a root transformation though. I'm afraid that what rviz does may actually be the best solution until Assimp fixed the underlying issue.

A manual workaround until we can fix this better is to change your `.dae` file to use:
```
<up_axis>Y_UP</up_axis>
```

@hersh: Could you confirm if this workaround "solves" your problem? If not we know that this is not the issue you are encountering.

[1] https://github.com/assimp/assimp/blob/23baecaff30b7a280648834368b1c0c792e4092e/code/ColladaLoader.cpp#L192
[2] https://github.com/ros-visualization/rviz/blob/0d423d6249a39f7f3ddd591007952c6dc940b6ac/src/rviz/mesh_loader.cpp#L232

de-vri-es, 2016-08-25:
Note: If the above is the issue here, then it has to be addressed in `geometric_shapes` which does the actual loading of the mesh.