moveit/moveit #161: Improve IK performance for unreachable poses

mikeferguson, 2016-08-25:
Ported from https://github.com/ros-planning/moveit_core/issues/40:

The IK seems quite fast but slows down a lot when requested to calculate an unreachable pose.

Idea: a certain set of poses could be precalculated as impossible. For example for a chain with only revolute joints any end position farther than the sum of the link lengths is impossible. Would it make sense to store such info (e.g. a sphere radius per joint group) so that some requests can be trivially rejected? Such calculations could be done per group by the SetupAssistant and stored in the srdf.

v4hn, 2016-08-25:
The proposed feature should be relatively simple to add to any (custom) `KinematicsBase` plugin.
The information does not belong in the srdf at least.

On the other hand I never had trouble with this...

mikeferguson, 2024-12-08:
Original issue was 2013, after 10 years, this probably isn't going to happen

closed 2024-12-08 by mikeferguson