moveit/moveit #166: PlanningScene::distanceToCollision returning large (possibly uninitialized) distances

mikeferguson, 2016-08-25:
ported from https://github.com/ros-planning/moveit_core/issues/238:

I am getting some very large distances when using distanceToCollision. I have compared these to distances computed using checkCollision with distance set in the CollisionRequest. I am using the Universal Robot package on Indigo.

Output:

```
valid: 1 distance_checkCollision: 0.0148276 distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0145733 distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0147373 distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0140031 distance_distanceToCollision: 1.79769e+308
valid: 0 distance_checkCollision: 0         distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0148941 distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0083631 distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0146611 distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0142101 distance_distanceToCollision: 1.79769e+308
valid: 1 distance_checkCollision: 0.0146178 distance_distanceToCollision: 1.79769e+308
```

Code:

```
robot_model_loader::RobotModelLoader robot_model_loader("robot_description"); 
robot_model::RobotModelPtr kinematic_model = robot_model_loader.getModel();
planning_scene::PlanningScene scene(kinematic_model);

robot_state::RobotState state = scene.getCurrentState();
const int attempts = 10;
for (int i = 0; i < attempts; ++i)
{
    state.setToRandomPositions();
    state.update(true);
    collision_detection::CollisionRequest request;
    request.distance = true;
    
    collision_detection::CollisionResult result;
    result.clear();
    scene.checkCollision(request, result, state);
    const bool valid = result.collision == false;
    double distance_checkCollision = result.distance;
    double distance_distanceToCollision = scene.distanceToCollision(state);
    ROS_ERROR_STREAM("valid: " << valid << " distance_checkCollision: " << distance_checkCollision << " distance_distanceToCollision: " << distance_distanceToCollision);
}
```

v4hn, 2016-08-26:
@scottpaulin I'm not sure this interests you anymore, but here's an explanation:

`PlanningScene::distanceToCollision(RobotState&)` only checks the robot state against *the scene*.
It does **not** perform self-collision checking.

`PlanningScene::checkCollision` on the other hand exposes a general collision checking interface,
so if you ask for collisions there, this *includes* self collisions. If you specify to compute distances, the distances for self-collisions will be included.

Now, the problem with your example is that you created an empty scene, and ask for the distance of your robot state to obstacles *in the empty scene*.
The resulting distance is max double.
If you add an object to your scene or use a PlanningSceneMonitor to fetch a non-empty scene from a running `move_group`, `distanceToCollision` returns meaningful values.

I just filed #174 to improve the documentation on this.

Closing as this is actually expected (although badly documented) behavior.

closed 2016-08-26 by v4hn

mikeferguson, 2016-08-26:
@v4hn -- thanks for looking into this

scottpaulin, 2016-08-30:
Ah that makes sense. Thanks for the explanation and doc updates @v4hn ! I'm sure the improved docs will save other developers time and possibly bugs in the future.