moveit/moveit #169: OMPL/catkin include order

mikeferguson, 2016-08-25:
Ported from https://github.com/ros-planning/moveit_planners/issues/66

The include order for OMPL and catkin in ompl/CMakeLists.txt seems wrong to me. However, there are already two commits switching this around: https://github.com/ros-planning/moveit_planners329fb6300c6cb8540d551f1e002fa9c57c9590b9 and https://github.com/ros-planning/moveit_plannersb888117ebb56e71e10b3d4da110d8ac9d95190a0, so making another pull request without discussion does not make sense.

The question is basically if catkin_INCLUDE_DIRS or OMPL_INCLUDE_DIRS should be preferred. I believe OMPL_INCLUDE_DIRS should be preferred. Otherwise, when working with a source build OMPL on a standard ROS install, which includes the released OMPL, this will shadow the source build OMPL.

In contrast https://github.com/ros-planning/moveit_planners329fb6300c6cb8540d551f1e002fa9c57c9590b9 states that this prevents ROS package shadowing. Shouldn't that still work as either OMPL_INCLUDE_DIRS is not set or should be set to the lowest overlay?

In either case, the order of link_directories should probably match the include order.

v4hn, 2016-08-25:
@dornhege

In theory I agree with you and `OMPL_INCLUDE_DIRS` should go first.
But this breaks *arbitrary package overlays* in OSRF's ROS indigo, because osrf installs OMPL in `/opt/ros/indigo/`.
Thus, no matter which workspace you sourced, the headers from `/opt/ros/indigo/` will be preferred.

Does this explanation resolve this ticket?

davetcoleman, 2016-08-26:
I've always been frustrated that a binary install of OMPL over-rides my source install version, and this has tripped me many times. So I would prefer it to be changed back. But I'd like to understand @v4hn's explanation better - what do you mean by 

> this breaks *arbitrary package overlays*

? Are you referring to the Ubuntu-universe ``libompl-dev``?

v4hn, 2016-08-26:
This means that if you use the standard ROS version of OMPL installed to `/opt/ros/indigo`,
moving `OMPL_INCLUDE_DIRS` first will entail that the sources will *always* favor the headers in `/opt/ros/indigo/include/`.
Thus, if you overlay the `tf` package in your current workspace and changes something there, `moveit_planners_ompl` will still include the "old" headers in `/opt/ros/indigo/include/tf/...` instead.
This is probably what @hersh ran into when he changed it back.

davetcoleman, 2016-08-26:
Why did you use ``tf`` as your example - this is confusing to me...

I would like the source-built version of OMPL to override ``/opt/ros/indigo`` debian version - in the current configuration:

```
include_directories(ompl_interface/include
                    ${catkin_INCLUDE_DIRS}
                    ${OMPL_INCLUDE_DIRS})
```

This is not what happens - correct?

v4hn, 2016-08-26:
On Fri, Aug 26, 2016 at 12:47:16PM -0700, Dave Coleman wrote:
> Why did you use ``tf`` as your example - this is confusing to me...

Because `tf` is one of the dependencies of the `moveit_planners_ompl` packages and we are talking about the `moveit_planners_ompl` (and only this) module, that would falsely prefer the headers in `/opt/ros/indigo/include/tf` to those in your overlay.

> I would like the source-built version of OMPL to override ``/opt/ros/indigo`` debian version
> [..]
> This is not what happens - correct?

This is not about debian's version. Ubuntu xenial's version 1.0 is installed to /usr where it belongs.
OSRF overload that package and [install 1.2](https://github.com/ros/rosdistro/blob/master/kinetic/distribution.yaml#L2554) to `/opt/ros/kinetic`.
As long as OSRF installs OMPL to `/opt/ros` and OMPL does not (as generally recommended for system-wide installed libraries) installs its headers to `include/ompl-$version/ompl` but to `include/ompl`.
This is really a clash between two sides that both break FOSS standards they should maintain..

So in a nutshell:

*EITHER* we can't overlay a custom installation of ompl and use it in `moveit_planners_ompl` (this is what you want)

*OR* the `moveit_planners_ompl` package will not respect overlays for its dependencies and its build will silently use the `/opt/ros/` versions of

- `moveit_core`

- `moveit_ros_planning`

- `roscpp`

- `rosconsole`

- `pluginlib`

- `tf`

- `dynamic_reconfigure`

- and `eigen_conversion`

instead of the versions you have in your local workspace.

I'm pretty sure we *don't* want `moveit_planners_ompl` to include headers from /opt/ros/kinetic/include/moveit/ when we have `moveit_core` and `moveit_ros_planning` also available in the same workspace as `moveit_planners_ompl`.

This is impossible to fix at our end.

Your best chance to get this fixed is to talk the OMPL maintainers into installing their headers by default to `include/ompl-$VERSION/ompl` (and export `OMPL_INCLUDE_DIRS=include/ompl-$VERSION/`) or to talk the OSRF responsible for OMPL into changing the package they distribute in ROS to install to `/opt/ros/kinetic/include/ompl-$VERSION`. Both ways will require a change in OSRFs package of ompl.

davetcoleman, 2016-08-26:
Thanks for that explanation. I do not have the motivation to push for this right now (too many MoveIt! emails already) so instead I just documented it and maybe someone in the future will pursue it

https://github.com/ros-planning/moveit_tutorials/pull/8

I will assume no one else does either. Please re-open if you want to work on this low-priority annoyance

mamoll, 2017-07-24:
@lepalom, something to keep in mind for future OMPL Debian releases?

v4hn, 2017-07-24:
@mamoll Actually, I would propose to change the include folder in the upstream project
and not just for the debian release. That's the way the PCL, among others, handles this.

mamoll, 2017-07-24:
@v4hn, where is this directory layout / naming convention documented? Does this apply to all Linux/Unix based systems?

v4hn, 2017-07-25:
> where is this directory layout / naming convention documented?

Good question indeed. Actually, I'm not sure it is documented explicitly,
or at least I don't have a reference at hand, unless you count [template `.pc` files as this one](https://tecnocode.co.uk/2014/12/09/a-checklist-for-writing-pkg-config-files/#).

> Does this apply to all Linux/Unix based systems?

The reasoning behind the nested layout is the exact problem we have above and to the best of my knowledge it exists throughout the Linux/Unix world: libraries don't want to pollute the ordering of include directories in case their install location coincides with other packages. In our case: make the dependent project include `/opt/ros/*/include` in the position of  the OMPL include path, even though other include paths should come in between the two.

The important behavior many of the standard libs in the Linux world exhibit upstream, but ompl doesn't, is that they do not export `${prefix}/include` as include path, but some subfolder instead. Note that `pkg-config` for example [even *drops* includes to default include locations](https://cgit.freedesktop.org/pkg-config/tree/pkg.c#n827) by default.
Everything would be fine, if OMPL would add `<install-dir>/include/ompl` as include directory and require users to include the headers as e.g. `#include <base/SpaceInformation.h>`. Personally I don't like that style, but it's common enough in standard packages.
The alternative is the additional folder.

v4hn, 2017-10-11:
As mentioned above, this is not an issue that can be fixed within MoveIt, but it requires either a change in OMPL (my preferred solution) or in OSRF's build description.

If you understand the issue, this is something that can be done in a day and the outcome should be a concrete patch that adjusts either end and a corresponding issue/pull-request.

davetcoleman, 2017-10-18:
fyi we've been discussing this recently with @wjwwood and the Kavraki Lab but have not finalized a solution

v4hn, 2017-10-18:
It would still be great to see a pull-request for this emerging from today :-)

wjwwood, 2017-10-18:
So the proposed solution, of putting the ompl headers in a subfolder, looks straightforward because there is already a `OMPL_VERSIONED_INSTALL` option in ompl, added in this commit:

https://github.com/ompl/ompl/commit/60c352848edc229c1eaa638c4817a7cfd1b6ca00

In principle all we need to do is turn it on (it is off by default as of https://github.com/ompl/ompl/commit/17781b6cebf4d307c0e15ff6ffffc7f4a24541fe) when building on the build farm.

Now, that might break other things, but I'll test it out locally by enabling this option and installing ompl to `/opt/ros/kinetic` manually. Since @isucan turned it off at some point, I expect there might be issues with packages consuming the versioned include folder.

wjwwood, 2017-10-18:
Seems to work, though the bug reproduction is a bit tricky so I might be getting a false positive inadvertently.

I'll make a pr to ompl's release repository and see if I can run a prerelease.

mamoll, 2017-10-18:
The versioned install of OMPL is supposed to make side-by-side installations easier, but it has been only lightly tested. Upon install, it'll create some symlinks (to make the latest installed version the default one) that might still cause issues. Rather than hard-coding version numbers in install locations (which seems like a hack to me), the proper solution seems to me to fix the include order so that local catkin ws is preferred over /opt/ros, which in turn is preferred over /usr. I just don't know where the problem comes from.

wjwwood, 2017-10-18:
> Rather than hard-coding version numbers in install locations (which seems like a hack to me)

That's a pretty common pattern, and what @v4hn suggested.

> Rather than hard-coding version numbers in install locations (which seems like a hack to me)

@mamoll I don't believe that is possible.

`catkin` works around this problem because it has more information. It has the list of include headers and the location of the packages which provide them (i.e. whether it the package is in `/opt/ros/kinetic` or `~/path/to/ws/install`) and the topological order of the packages. See:

https://github.com/ros/catkin/blob/5cd8c81575561ef23b291a6cee9049f2568a57ef/cmake/list_insert_in_workspace_order.cmake#L23

and

https://github.com/ros/catkin/blob/5a7f61d594a56ca4b3ae1705e64140e9f29e0bd7/cmake/order_paths.py#L16-L18

To see where catkin is doing this. That feature was added in https://github.com/ros/catkin/commit/731543097f7400ce2ff1027c89bc16dadd25647f and was reported in https://github.com/ros/catkin/issues/200.

wjwwood, 2017-10-18:
I'll wait to execute what I described above until we have some consensus on what to do.

mamoll, 2017-10-18:
Ok, we can use the OMPL_VERSIONED_INSTALL option for ROS/catkin, but in that case we should probably disable the symlinks, right?:
https://github.com/ompl/ompl/blob/c2ffc537dcc79bce389743c46abb155cce914432/CMakeLists.txt#L178-L184

wjwwood, 2017-10-18:
That's possible. I didn't investigate in detail how that feature is implemented. If the symlink is something like `<prefix>/include/ompl-x.y` -> `<prefix>/include/ompl` then we don't want that.

v4hn, 2017-10-18:
> Rather than hard-coding version numbers in install locations (which seems like a hack to me),

Out of 456 header folders of standard linux packages in my `/usr/include` folder (more vanilla than standard ubuntu dev packages), roughly 80 use this pattern. I believe ubuntu packaged more of these as -dev packages like that. If it's a "hack", then it's a common one.

> the proper solution seems to me to fix the include order so that local catkin ws is preferred over /opt/ros, which in turn is preferred over /usr.

I agree. If ROS would use a ubuntu-system package of OMPL in `/usr` this would be the answer, yes.
But actually we don't discuss *local ws* -> */opt/ros/* -> */usr/include*.
The question is: which should you prefer, include paths for ompl or include paths for catkin packages?
If I map your answer from above to this question, it becomes: It depends on in which of the locations you install the version of ompl you want to use... And we don't know that in our `CMakeLists.txt` without ugly, fragile logic...

As stated before, my proposal is to install OMPL headers into
`/opt/ros/kinetic/include/ompl-1.2/ompl/config.h` to get rid of cross-talk in header includes
and add the include path as `-I/opt/ros/kinetic/include/ompl-1.2`.
If this is what `OMPL_VERSIONED_INSTALL` does, then I agree.
I didn't verify and based on the previous explanations I'm not sure.

wjwwood, 2017-10-19:
> > the proper solution seems to me to fix the include order so that local catkin ws is preferred over /opt/ros, which in turn is preferred over /usr.

> I agree. 

That is controlled by the cmake code, basically you'd need to make sure that the ompl include directory from your local workspace ended up ahead of everything, but not only within one package but also all downstream packages which use the include directories of a package that happens to export the ompl include directories. As you said that would be fragile...

> If this is what `OMPL_VERSIONED_INSTALL` does, then I agree.

`OMPL_VERSIONED_INSTALL` does this, but also other things apparently. I'm going to look at using it while also disabling some of the features we don't want as patches in the bloom release process.

mamoll, 2017-10-19:
I can disable the sym links created by the versioned install in the OMPL repo. Maybe at some point someone can create the right kind of /etc/alternatives-like code, in needed/desired.

wjwwood, 2017-10-19:
@mamoll it's up to you if you want to disable the symlinks upstream, I was just going to do that as a patch in the bloom release. If you do make a commit that removes the symlinks then we'll need to include it in the forthcoming 1.2.2 release as described here: https://github.com/ros-planning/moveit/issues/274#issuecomment-337960158

mamoll, 2017-10-20:
This is what currently gets installed with `OMPL_VERSIONED_INSTALL=ON`:
```
${prefix}/bin/plannerarena-1.3
${prefix}/bin/ompl_benchmark_statistics-1.3.py
${prefix}/include/ompl-1.3/config.h
...
${prefix}/lib/pkgconfig/ompl-1.3.pc
${prefix}/lib/libompl.dylib
${prefix}/lib/libompl.1.3.1.dylib
${prefix}/lib/libompl.14.dylib
${prefix}/share/man/man1/plannerarena.1
${prefix}/share/man/man1/ompl_benchmark_statistics.1
${prefix}/share/ompl-1.3/ompl-config-version.cmake
${prefix}/share/ompl-1.3/ompl-config.cmake
...
```
I am guessing the headers should actually be installed in `${prefix}/include/ompl-1.3/ompl`. What about executables, man pages or .pc files? Is appending the version name ok?

wjwwood, 2017-10-20:
> I am guessing the headers should actually be installed in ${prefix}/include/ompl-1.3/ompl.

Yes, I hadn't noticed that, but yes we would want something more like:

```
${prefix}/include/ompl-1.3/ompl/config.h
...
```

With the cmake and pkg-config files giving the equivalent of `-I${prefix}/include/ompl-1.3`, so user code still just does `#include <ompl/config.h>`.

For the binaries and man pages, I'm not sure it matters but we don't need them to be namespaced because we're not going for side-by-side installation.

It sounds like we need to do some work on the `OMPL_VERSIONED_INSTALL` feature before we can use it though. I'm not sure if the version being in the name of the pkg-config and/or cmake folder/files will matter.

mamoll, 2017-10-20:
> I'm not sure if the version being in the name of the pkg-config and/or cmake folder/files will matter.

It will determine how you want other packages to find ompl. Do you want this:
```
find_package(ompl 1.3)
pkg-config --cflags --libs --exact-version=1.3 ompl
```
or this:
```
find_package(ompl-1.3)
pkg-config --cflags --libs ompl-1.3
```

wjwwood, 2017-10-20:
I think ideally we would just do:

```cmake
find_package(ompl REQUIRED)
```

And I don't think we use pkg-config, but someone else might.

mamoll, 2017-10-24:
I have made some fixes today  in the master branch of ompl that hopefully address all of the above concerns. See https://github.com/ompl/ompl/commits/master.

wjwwood, 2017-10-24:
Thanks @mamoll I'll have a look hopefully today. I'll probably need to cherry-pick those to the 1.2.2 release, or maybe cherry-pick them in ompl and do a 1.2.3 release.

Either way, let me have a look at them and test them to see if it works for moveit.

wjwwood, 2017-11-07:
Haven't forgotten about this, I tried out the changes briefly but ran into a problem on my end. I haven't had time to pick back up on this yet. Hopefully this weekend?

davetcoleman, 2017-11-07:
Thanks for the update @wjwwood, looking forward to a solution here!

v4hn, 2017-11-17:
I just looked into your changes @mamoll and they look good to me.
Thank you for taking care of this on your end!

@wjwwood @mamoll If we need any further releases of OMPL in future versions of ROS (I guess we might want more current versions of OMPL in ROS also in the future), please make sure they are always build with `-DOMPL_VERSIONED_INSTALL=ON`. I would recommend it for the debian releases too, but that's outside my domain.

Also, I backported the respective cmake changes on top of 1.2.2. This includes not only https://github.com/ompl/ompl/commit/08e5b6c7ffba82d4bc976563fe3373f7e353027a, but also the three follow-up commits. Otherwise `find_package(ompl)` will specify the wrong include folder later on.
@wjwwood Have a look at https://github.com/v4hn/ompl/tree/1.2.2-new-cmake .
It works for my setup locally, but could you please check whether the pre-release test passes with the new cmake config logic for kinetic and take care of the rest?

wjwwood, 2017-12-15:
Thanks @v4hn, I used your `1.2.2-new-cmake` branch to release ompl into my fork of the release repository as "1.2.3":

https://github.com/wjwwood/ompl-release#ompl-kinetic---123-2

I added a patch to turn the versioned install on by default, this way people downloading it from source will get the same behavior as the deb, though I think it might be sufficient to only turn this on in the deb. However, it made it easier for me to test, here's that patch:

https://github.com/wjwwood/ompl-release/commit/d78a2bbe9e144b895b147d844a02d1df1423de72

However, after running a prerelease with this command:

```
mkdir -p /tmp/prerelease_job
cd /tmp/prerelease_job
generate_prerelease_script.py \
  https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml \
  kinetic default ubuntu xenial amd64 \
  --custom-repo \
    ompl__custom-1:git:https://github.com/wjwwood/ompl-release.git:debian/kinetic/xenial/ompl \
  --level 1 \
  --output-dir ./
./prerelease.sh
```

I get an error like this:

```
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "FindOMPL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OMPL", but
  CMake did not find one.

  Could not find a package configuration file provided by "OMPL" with any of
  the following names:

    OMPLConfig.cmake
    ompl-config.cmake

  Add the installation prefix of "OMPL" to CMAKE_PREFIX_PATH or set
  "OMPL_DIR" to a directory containing one of the above files.  If "OMPL"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/tmp/catkin_workspace_overlay/build_isolated/moveit_planners_ompl/CMakeFiles/CMakeOutput.log".
See also "/tmp/catkin_workspace_overlay/build_isolated/moveit_planners_ompl/CMakeFiles/CMakeError.log".
<== Failed to process package 'moveit_planners_ompl':
  Command '['cmake', '/tmp/catkin_workspace_overlay/src/moveit_planners_ompl', '-DCATKIN_DEVEL_PREFIX=/tmp/catkin_workspace_overlay/devel_isolated/moveit_planners_ompl', '-DCMAKE_INSTALL_PREFIX=/tmp/catki
n_workspace_overlay/install_isolated', '-DCATKIN_ENABLE_TESTING=1', '-DCATKIN_SKIP_TESTING=0', '-DCATKIN_TEST_RESULTS_DIR=/tmp/catkin_workspace_overlay/test_results', '-G', 'Unix Makefiles']' returned non
-zero exit status 1

Reproduce this error by running:
==> cd /tmp/catkin_workspace_overlay/build_isolated/moveit_planners_ompl && cmake /tmp/catkin_workspace_overlay/src/moveit_planners_ompl -DCATKIN_DEVEL_PREFIX=/tmp/catkin_workspace_overlay/devel_isolated/
moveit_planners_ompl -DCMAKE_INSTALL_PREFIX=/tmp/catkin_workspace_overlay/install_isolated -DCATKIN_ENABLE_TESTING=1 -DCATKIN_SKIP_TESTING=0 -DCATKIN_TEST_RESULTS_DIR=/tmp/catkin_workspace_overlay/test_re
sults -G 'Unix Makefiles'

Command failed, exiting.
# END SUBSECTION
```

While building `moveit_planners_ompl`. So it looks like the cmake module/config file is not being found. I'll have to look into this in more detail to figure out why.

----

Separately, which versions of ROS would you want this released into? I'm guess just Kinetic, since we minted 1.2.2 and that's not a direct ancestor of the version released into Indigo.

wjwwood, 2017-12-16:
So the issue is this line (from `kinetic-devel`):

https://github.com/ros-planning/moveit/blob/4bce541153b26c184cb5d2a335fb894d744b1732/moveit_planners/ompl/CMakeLists.txt#L23

Which uses `OMPL` as uppercase, but the 1.2.2 I'm basing this new release off of only installs `<prefix>/lib/cmake/ompl/omplConfig.cmake`, which does not match `OMPL`, either due to the `ompl` folder it's in or the name of the file.

If I change the `find_package()` to `ompl`, then it works because the `omplConfig.cmake` still sets `OMPL_INCLUDE_DIRS` like variables despite being lowercase in the file name.

Above (https://github.com/ros-planning/moveit/issues/169#issuecomment-338257227) I casually mentioned that doing `find_package(ompl REQUIRED)` would be ideal, but that was just because I default to lower case if I can. Since it was already `OMPL`, I guess that's what is preferred.

----

So we need to either:

- change `moveit_planners_ompl` to use `find_package(ompl ...)`
  - and leave uses of vars like `OMPL_INCLUDE_DIRS`
- change the ompl we ship to install to a CMake file that will work with `find_package(OMPL ...)`
  - either upsteam before making the 1.2.3 release or
  - as a patch in the release repository
- something else?

@mamoll and others please let me know what you'd like to do and I'll try to do that.

davetcoleman, 2017-12-17:
I do not have strong opinions or insights into the various repercussions, but my thought is your first option "change moveit_planners_ompl" would cause the least breakages.

mamoll, 2017-12-21:
I strongly favor the option to change `moveit_planners_ompl` to use `find_package(ompl ...)`.

v4hn, 2018-07-19:
I just noticed this issue got stuck...

@wjwwood The problem you hit last time was fixed in the meanwhile in #907 .
Please try again.

I would still very much like to have a released version of OMPL in kinetic that uses
`-DOMPL_VERSIONED_INSTALL=ON` with my patch above.

v4hn, 2018-11-02:
@wjwwood I just checked and even the melodic package in `ros-melodic-ompl`
is still built without `-DOMPL_VERSIONED_INSTALL=ON`.
Could we please change that?

davetcoleman, 2019-06-24:
ping is this still an issue? can we close this as no one has complained in quite a while.

mamoll, 2019-06-25:
How do I specify in the ompl-release repo that `cmake` should be run with the extra flag `-DOMPL_VERSIONED_INSTALL=ON`? I was hoping that I could add this to the right package.xml files, but I don't see other packages doing that. Is it a good idea to change this for melodic or should I only do it for dashing at this point?

v4hn, 2019-06-25:
Last time I checked this was still relevant.
I have been waiting for @wjwwood to rebuild the melodic package for half a year now.

@mamoll: I strongly request this to be changed for melodic. Also don't forget noetic.

As far as I know, the parameter has to be specified with the build instructions for the buildfarm.
I don't think it's the package author's responsibility to ensure the flag (unless you agree to change the default).

mamoll, 2019-06-25:
@v4hn, who is responsible for the build instructions for the buildfarm? Where is this process documented? I imagine that other libraries used by ROS also have optional features that need to be disabled/enabled. How is this done?

gavanderhoorn, 2019-06-26:
This is probably possible using a patch that you add directly to the release repository. There are more packages doing this.

I don't have the time right now to find an example, but this should point you in the right direction.

mamoll, 2019-06-26:
I followed the bloom directions in section 2 of http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty and changed the relevant setting in the release/{melodic,dashing}/ompl branch of the ompl-release repo (see https://github.com/ros-gbp/ompl-release/commit/0d30b5953de9c08642d7837d31e758efc3c44b49).

I created pull requests for 
- Melodic: https://github.com/ros/rosdistro/pull/21624
- Dashing: https://github.com/ros/rosdistro/pull/21625

The bloom release process is still a little opaque to me, but hopefully this fixes things. :ring

v4hn, 2019-08-27:
Thank you for that cleanup @mamoll .
This means we can finally close this issue! :champagne:

closed 2019-08-27 by v4hn