moveit/moveit #176: setup assistant: split command line and gui tools

mikeferguson, 2016-08-26:
Ported from https://github.com/ros-planning/moveit_setup_assistant/issues/113

It would be great have standalone tools library and command line tool without QT dependencies.
I suggest to move the moveit_setup_assistant into a sub-directory and to factor out the non-GUI code into a moveit_setup_tools package.

mikeferguson, 2016-08-26:
Might also be good to add additional command-line arguments as mentioned in https://github.com/ros-planning/moveit_setup_assistant/issues/87

v4hn, 2016-08-26:
@ipa-mdl Why? you never gave a reason why this would make sense.

ipa-mdl, 2016-08-27:
> @ipa-mdl Why? you never gave a reason why this would make sense.

The key is "without QT dependencies". I have written the command line tools (ros-planning/moveit_setup_assistant#108) for headless creation and maintenance of SRDFs. With extensive automated testing (Travis etc.) and container-based deployment (Docker etc.) clean dependencies get more important again.

`moveit_ros_visualization` adds 100 debian packages and 350 MB of disk usage which are not needed for headless operation.

@mikeferguson: I think ros-planning/moveit_setup_assistant#87 is already addressed by `collisions_updater`

v4hn, 2016-08-27:
Oh, I haven't seen the `collision_updater` before. Sounds like it's pretty useful!
@ipa-mdl: Could you maybe add a short description of this tool to [the setup_assistant tutorial](http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html#additional-reading)?
That would be awesome!
Note the "Edit on Github" link on top of the page :)

davetcoleman, 2016-08-28:
+1 to adding to the tutorial

davetcoleman, 2019-06-24:
Given this issue is 3 years old, I will close this until someone with a strong use case presents a prototype...

closed 2019-06-24 by davetcoleman