moveit/moveit #185: UnionConstraintSampler does not satisfy constraints for one sampler

scottpaulin, 2016-08-28:
In the case of samplers_.size() == 1, it looks like the UnionConstraintSampler just samples a random state (set with state.setRandomPositions()) instead of using samplers_[0]. This means that the returned state might not satisfy the constraints of samplers_[0].

```
bool constraint_samplers::UnionConstraintSampler::sample(robot_state::RobotState &state, const robot_state::RobotState &reference_state, unsigned int max_attempts)
{
  state = reference_state;
  state.setToRandomPositions(jmg_);

  if (samplers_.size() >= 1)
  {
    if (!samplers_[0]->sample(state, reference_state, max_attempts))
      return false;
  }

  for (std::size_t i = 1 ; i < samplers_.size() ; ++i)
  {
    if (!samplers_[i]->sample(state, state, max_attempts))
      return false;
  }
  return true;
}
```

de-vri-es, 2016-08-28:
From the snippet you quote it looks like there is always a call to `samplers_[0]->sample(...)` if `samplers.size() > 0`. Note the condition on the `if` statements is `>= 1`, or in other words `> 0`.

It also looks like that can be folded into a single loop starting at `i = 0`, but it doesn't seem to ignore the samplers.

v4hn, 2016-08-28:
@scottpaulin Sorry, but I agree with @de-vri-es , your explanation does not match the code you reference.
In the first condition `samplers_[0]->sample` is called if it exists.
Please elaborate or close the issue :)

scottpaulin, 2016-08-30:
Ah sorry I missed the = part :/.

closed 2016-08-30 by scottpaulin