moveit/moveit #190: Attached bodies in rviz trail

Jntzko, 2016-08-30:
At the moment attached bodies are not displayed in the trail of the generated robot trajectory.
However, it would be nice to see them there.

[image]

davetcoleman, 2016-09-07:
This should be possible since that Rviz display plugin takes as input a ``moveit_msgs/DisplayTrajectory.msg`` that includes a ``moveit_msgs/RobotState.msg`` that includes a vector ``AttachedCollisionObject[]``

You simply need to add this notion to ``moveit/moveit_ros/visualization/rviz_plugin_render_tools/src/trajectory_visualization.cpp``

thummel26, 2017-10-18:
Presently working on this

davetcoleman, 2017-10-18:
i suspect this image is misleading and collision objects are never attached - rather they are only in the RobotState for start and goal

rhaschke, 2017-10-19:
@davetcoleman There is a fundamental issue in the design of `moveit_msgs/DisplayTrajectory.msg`: There is only a single field for the (initial) `RobotState`. However, if we a have a more complex trajectory, e.g. due to pick + place, the RobotState, particularly the set of attached objects, might change multiple times along the whole trajectory.
Hence, I suggest to extend `DisplayTrajectory.msg` to take an array of `RobotState`s (of same dimension as `trajectory`). What do you think? Probably this will break some things elsewhere?

v4hn, 2017-11-17:
> Hence, I suggest to extend `DisplayTrajectory.msg` to take an array of `RobotState`s (of same dimension as `trajectory`).

I gave this a bit of thought and I like the idea.

Thoughts from my side:

- Filling in the array should be optional and partially filled-in arrays should be supported to

- I would like to encourage users to set `is_diff == true` for all RobotState msgs in there in a comment.
  Also to make clear that they *don't* have to fill in the whole robot state if they want to specify a changed attached object

- The entries in the array should be applied (usually as diffs) at the end of each trajectory segment in `trajectory`

- I would propose `changed_state` as a name

@rhaschke It would be nice to have this, but someone has to add support for it to rviz.
I don't see merit in adding it to `moveit_msgs` before there's a patch for the rviz interpretation too.
Otherwise it's just bloat in the msg.

If anyone wants to continue on the change proposed by @rhaschke please create a new issue or a pull-request for it.
This issue is about a more basic bug.

v4hn, 2017-11-17:
@thummel26 Did you manage to implement a patch for this issue last month?

felixvd, 2020-03-23:
I think this can be closed.

rhaschke, 2020-03-23:
Yes, fixed in #1766

closed 2020-03-23 by rhaschke