moveit/moveit #192: CollisionRobot distance calculation isn't group specific

jrgnicho, 2016-08-31:
The [CollisionRobot::distanceSelf](http://docs.ros.org/indigo/api/moveit_core/html/classcollision__detection_1_1CollisionRobot.html) returns  the shortest distance between any two links in the robot at the given state.  However, the current interface doesn't allow to calculate the distance between a link in a given planning group and another link that's not part of the group.  Since motion planning in moveit is always computed for a specific planning group then allowing  a planner  to perform distance calculations for the planning group of interest would be very helpful.

v4hn, 2016-09-18:
Interesting idea.

Did you check how `CollisionRobot::checkSelfCollision` behaves when you specify the `group_name` field of the `CollisionRequest`?

Strakovo, 2023-09-22:
That's true. The planning_scene->checkSelfCollision() computes collision distances of links which are not included in the specified group. This happened me in an example similar to this:
collision_detection::CollisionRequest collision_request;
collision_request.distance=true;
collision_request.group_name="self";
...
planning_scene->checkSelfCollision(collision_request, collision_result, *kinematic_state, acm);
Part of cell.srdf:
    <group name="self">
     <link name="robot_link_1_s" />
    </group>
But the checkSelfCollision() result calculates distance of <link name='tank'> and <link name='extruder'> which are both not a member of the group "self". It is not clear to me when and why this happens. With some links this happens, with some other does not happen.
Xacro is defined with several included xacros. The shortened structure of xacro is similar like this:
<robot name="cell">
 <robot>
  <link name="robot_base_link">
  <link name="robot_link_1_s">
  <link name="robot_link_2_l">
  <link name="robot_link_3_u">
  <link name="robot_link_4_r">
  <link name="robot_link_5_b">
  <link name="robot_link_6_t">
 </robot>
 <link name="tank" meshfile="..." />

 <link name="extruder">
  <collision><geometry><box size="0.38 0.22 0.32"/></geometry></collision>
  <collision><geometry>
    <origin rpy="0 ${pi/2} 0" xyz="0.245 0 0.2" /><cylinder radius="0.09" length="0.15"/>
  </geometry></collision>
 </link>

 <link name="column">
  <collision><origin rpy="0 0 0" xyz="0 0 2.25"><geometry><box size="1.020 1.440 4.5"/></geometry></collision>
 </link>
</robot>

The links are concatenated with joints.

Strakovo, 2023-09-22:
That's true. The planning_scene-&gt;checkSelfCollision() computes collision distances of links which are not included in the specified group. This happened me in an example similar to this:
collision_detection::CollisionRequest collision_request;
collision_request.distance=true;
collision_request.group_name="self";
...
planning_scene-&gt;checkSelfCollision(collision_request, collision_result, *kinematic_state, acm);
Part of cell.srdf:
    &lt;group name="self"&gt;
     &lt;link name="robot_link_1_s" /&gt;
    &lt;/group&gt;
But the checkSelfCollision() result calculates distance of &lt;link name='tank'&gt; and &lt;link name='extruder'&gt; which are both not a member of the group "self". It is not clear to me when and why this happens. With some links this happens, with some other does not happen.
Xacro is defined with several included xacros. The shortened structure of xacro is similar like this:
&lt;robot name="cell"&gt;
 &lt;robot&gt;
  &lt;link name="robot_base_link"&gt;
  &lt;link name="robot_link_1_s"&gt;
  &lt;link name="robot_link_2_l"&gt;
  &lt;link name="robot_link_3_u"&gt;
  &lt;link name="robot_link_4_r"&gt;
  &lt;link name="robot_link_5_b"&gt;
  &lt;link name="robot_link_6_t"&gt;
 &lt;/robot&gt;
 &lt;link name="tank" meshfile="..." /&gt;

 &lt;link name="extruder"&gt;
  &lt;collision&gt;&lt;geometry&gt;&lt;box size="0.38 0.22 0.32"/&gt;&lt;/geometry&gt;&lt;/collision&gt;
  &lt;collision&gt;&lt;geometry&gt;
    &lt;origin rpy="0 ${pi/2} 0" xyz="0.245 0 0.2" /&gt;&lt;cylinder radius="0.09" length="0.15"/&gt;
  &lt;/geometry&gt;&lt;/collision&gt;
 &lt;/link&gt;

 &lt;link name="column"&gt;
  &lt;collision&gt;&lt;origin rpy="0 0 0" xyz="0 0 2.25"&gt;&lt;geometry&gt;&lt;box size="1.020 1.440 4.5"/&gt;&lt;/geometry&gt;&lt;/collision&gt;
 &lt;/link&gt;
&lt;/robot&gt;

The links are concatenated with joints.