moveit/moveit #197: Change Default Planner

v4hn, 2016-09-02:
Users often complain about ridiculous trajectories when they first use MoveIt.
@130s ' video https://www.youtube.com/watch?v=e4XMzfvK6II beautifully demonstrates that the default planner is not really adequate anymore.

@davetcoleman could you look into the details here and provide a patch that changes the default planner to something that - in your experience - produces better trajectories?

mikeferguson, 2016-09-02:
Is it really that the planner is wrong? I've seen this as well (recently) and based on the output I saw on the terminal, it looked like it was the right planner but with either no (or default) parameters.

rbbg, 2016-09-08:
This is pretty much expected from (non-optimal) random sampling based planners and they will all occasionally provide these paths. Better configuration/parameters help but do not eliminate it completely. The planner that is selected as an alternative (RRTConnect) does the same, but perhaps a little less than the current default (LBKPIECE).

However, the current default planner is not set by MoveIt! but by OMPL and they have the following snippet on their [planners page](http://ompl.kavrakilab.org/planners.html):

> If the state space has a default projection (which is going to be the case if you use any of the built-in state spaces), then it will use LBKPIECE if a bidirectional planner can be used and otherwise it will use KPIECE. These planners have been shown to work well consistently across many real-world motion planning problems, which is why these planners are the default choice. In case the state space has no default projection, RRTConnect or regular RRT will be used, depending on whether a bidirectional planner can be used.

Changing the default planner to something else is pretty easy, but I wanted to point out that the current planner is not an arbitrary choice. Perhaps it is better to leave the default selection to the planning library?

v4hn, 2016-09-08:
Thanks for the background Ruben!

Most of all, I wanted to point out that, in my experience, the out-of-the-box experience of MoveIt suffers *a lot* (and I mean **a lot**) because the default planner generates pretty random trajectories 9/10 times and things look **much** better once you use a different planner (e.g. BKPIECE).

I think it might even be a good idea to switch to a Path-Length-optimizing planner for the default (e.g. PRMstar works really well over here),
but I do not have enough enough background with OMPL for now, so I'll leave that decision to others.

Either way, we should only change the default planner in a fresh ROS distro release, so the default behavior will not change for existing setups. I would really like to see the current state improved in kinetic.

gavanderhoorn, 2016-09-08:
Just another data point, but

@v4hn wrote:
> [..] things look **much** better once you use a different planner (e.g. BKPIECE).

We've seen some pretty convoluted paths coming out of that for certain robot geometries as well. So much so, that some people have even decided to avoid any kind of KPIECE planner by removing the projection parameters from the srdf (see ros-industrial/universal_robot#193 and ros-industrial/universal_robot#196 fi). That was before ros-planning/moveit_ros#625 was merged.

v4hn, 2016-09-24:
ping @davetcoleman 
I just saw you removed this from the kinetic milestone last week.
What's your opinion on this matter?
I believe if we want to change the default, it would be wise to do so with the release into kinetic.

davetcoleman, 2016-09-24:
I don't use the Rviz Motion Planning Plugin, nor moveit_planners_ompl, for various reasons. So I am not motivated to spend my time on this. Certainly it would be nice to have fixed and I'll review, but I have my own things to focus on.

v4hn, 2016-09-24:
@davetcoleman Ah, I didn't know you don't use `moveit_planners_ompl`.
I basically wanted to know if you would have a preferred OMPL planner that we could use to improve the default planning behavior in MoveIt!

@mamoll From the side of OMPL, what would you propose as a good default planner for MoveIt! ? As discussed above, all moveit configs include projection parameters by default and thus OMPL defaults to LBKPIECE. However, this often yields *very* impractical and unintuitive trajectories and most beginners are very puzzled because of this.
Team Delft successfully used RRT-Connect at APC2016.
Should it be an option to change the default planner to an optimizing planner? Personally, I have positive experience with PRMstar.

Overall, I simply have not enough experience with different planners myself to make a justified choice for a default planner, but I noticed that many people are not happy with the current default.

davetcoleman, 2016-09-25:
I hope @mamoll provides his expert advice, but I'll add that I almost always use RRTConnect when not using my own experimental stuff. PRMstar is slower so I wouldn't suggest it be default.

mamoll, 2016-09-25:
The TLDR version of text below: There is no easy solution.

It is difficult to pick a universal planner that works well for everything, esp. if you also want universal parameter settings. OMPL has some logic to pick a reasonable planner and reasonable parameters for that planner, but there is no magical silver bullet. Many of the optimizing planners like RRT\* and PRM\* are *asymptotically* optimal and will use *all* the time you give it to find a solution, even if a "reasonable" solution is found with a fraction of the total time budget. The convergence rate tends to be pretty slow. In fact, we [have shown](http://www.cs.rice.edu/~mmoll/publications/luna2013anytime-solution-optimization.pdf) (as well as [Jingru Luo and Kris Hauser](http://motion.pratt.duke.edu/papers/IROS2014-OptimalBenchmarking.pdf)) that the convergence rate of repeatedly running a non-optimizing planner, combined with path smoothing and path hybridization, converges faster to optimal. You *can* have an optimizing planner terminate whenever it finds the first (reasonable) solution by adjusting an optimization threshold, but it's not clear whether this is any better than the current defaults. In my experience, post-processing a path from a non-optimizing planner seems to work reasonably well. It might be a good idea if someone could define a:

1. MoveGroup-specific default planner.
2. Corresponding planning parameters.
3. The bag of tricks to post-process a path (i.e., local optimization). This can be as simple as path short-cutting, more complex path smoothing, or full on path optimization (TrajOpt, CHOMP).
4. How much time to spent on the planner vs. the optimization.

(Addressing issue #117 would go a long way in making this easy.)

If you think finding a feasible path is easy and optimization is hard, then you can think of finding a feasible path as the pre-processing step that bootstraps a candidate solution for the optimization.

v4hn, 2016-09-26:
So overall, there's a lot to be done. :D

Changing to an optimizing planner for the default does not seem to be welcomed by both you and @davetcoleman, so I believe that's out of question.

Of course there is no general purpose planner & parameterization that wins gold in all categories,
otherwise why bother to write a huge planning framework such as OMPL at all. :)
The original question was meant to be whether e.g. RTTConnect (which is quite popular recently) is a justified improvement over the current default LBKPIECE *for the default* planner in MoveIt. If people are not happy with the default, they can of course change it.

I just opened one of my MoveIt setups (a UR5) in demo mode, set a random goal state and planned with LBKPIECE and RRTConnect. These are the resulting trajectories:

LBKPIECE (after around 15s of planning time):
[image]

RRTConnect (around 3s):
[image]

This is the first impression many users got over the last year(s?). I sure know which I would prefer to see when working with MoveIt for the very first time.

So I would propose to either set the default planner explicitly within MoveIt to `RRTConnect` or remove the default projection parameters from the setup assistant templates as @gavanderhoorn mentioned [above](https://github.com/ros-planning/moveit/issues/197?_pjax=%23js-repo-pjax-container#issuecomment-245537818). I believe @davetcoleman added the parameters there a while ago, so I'm interested in both your opinions on whether either of these makes some sense as a change for MoveIt kinetic.

mamoll, 2016-09-26:
Wow! I'd not be very happy if LBKPIECE would often take that long to produce a path that crappy! The \*KPIECE family of planners relies heavily on a projection to guide the sampling/exploration. Ideally, the projection is chosen so that you can distinguish collision-free configurations as best as possible. If the projection is very bad, you might as well not have any projection at all! You can simulate that by setting the cell dimensions to be very large so that all configurations project to the same cell.

Is the projection robot-specific? How is it computed / chosen?

@isucan, if you have time, you should check this out. It's a little disappointing that the best out-of-the-box experience is provided by a planner proposed more than 15 years ago. Many newer planners rely on additional information (like a projection) that *someone* has to provide. If that additional information cannot be provided/computed, then you are stuck with RRTConnect.

gavanderhoorn, 2016-09-26:
@mamoll wrote:
> Is the projection robot-specific? How is it computed / chosen?

It is generated by the Setup Assistant using [MoveItConfigData::decideProjectionJoints(..)](https://github.com/ros-planning/moveit/blob/0ceb71ca634f7f9e25adc417304a9eca460465dc/moveit_setup_assistant/src/tools/moveit_config_data.cpp#L673-L705).

mamoll, 2016-09-26:
I can see the logic behind choosing the first two joints (which cause most likely the largest swept volume on average), but perhaps projecting the joints to the end effector position using the FK would be more informative. Would that be too expensive? Is that easy to try out?

davetcoleman, 2016-09-26:
Yes, years ago I wanted to be able to test LBKPIECE and it wouldn't work by default in MoveIt! for robots unless a projection was provided. I talked to @isucan and he suggested the default projection just be the first two joints, so I added that to the [setup assistant](https://github.com/ros-planning/moveit/commit/2bc9a9885cd02b79c6626ffa05c43810843a46ae). It seems that adding that to the MSA had the side-effect of making it the default planner, due to the logic [here](https://github.com/ompl/ompl/blob/master/src/ompl/tools/config/src/SelfConfig.cpp#L275). 

It would be really great to get https://github.com/ros-planning/moveit/issues/117 merged into the main MoveIt! release, but I don't know who has the cycles for this right now. I really dislike the ``moveit_planners_ompl`` class as it is now, it has a lot of dead code and unnecessary abstraction layers. I've forked and cleaned up that class for myself, but @mamoll's proposal is even better.

If there is no interest in that right now, we can still easily implement our own [getDefaultPlanner](https://github.com/ros-planning/moveit/blob/477b7a3ba20fb6c41f165850a6420e12ef520c34/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L586) and just make it point at RRTConnect. I would support this.

mamoll, 2016-09-26:
It'd be great if #117 could get merged. I'd be happy to help, but I simply don't know what the next step is. All I can say that even though our replacement moveit_ompl_interface hasn't been touched for a while, it still works great for us. I guess the problem is that there might be improvements in the `official' ompl_interface that need to be incorporated in our replacement? Even if we do that, that still doesn't solve the problem of automatically defining reasonable parameter settings and projections. It just makes it easier to solve.

v4hn, 2016-09-26:
> Wow! I'd not be very happy if LBKPIECE would often take that long to produce a path that crappy!

LOL. That's pretty much the reaction I expected. :D 
So now you see what I've been talking about.

So it seems we all agree that RRTConnect should be the default planner for MoveIt! for now :)
Whoever has time to do that in the near future is welcome to!

The question of how to choose better parameters/projections is separate from this issue imho, but surely every useful insight is welcome there.

Also the request to merge `moveit_ompl_interface` in there is out in #117 and we'll see when someone gets around to push this forward...

mamoll, 2016-09-26:
> So it seems we all agree that RRTConnect should be the default planner for MoveIt! for now :)

Agreed. Until there is a more satisfactory solution, this seems to be the least bad fix.

davetcoleman, 2016-10-31:
I'm currently testing/fixing up the MoveIt! tutorials, and it seems there are often failed plans because its using LBKPIECE by default. By removing the ``projection_evaluator`` from ``ompl_planning.yaml`` RRTConnect is chosen by default, and the tutorials work reliably

tutorial issue background: https://github.com/ros-planning/moveit_tutorials/pull/34

gavanderhoorn, 2016-11-01:
@davetcoleman: do you really want to remove the projection evaluator? We could just set the `default_planner_config` entry in `ompl_planning.yaml`?

v4hn, 2016-11-01:
On Tue, Nov 01, 2016 at 03:02:10AM -0700, G.A. vd. Hoorn wrote:
> @davetcoleman: do you really want to remove the projection evaluator? We could just set the `default_planner_config` entry in `ompl_planning.yaml`?

Is it that complex to just specify RRTConnect as the default planner if the field is not overriden in ompl_planning.yaml?
This sounds much better to me than fixing this on an individual basis...

gavanderhoorn, 2016-11-01:
@v4hn wrote:
> Is it that complex to just specify RRTConnect as the default planner if the field is not overriden in ompl_planning.yaml?
This sounds much better to me than fixing this on an individual basis...

I actually was just reminded of the fact that `default_planner_config` only sets the planner in the RViz plugin. It doesn't affect planner selection globally.

Perhaps we should add support for that, and in the absence of that parameter, or a specific planner configured in the request, use `RRTConnect`. That would seem to result in what @davetcoleman wants, would solve the 'default planner results are crap' for (a) (the) large(st) share of moveit users.

rbbg, 2016-11-01:
The `default_planner_config` entry only changes the default setting of the Rviz motion planning plugin, so that's not a solution for anyone using the moveit_commander or the move_group_interface.

Also, commenting or removing the `projection_evaluator` setting only helps for updated moveit_configs. I would personally be in favor or making the change somewhere else so that old configs (_with_ the `projection_evaluator` and _without_ the `default_planner_config`) will also use RRTConnect as a default planner. This will also allow you to keep the `projection_evaluator` for when people do want to use LBKPIECE. Thoughts?

davetcoleman, 2016-11-01:
No I wasn't advocating removing the ``projection_evaluator``, but confirming for lots of benchmarking that using LBKPIECE as the default is a bad idea. We should change it elsewhere instead.

rbbg, 2016-11-01:
I just ran some tests (In [this](https://github.com/rbbg/moveit/tree/ompl_default) branch) with a potential approach.

 - When the planner configurations are loaded from the server, the PlannerConfigurationMap is checked for a `'group_name + "[default]"'` configuration, when it is not found, it creates one with the type set to `geometric::RRTConnect`
 - When the planningContext is generated and the planner_id is empty, the `"group_name"[default]` configuration is selected.

Doing it like this, RRTConnect will be used by default, but you can also override it from the `ompl_planning.yaml` by specifying a "default" configuration. One limitation is that you cannot have different overrides for different planning groups as you can only have a single configuration named "default". This would also cause _old_ configurations (_with_ a `projection_evaluator` and _without_ a `default_planner_config`) to default to RRTConnect. In addition, you can still use planners that require a projection evaluator without the need to change the `ompl_planning.yaml` config.

davetcoleman, 2016-11-01:
I looked at your branch, seems like a good fix! Mind creating a PR? Be sure to run clang-format on it.

davetcoleman, 2016-11-02:
Thank you @rbbg for addressing this crucial reliability fix!

closed 2016-11-02 by davetcoleman

mehditlili, 2019-07-23:
Is it normal that RTTConnect results in such jumping plans when  used with constraints?

[image]

felixvd, 2019-07-24:
1) This probably belongs in a separate issue

2) I haven't used it with a 7-DOF robot like this, but it looks like the planner might be checking that the IK result for each of the 6D poses is not in collision, not that those IK results form a feasible joint trajectory? The path of your purple object itself looks pretty smooth.