moveit/moveit #200: Can't locate move_group node in package moveit_ros_move_group

bhavyadoshi26, 2016-09-05:
I have done the binary installation of the MoveIt for Jade using [this](http://moveit.ros.org/install/) link. After doing this, I followed [this](https://www.youtube.com/watch?v=asg-thB3mwA) video and created the config package for my manipulator. After my package was created, I ran the demo.launch for the first time and I got the following error : 
`...
auto-starting new master
process[master]: started with pid [43949]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to eb76e8ea-711b-11e6-8339-480fcf447f3c
process[rosout-1]: started with pid [43963]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [43980]
process[robot_state_publisher-3]: started with pid [43981] 
ERROR: cannot launch node of type [moveit_ros_move_group/move_group]: can't locate node [move_group] in package
[moveit_ros_move_group]
process[rviz_ipt_d_0342_43937_3307353756922455826-5]: started with pid [43982]
[ WARN] [1472827541.765044325]: The root link base_link has an inertia specified in the URDF, but KDL does not     
support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.`

The URDF works perfectly fine with RViZ and Joint State Publisher GUI sliders (without MoveIt) and after a while I get the interactive markers in RViZ (with MoveIt) and I am able to move the arm around but it says 'No Planning Library Loaded'. Also system dependencies seem to be satisfied. I'm unable to find out what the exact issue is. 

Here is a screenshot of RviZ after running demo.launch : 

[image]

gavanderhoorn, 2016-09-05:
Directed here from [MoveIt demo.launch error](http://answers.ros.org/question/243002) on ROS Answers.

lakehanne, 2016-10-04:
Has this been fixed, guys?

bhavyadoshi26, 2016-10-06:
I don't think so. I went ahead with source installation of moveit and I could continue working.

lakehanne, 2016-10-08:
Looks like WS Tool is recommended in installing Move-it! http://moveit.ros.org/install/source/

130s, 2017-01-12:
We need more info. Could you provide more info, particularly:

- output of followings:

```
env|grep ROS
rospack find moveit_ros_move_group
```
- Launch file that causes the issue.

I just saw exactly the same error while I'm using source built MoveIt! but that was simply because `move_group` node wasn't built.

v4hn, 2017-03-20:
We had more than one release since this was filed. if you still encounter this problem, feel free to continue this issue. I'll close it for now.

manacoa, 2017-04-03:
+1. ROS Kinetic on ARM.

```
env|grep ROS
ROS_ROOT=/opt/ros/kinetic/share/ros
ROS_PACKAGE_PATH=/home/ubuntu/catkin_ws/src:/home/ubuntu/ws_moveit/src/fcl:/home/ubuntu/ws_moveit/src/geometric_shapes:/home/ubuntu/ws_moveit/src/moveit/moveit:/home/ubuntu/ws_moveit/src/moveit/moveit_commander:/home/ubuntu/ws_moveit/src/moveit_msgs:/home/ubuntu/ws_moveit/src/moveit/moveit_planners/moveit_planners:/home/ubuntu/ws_moveit/src/moveit/moveit_plugins/moveit_plugins:/home/ubuntu/ws_moveit/src/moveit_resources:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/moveit_ros:/home/ubuntu/ws_moveit/src/moveit/moveit_runtime:/home/ubuntu/ws_moveit/src/ompl:/home/ubuntu/ws_moveit/src/srdfdom:/home/ubuntu/ws_moveit/src/moveit/moveit_core:/home/ubuntu/ws_moveit/src/moveit/moveit_plugins/moveit_controller_manager_example:/home/ubuntu/ws_moveit/src/moveit/moveit_experimental:/home/ubuntu/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner:/home/ubuntu/ws_moveit/src/moveit/moveit_planners/chomp/chomp_interface:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/perception:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/planning:/home/ubuntu/ws_moveit/src/moveit/moveit_plugins/moveit_fake_controller_manager:/home/ubuntu/ws_moveit/src/moveit/moveit_kinematics:/home/ubuntu/ws_moveit/src/moveit/moveit_planners/ompl:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/move_group:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/manipulation:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/robot_interaction:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/warehouse:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/benchmarks:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/planning_interface:/home/ubuntu/ws_moveit/src/moveit/moveit_ros/visualization:/home/ubuntu/ws_moveit/src/moveit/moveit_setup_assistant:/home/ubuntu/ws_moveit/src/moveit/moveit_plugins/moveit_simple_controller_manager:/home/ubuntu/ws_moveit/src/moveit/moveit_plugins/moveit_ros_control_interface:/home/ubuntu/ws_moveit/src/moveit_visual_tools:/opt/ros/kinetic/share
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=/home/ubuntu/catkin_ws/devel/share/common-lisp:/home/ubuntu/ws_moveit/devel/share/common-lisp
ROS_DISTRO=kinetic
ROS_IP=192.168.2.2
ROS_ETC_DIR=/opt/ros/kinetic/etc/ros
```
and

```
rospack find moveit_ros_move_group
/home/ubuntu/ws_moveit/src/moveit/moveit_ros/move_group
```
If it helps, the directory /home/ubuntu/ws_moveit/src/moveit/moveit_ros/move_group/scripts/ contains only two scripts load_map and save_map. There are other executable scripts anywhere in the package that I can see.

130s, 2017-04-03:
For Kinetic the latest version 0.9.5 is recommended where the issue should be solved. If that doesn't solve please let us know your version:

```
rosversion moveit_ros_move_group
```

manacoa, 2017-04-03:
rosversion moveit_ros_move_group
gives 
    0.9.4

However, when I deinstall and reinstall it again with
   sudo apt-get install ros-kinetic-moveit-ros-move-group
it claims the version being installed is
   0.9.5-0xenial-20170315-215430-0700

rosversion still claims it to be 0.9.4 though.

130s, 2017-04-03:
> ```
> sudo apt-get install ros-kinetic-moveit-ros-move-group
> ```
> 
> it claims the version being installed is
> 
> ```
> 0.9.5-0xenial-20170315-215430-0700
> ```
> 
> rosversion still claims it to be 0.9.4 though.

That's strange. Please make sure that your `ROS_PACKAGE_PATH` points to where the latest package is installed (e.g. if you want to use binary, that should be `/opt/ros/...`).

I got the following:

```
$ rosversion moveit_ros_move_group
0.9.5
[1]+  Done                    sudo synaptic
$ apt-cache policy ros-kinetic-moveit-ros-move-group 
ros-kinetic-moveit-ros-move-group:
  Installed: 0.9.5-0xenial-20170327-171505-0700
  Candidate: 0.9.5-0xenial-20170327-171505-0700
:
$ rospack find moveit_ros_move_group
/opt/ros/kinetic/share/moveit_ros_move_group
```

SimonBirrell, 2017-04-10:
Oops, sorry. I had an old attempt at installing MoveIt from source lying around.

0.9.5 works fine, thanks.

v4hn, 2017-04-11:
@SimonBirrell are you related to one of the previous posters here?
Your nick is not mentioned in this thread before.

manacoa, 2017-04-11:
Oops, sorry. I had an old attempt at installing MoveIt from source lying around.

0.9.5 works fine, thanks.

closed 2017-04-11 by v4hn

GNREX, 2018-03-30:
Having similar problem as manacoa ,
 
$ rospack find moveit_ros_move_group   returns
/home/robin/ros_ws/src/moveit/moveit_ros/move_group

I also tried to install MoveIt from source (failed) and then installed binary.

How do i now make it so ROS sees the binary files that are in 

/opt/ros/kinetic/share/moveit_ros_move_group  ??

Murtaza786-oss, 2019-08-12:
Hello everyone, I have been following your book 'Mastering_ROS_for_Robotics_Programming'. I am stuck in chapter 10 where I am not able to execute a launch file i.e. `roslaunch seven_dof_arm_gazebo grasp_generator_server.launch.`

When I execute this I get the following error,
[image]

I have download the updated folder of moveit_simple_grasps as provided above but when I run catkin_make , I get the following errors 

```
Scanning dependencies of target moveit_simple_grasps_test
Scanning dependencies of target moveit_simple_grasps_server
Scanning dependencies of target moveit_simple_grasps_filter_test
[ 96%] Building CXX object moveit_simple_grasps/CMakeFiles/moveit_simple_grasps_test.dir/src/simple_grasps_test.cpp.o
[ 96%] Building CXX object moveit_simple_grasps/CMakeFiles/moveit_simple_grasps_server.dir/src/simple_grasps_server.cpp.o
[ 96%] Building CXX object moveit_simple_grasps/CMakeFiles/moveit_simple_grasps_filter_test.dir/src/grasp_filter_test.cpp.o
In file included from /home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp:49:0:
/home/murtaza/catkin_ws/src/moveit_simple_grasps/include/moveit_simple_grasps/custom_environment2.h: In function ‘void baxter_pick_place::createEnvironment(moveit_visual_tools::MoveItVisualToolsPtr)’:
/home/murtaza/catkin_ws/src/moveit_simple_grasps/include/moveit_simple_grasps/custom_environment2.h:86:126: warning: ‘bool moveit_visual_tools::MoveItVisualTools::publishCollisionTable(double, double, double, double, double, double, std::__cxx11::string, const rviz_visual_tools::colors&)’ is deprecated [-Wdeprecated-declarations]
   visual_tools_->publishCollisionTable(0.78,    -0.8,    0,     0.4,         1.4,          0.47,        SUPPORT_SURFACE1_NAME); // computer monitor
                                                                                                                              ^
In file included from /home/murtaza/catkin_ws/src/moveit_simple_grasps/include/moveit_simple_grasps/simple_grasps.h:60:0,
                 from /home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp:46:
/home/murtaza/catkin_ws/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:464:8: note: declared here
   bool publishCollisionTable(double x, double y, double angle, double width, double height, double depth,
        ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp:49:0:
/home/murtaza/catkin_ws/src/moveit_simple_grasps/include/moveit_simple_grasps/custom_environment2.h:87:126: warning: ‘bool moveit_visual_tools::MoveItVisualTools::publishCollisionTable(double, double, double, double, double, double, std::__cxx11::string, const rviz_visual_tools::colors&)’ is deprecated [-Wdeprecated-declarations]
   visual_tools_->publishCollisionTable(0.78,    -0.45,   0,     0.4,         0.7,          0.47,        SUPPORT_SURFACE2_NAME); // my desk
                                                                                                                              ^
In file included from /home/murtaza/catkin_ws/src/moveit_simple_grasps/include/moveit_simple_grasps/simple_grasps.h:60:0,
                 from /home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp:46:
/home/murtaza/catkin_ws/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:464:8: note: declared here
   bool publishCollisionTable(double x, double y, double angle, double width, double height, double depth,
        ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp:49:0:
/home/murtaza/catkin_ws/src/moveit_simple_grasps/include/moveit_simple_grasps/custom_environment2.h:88:126: warning: ‘bool moveit_visual_tools::MoveItVisualTools::publishCollisionTable(double, double, double, double, double, double, std::__cxx11::string, const rviz_visual_tools::colors&)’ is deprecated [-Wdeprecated-declarations]
   visual_tools_->publishCollisionTable(TABLE_X, TABLE_Y, 0,     TABLE_WIDTH, TABLE_HEIGHT, TABLE_DEPTH, SUPPORT_SURFACE3_NAME); // andy table
                                                                                                                              ^
In file included from /home/murtaza/catkin_ws/src/moveit_simple_grasps/include/moveit_simple_grasps/simple_grasps.h:60:0,
                 from /home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp:46:
/home/murtaza/catkin_ws/src/moveit_visual_tools/include/moveit_visual_tools/moveit_visual_tools.h:464:8: note: declared here
   bool publishCollisionTable(double x, double y, double angle, double width, double height, double depth,
        ^~~~~~~~~~~~~~~~~~~~~
/home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp: In constructor ‘baxter_pick_place::GraspGeneratorTest::GraspGeneratorTest(int)’:
/home/murtaza/catkin_ws/src/moveit_simple_grasps/src/simple_grasps_test.cpp:99:20: error: ‘using element_type = class moveit_visual_tools::MoveItVisualTools {aka class moveit_visual_tools::MoveItVisualTools}’ has no member named ‘setMuted’
     visual_tools_->setMuted(false);
                    ^~~~~~~~
In file included from /home/murtaza/catkin_ws/src/moveit_simple_grasps/src/grasp_filter_test.cpp:66:0:
[... cut ...]
CMakeFiles/Makefile2:8118: recipe for target 'moveit_simple_grasps/CMakeFiles/moveit_simple_grasps_server.dir/all' failed
make[1]: *** [moveit_simple_grasps/CMakeFiles/moveit_simple_grasps_server.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

I have also edited the `custom_environment2.h` providing 7 arguments instead of 5 which is,
[image]

I have tried it on both ros-melodic and ros-kinetic I get the same error on both the versions

I would be glad if you can help me out