moveit/moveit #201: MoveGroupExecuteTrajectoryAction vs. MoveGroupExecuteService

AdrianZw, 2016-09-06:
Recently, I did an update from a very old version of MoveIt to the current version of the jade-devel branch.  So far everything was working fine with the RVIZ interface and my configuration for the Jaco2.
Using RVIZ I can plan and execute path on my jaco2. However, I realized that previously working programs using the MoveGroup interface do not execute any path any more. 
Instead I get the following exception (move_group.cpp Line 224):

terminate called after throwing an instance of 'std::runtime_error'
  what():  No Trajectory execution capability available.
Aborted (core dumped)

Switching back to the deprecated MoveGroupExecuteService solves this issue. 
Unfortunately, I did not find any information, if the Moveit config/ a controller or the usage of the MoveGroup has to be changed in order to make everything run with the TrajectoryAction. 
I would be very happy if someone could give me some hints to this issue.

Here is the code I tried to run:
`int main(int argc, char *argv[])
{
    ros::init (argc, argv, "home_arm");
    ros::NodeHandle node_handle("~");
    ros::AsyncSpinner spinner(1);
    spinner.start();

    ros::Duration sleep_time(1.0);
    sleep_time.sleep();
    sleep_time.sleep();

    moveit::planning_interface::MoveGroup::Plan my_plan;
    moveit::planning_interface::MoveGroup group("manipulator");
    moveit::planning_interface::PlanningSceneInterface planning_scene_interface;

    group.setPlannerId("RRTkConfigDefault");
    group.setStartStateToCurrentState();
    group.setPlanningTime(5.0);
    group.setNamedTarget("home");
    moveit_msgs::MoveItErrorCodes success = group.plan(my_plan);
    if(success.val == moveit_msgs::MoveItErrorCodes::SUCCESS) {
        ROS_INFO("Success! Now move");
        success = group.move();
    }
    spinner.stop();
    ros::shutdown();
    return 0;}`

130s, 2016-09-10:
> Switching back to the deprecated MoveGroupExecuteService solves this issue. 

Where do you do this exactly?

AdrianZw, 2016-09-10:
I added the deprecated MoveGroupExecuteService in the move group launch file:

`    
    <param name="capabilities" value="move_group/MoveGroupCartesianPathService
				      move_group/MoveGroupExecuteService
				      move_group/MoveGroupExecuteTrajectoryAction
				      move_group/MoveGroupKinematicsService
				      move_group/MoveGroupMoveAction
				      move_group/MoveGroupPickPlaceAction
				      move_group/MoveGroupPlanService
				      move_group/MoveGroupQueryPlannersService
				      move_group/MoveGroupStateValidationService
				      move_group/MoveGroupGetPlanningSceneService
				      move_group/ApplyPlanningSceneService
				      move_group/ClearOctomapService
				      " />`

It seems that by default the MoveGroupExecuteService will not appear if you create the moveit config. using the setup assistant. However, at least with my config., I am not able to control the Jaco2 using the MoveGroup planning interface, if the Service is not listed in the move_group.launch file.

To communicate with my controller I use the ROS FollowJointTrajectoryAction and the
MoveItSimpleControllerManager, as described in the moveit tutorials. 

I do not understand why everything works fine using RVIZ, but if I run the above program the execute_action_client_ seems not to be able to connect to the FollowJointTrajectoryActionServer which is running. Am I missing something?

v4hn, 2016-11-11:
I'm pretty confident #335 fixed your problem @AdrianZw could you please test and verify this?

AdrianZw, 2016-11-14:
Indeed #335 fixed this problem. Thank you !

closed 2016-11-14 by AdrianZw

rkeatin3, 2017-12-04:
I assume the cause is different, but I am now having this issue.

I am running ROS Jade on Ubuntu 14.04 and recently moved from binary to a source installation of MoveIt. When I moved to the source installation I had to rerun the setup assistant, and doing so populated the capabilities into move_group.launch. 

When running MoveIt!, however, I get the following warning: 

```
Deprecation warning: Trajectory execution service is deprecated (was replaced by an action).
Replace 'MoveGroupExecuteService' with 'MoveGroupExecuteTrajectoryAction' in move_group.launch
```

If I actually do make the modifications to move_group.launch, I then get the following error when trying to launch MoveIt!: 

```
[ERROR] [1512426794.431275074]: Exception while loading move_group capability 'move_group/MoveGroupExecuteTrajectoryAction': According to the loaded plugin descriptions the class move_group/MoveGroupExecuteTrajectoryAction with base class type move_group::MoveGroupCapability does not exist. Declared types are  move_group/ApplyPlanningSceneService move_group/ClearOctomapService move_group/MoveGroupCartesianPathService move_group/MoveGroupExecuteService move_group/MoveGroupGetPlanningSceneService move_group/MoveGroupKinematicsService move_group/MoveGroupMoveAction move_group/MoveGroupPickPlaceAction move_group/MoveGroupPlanService move_group/MoveGroupQueryPlannersService move_group/MoveGroupStateValidationService
Available capabilities: move_group/ApplyPlanningSceneService, move_group/ClearOctomapService, move_group/MoveGroupCartesianPathService, move_group/MoveGroupExecuteService, move_group/MoveGroupGetPlanningSceneService, move_group/MoveGroupKinematicsService, move_group/MoveGroupMoveAction, move_group/MoveGroupPickPlaceAction, move_group/MoveGroupPlanService, move_group/MoveGroupQueryPlannersService, move_group/MoveGroupStateValidationService
```
Is the warning erroneous in Jade (i.e. I should keep using MoveGroupExecuteService despite that it's deprecated)? How should I fix this issue?

leader1313, 2019-09-27:
it will be fixed install ros-move-group
 `sudo apt-get install ros-indigo-moveit-ros-move-group`