moveit/moveit #210: Specify the robots' movement

mat-l, 2016-09-09:
Hello,

how can I plan and execute linear (LIN) or point to point (PTP) movements using MoveIt!   
I would like to define the distance between two waypoints using LIN and PTP movements in the command line, C++/Python programms and I would like to integrate it later maybe into the Cartesian Path Planner (https://github.com/ros-industrial-consortium/fermi)

Any ideas?
Best regards

v4hn, 2016-09-09:
Could you elaborate on this please?

mat-l, 2016-09-11:
Hi,
the point is, that i would like to tell ROS how the path should be planned for the robot. Common kinds of pathes are synchronous or asynchronous point to point movements or even linear ones.

If you have different waypoints named for example w01, w02, w03, and so one, then you can tell the robot how to move between the waypoints.

For example:
move_lin(w01, w02) //linear movement between the two waypoints   

move_ptp(w02, w03) //point to point movement between the two waypoints

My question is, how to tell ROS how to plan the path between different kinds of waypoints?

Besr regards

CMobley7, 2016-09-16:
mat-l,

For linear and point to point motion use the compute_cartesian_path() and plan() functions respectively.

V/R,
CMobley7

closed 2016-10-12 by mat-l