moveit/moveit #212: sudo for warehouse source install instructions?

davetcoleman, 2016-09-09:
@ugocupcic I'm testing you [source install](https://github.com/ros-planning/moveit.ros.org/pull/57) instructions for warehouse and on my system for the line

```
scons --prefix=/usr/local/ --full --use-system-boost --disable-warnings-as-errors
``` 

I get the error:

```
Install file: "build/linux2/use-system-boost/libmongoclient.a" as "/usr/local/lib/libmongoclient.a"
scons: *** [/usr/local/lib/libmongoclient.a] /usr/local/lib/libmongoclient.a: Permission denied
scons: building terminated because of errors.
```

But putting ``sudo`` in front of the command fixes the problem. Would you agree the instructions should be updated?

davetcoleman, 2016-09-09:
For the next step

```
sudo scons install
```

I also get the error
```
Error: You must use --use-system-boost when building the C++ driver
```

I'm now reading into mongo-cxx-driver's [install instructions](https://github.com/mongodb/mongo-cxx-driver/wiki/Download-and-Compile-the-Legacy-Driver#compile-the-driver) and wondering why we are using the ``26compat`` branch.

I'm on Ubuntu 14.04 / ROS Jade, is this similar to your setup when you tested this?

Thanks!

ugocupcic, 2016-09-12:
Yes sorry I've been running that in a docket so there was no need to sudo!

Sent from a tiny keyboard, excuse the brevity.

> @ugocupcic I'm testing you source install instructions for warehouse and on my system for the line
>
> scons --prefix=/usr/local/ --full --use-system-boost --disable-warnings-as-errors
> I get the [...]

ugocupcic, 2016-09-12:
No I did this pull request for 16.04 / kinetic. I thought I had the use system boost in there.

Sent from a tiny keyboard, excuse the brevity.

> For the next step
>
> sudo scons install
> I also get the error
>
> Error: You must use --use-system-boost when building the C++ driver
> I'm now reading into mongo-cxx-driver's install instructions and [...]

v4hn, 2016-09-13:
@davetcoleman if this is fixed via https://github.com/ros-planning/moveit.ros.org/pull/65 please close this issue.

davetcoleman, 2016-09-13:
I suspect the mongo-db install instructions are broken for 14.04 (only working on 16.04) but that should go in a different issue

closed 2016-09-13 by davetcoleman