moveit/moveit #216: moveit_core build not passing

OwlCoder, 2016-09-12:
I am getting the following error message while "catkin build"ing the moveit_core package. What could be the issue?

/home/arclab/catkin_ws/src/moveit_core/collision_detection/src/collision_octomap_filter.cpp: In function ‘int collision_detection::refineContactNormals(const ObjectConstPtr&, collision_detection::CollisionResult&, double, double, bool, double, double)’:
/home/arclab/catkin_ws/src/moveit_core/collision_detection/src/collision_octomap_filter.cpp:118:71: error: **conversion from ‘const boost::shared_ptr<const octomap::OcTree>’ to non-scalar type ‘std::shared_ptr<const octomap::OcTree>’ requested**
         std::shared_ptr<const octomap::OcTree> octree = shape_octree->octree;
                                                                       ^
make[2]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/src/collision_octomap_filter.cpp.o] Error 1
make[1]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/all] Error 2
make: *** [all] Error 2

davetcoleman, 2016-09-13:
Please use ``` to properly format your console output e.g.

```
output
```

please post your OS version and ROS version

OwlCoder, 2016-09-13:
Done. Sorry for the incorrect format.

davetcoleman, 2016-09-13:
you've ``git clone``d the kinetic-devel branch when you need the indigo branch

OwlCoder, 2016-09-15:
Yes, you were absolutely right! It was my blunder.

closed 2016-09-15 by OwlCoder

davetcoleman, 2016-09-15:
Is there something in the docs that was not clear that we can improve? Did you see [this install page](http://moveit.ros.org/install/source/)?