moveit/moveit #222: Supporting Ubuntu Wily in Continuous Integration / MoveIt!

davetcoleman, 2016-09-16:
@rhaschke requested I add Wily (Ubuntu 15.10) testing to our Travis setup. It no longer appears as easy than when I first agreed to it - there are not ROS Docker images for it, for example (@ruffsl).

I also noticed that Wily is already [end of life](https://wiki.ubuntu.com/Releases), so I was surprised that [ROS Kinetic is still supporting it](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021)

All of this is important because we've been switching to new urdfdom headers [here](https://github.com/ros-planning/moveit/pull/209) and [here](https://github.com/ros-planning/srdfdom/issues/23) but they are missing in Wily. 

Do we really need to support Wily in MoveIt? Do we have users who are going to, in the future when we finally release MoveIt! for Kinetic, choose to use an end of life non-LTS version of Ubuntu? 

This might be an outrageous question and I suppose we'll run into issues with the ROS buildfarm, but just thought I'd throw the question out there. At bare minimum, I'm not sure its worth me building custom ROS Dockers and doing lots of modifications to our Travis scripts.

v4hn, 2016-09-16:
Wow, I didn't expect wily is eol already, but sure, time flies..
For how long does OSRF plan to build packages for wily?
I'd say that we do not need to support it anymore the moment they stop building packages.
No idea who would be best to ping about this at osrf.

davetcoleman, 2016-09-16:
[This guys](http://answers.ros.org/question/220064/ros-on-ubuntu-wily/) claims support should have already been dropped, but @tfoote is certainly the best person to ask

tfoote, 2016-09-16:
As linked Wily is a Kinetic target platform. It should generally be thought of as the low water mark for changes. 

We are still building Kinetic debs for Wily. There's one or two changes we need to make to the buildfarm to support turning it off but we have not had time to make the upgrades.

davetcoleman, 2016-09-16:
So it will be turned off soon? Do you have a estimate for when? I think this is a nod that we will skip special rules to allow MoveIt! to be released for Wily

tfoote, 2016-09-16:
I doubt that it will happen soon, in the scale of a potential Kinetic moveit release.

davetcoleman, 2016-09-19:
@tfoote If we avoid adding the special ``#ifdef`` commands for wily, etc, can we still release MoveIt! into Kinetic using bloom or will it complain about the broken Wily target?

tfoote, 2016-09-19:
If it's compile type issues you will get daily emails when the buildfarm attempts to build Wily on each architecture.

davetcoleman, 2016-10-18:
I believe we decided we had to continue to support Wily because of ROS buildfarm requirements

closed 2016-10-18 by davetcoleman

davetcoleman, 2016-10-25:
We are still being blocked from releasing because of Wily, so I've reopened this discussion for all of ROS [on Discourse](http://discourse.ros.org/t/drop-support-for-wily/727)

davetcoleman, 2016-11-01:
I've created a test for Wily https://github.com/ros-planning/moveit_ci/pull/20