moveit/moveit #231: Is there any plans to add the ompl planners FMT* and BFMT*?

Levi-Armstrong, 2016-09-20:


davetcoleman, 2016-09-20:
Someone easily could, but AFAIK there are no plans to. A PR is welcome, and I can give a quick pointer of where to start

rbbg, 2016-10-09:
I had already exposed a subset of the available planners before, and figured I'd finish that and put in a PR, but I ran into issue #274 and putting this on hold until that's fixed. (both FMT* and and BFMT* seemed to work fine)

I can also expose the experience based frameworks Lightning and Thunder, but that's a little more involved and I am not sure if anyone would be interested in using them?

davetcoleman, 2016-10-11:
One of the core maintainers is interested in Lightning and Thunder :-) !

rbbg, 2016-11-01:
I have opened a PR #338 that exposes several of the 'new' OMPL planners. The ones that I didn't expose were a bit more trouble (the informed planners, InformedRRT, BIT, etc, use an informed sampler that is not compatible with the type of state space that MoveIt! uses).

With regards to Lightning and Thunder, wouldn't it be a better idea to see what would need to be done for #117 ? I'd imagine that this would also affect the way you would integrate something like Thunder/Lightning.

v4hn, 2017-01-13:
as far as I see #338 implemented this. Closing.

closed 2017-01-13 by v4hn