moveit/moveit #236: moveit_commander: no collision avoidance when using Kinect on Baxter

bmmi, 2016-09-21:
Hi there,

I am using python interface (moveit_commander) for the moveit on Baxter robot (https://github.com/ros-planning/moveit_robots/tree/kinetic-devel/baxter/baxter_moveit_config)

I try to use moveit_commander to do the motion planning with collision avoidance (point cloud from Kinect). When I launched the demo_kinect.launch, and do the planning through Rviz GUI, then moveit will indeed give collision-aware path. However I directly use moveit_commander to plan, it seems ignore the collision.

Bellow is the python code for using moveit_commander:

#!/usr/bin/env python

import sys
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
import baxter_interface

def main():

	rospy.init_node('moveit_test')

	moveit_commander.roscpp_initialize('/joint_states:=/robot/joint_states')
	# moveit_commander.roscpp_initialize(sys.argv)

	robot = moveit_commander.RobotCommander()

	baxter_interface.RobotEnable().enable()

	# scene = moveit_commander.PlanningSceneInterface()

	# group_left = moveit_commander.MoveGroupCommander("left_arm")
	
	group_right = moveit_commander.MoveGroupCommander("right_arm")

	pose_target = geometry_msgs.msg.Pose()
	pose_target.orientation.x = 0.9566752755123588
	pose_target.orientation.y = 0.2514090215247311
	pose_target.orientation.z = 0.144068320430591
	pose_target.orientation.w = -0.028464717944120376
	pose_target.position.x = 0.831
	pose_target.position.y = -0.022
	pose_target.position.z = 0.3242

	group_right.set_pose_target(pose_target)
	group_right.plan()
	group_right.go(wait=True)

	rospy.spin()

if __name__ == "__main__":

	main()

This is the output from the terminal:

[ INFO] [1474437906.102495533]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1474437906.114322575]: No planner specified. Using default.
[ INFO] [1474437906.114460276]: LBKPIECE1: Attempting to use default projection.
[ INFO] [1474437906.117579588]: LBKPIECE1: Starting planning with 1 states already in datastructure
[ INFO] [1474437907.167631395]: LBKPIECE1: Created 83 (38 start + 45 goal) states in 72 cells (38 start (38 on boundary) + 34 goal (34 on boundary))
[ INFO] [1474437907.167735236]: Solution found in 1.053163 seconds
[ INFO] [1474437907.167923582]: SimpleSetup: Path simplification took 0.000077 seconds and changed from 40 to 2 states
[ERROR] [1474437907.188444213]: Computed path is not valid. Invalid states at index locations: [ 2 3 5 6 ] out of 10. Explanations follow in command line. Contacts are published on /move_group/display_contacts
[ INFO] [1474437907.189710977]: Found a contact between '<octomap>' (type 'Object') and 'right_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1474437907.189763403]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1474437907.193585770]: Found a contact between '<octomap>' (type 'Object') and 'right_gripper_base' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1474437907.193644029]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1474437907.198037496]: Found a contact between '<octomap>' (type 'Object') and 'right_gripper_base' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1474437907.198096046]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1474437907.202010290]: Found a contact between '<octomap>' (type 'Object') and 'right_gripper_base' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1474437907.202076672]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1474437907.206050485]: Completed listing of explanations for invalid states.
[ INFO] [1474437916.940036555]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.

In the Rviz GUI I can choose the planner (RRTConnect) and collision-aware IK. But I don't know how to configure the planner (it chooses LBKPIECE1 as default) and collision avoidance with moveit_commander.
It does check for collision though.

Could you give me some information about using moveit_commander in python with collision avoidance using Kinect point cloud?

Thanks,
bmmi

v4hn, 2016-09-22:
On Wed, Sep 21, 2016 at 03:08:09PM -0700, Yinhao Zhu wrote:
> However I directly use moveit_commander to plan, it seems ignore the collision.
> [...]
> This is the output from the terminal:
> [...]
> [ERROR] [1474437907.188444213]: Computed path is not valid. Invalid states at index locations: [ 2 3 5 6 ] out of 10. Explanations follow in command line. Contacts are published on /move_group/display_contacts
> [ INFO] [1474437907.189710977]: Found a contact between '<octomap>' (type 'Object') and 'right_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored.

This looks like the code aborts because of the collision it found. Isn't that the case?
What happens? What exactly is the problem you experience?

bmmi, 2016-09-22:
I am expecting that moveit can compute a path that is collision free, not
just find a path and check for collision, and if it collides, aborts the
search. How to configure moveit to try more times to search other paths...
When I did the planning from Rviz, it gave me a collision free path. And it
used a different planner as I mentioned before.

v4hn, 2016-09-22:
> I am expecting that moveit can compute a path that is collision free, not just find a path and check for collision, and if it collides, aborts the search

Ah, sure. That makes sense. Your first comment sounded like your robot crashes into an obstacle...
It does not abort the search after it collided.
Probably one of two things happened here:

- Either your planner found a plan that did not collide on any point of the trajectory that the planner checked for collisions, but post-processing of the resulting trajectory found the collision.
 To increase the density of collision checks in the planner, reduce the value `longest_valid_segment_fraction` in `<robot>_moveit_config/config/ompl_planning.yaml`. The default 0.05 sometimes is not good enough.
- Or the planner found no non-colliding plan at all and the contact outputs represent one of the failed plans.
 Changing the planner parameters or changing the planner might help. To answer your question: you can set the planner with `group_right.set_planner_id("UseThisPlannerInstead")`.

bmmi, 2016-09-22:
Thanks, changing the planner to be `RRTConnectkConfigDefault` solves the problem. The planned path is with object-avoidance but sometimes looks unnatural...

v4hn, 2016-09-23:
ok, I'll consider this to be resolved then :)

closed 2016-09-23 by v4hn