moveit/moveit #237: CHOMP planner documentation

davetcoleman, 2016-09-22:
@ksatyaki remaining TODOs post https://github.com/ros-planning/moveit/pull/155:

- [ ] Open PRs for jade and kinetic
- [ ] Create moveit_tutorial page for using CHOMP

davetcoleman, 2016-10-20:
@ksatyaki do you mind finishing up the CHOMP integration by documenting how to use what you've done so far?

davetcoleman, 2016-10-20:
fyi I created a very basic page for you to expand upon: https://github.com/ros-planning/moveit_tutorials/pull/33

ksatyaki, 2016-11-08:
I have just completed the tutorial and issued a pull request. Sorry for the delay.

130s, 2017-10-18:
I'm interested in looking into this for WMD2017, although I'm very new to CHOMP/STOMP. Let's see how far a langsam learner :turtle: can go within a day.

@ksatyaki wrote:

> I have just completed the tutorial and issued a pull request. Sorry for the delay.

Where can we find it?

ksatyaki, 2017-10-18:
I think this issue should be closed. The tutorial is available at http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/chomp_interface_tutorial.html

davetcoleman, 2017-10-18:
I think the tutorial would be great to be reviewed and expanded upon

ksatyaki, 2017-10-19:
I will probably resurrect this planner in the near future and make sure it works reliably. At that time I shall also build upon the tutorial (unless someone wants to do it sooner, of course).

davetcoleman, 2018-05-10:
Closing because @mlautman recently reviewed this tutorial after @ksatyaki's original version

closed 2018-05-10 by davetcoleman