moveit/moveit #242: Auto-format code base with clang-format

davetcoleman, 2016-09-22:
I'd like to run [clang-format](https://github.com/davetcoleman/roscpp_code_format) on this entire repo right after the indigo and kinetic releases are finished, as I am expecting that the fewest number of pull request will be open at that moment. The code formatter is pretty amazing - in one line of code:

```
find . -name '*.h' -or -name '*.hpp' -or -name '*.cpp' | xargs clang-format-3.6 -i -style=file $1
```

We can make all our code comply to the roscpp style guidelines and then, in the future, include this as a check for CI. This change does have the tendency to break open PRs though - rebasing will likely have lots of merge conflicts (there are ways to mitigate this though) so that is why I suggest we wait until the moveit releases.

I've already run this formatter on the entire [ompl codebase](https://github.com/ompl/ompl/pull/81) successfully, and its been used by many ROS users for a while via the link on [rosccp style guidelines](http://wiki.ros.org/CppStyleGuide#Autoformatting_of_ROS_Code)

I'm announcing this intention now so that we can discuss it and, when the moment is right, make the merge quickly.

de-vri-es, 2016-09-22:
I'm very much in favor of this idea. It would allow me to work with my regular editor settings without worrying about style, and then apply the same formatting in one go before committing.

v4hn, 2016-09-22:
I don't really like syntactic rewriting like this but I see the advantages in this case. so +1 from my side.

rhaschke, 2016-10-14:
It's a huge rewrite of the whole repo. But I can see the benefits too. +1 if we enforce compliance afterwards as suggested.

davetcoleman, 2016-10-14:
The automated test has been created: https://github.com/ros-planning/moveit_ci/pull/19

davetcoleman, 2016-10-27:
code formatting is complete!

closed 2016-10-27 by davetcoleman