moveit/moveit #265: moveit kinetic build problem

arpit15, 2016-09-29:
[moveit_build_error.txt](https://github.com/ros-planning/moveit/files/501359/moveit_build_error.txt)

fcl is compiling % . When I catkin build there is the error as listed in the attached file.

davetcoleman, 2016-09-29:
please use https://gist.github.com/

davetcoleman, 2016-09-29:
do you have ``libfcl-0.5.-dev`` and ``libfcl0.5`` installed via debian? have you run ``rosdep update``? did you run the ``rosdep install`` command?

arpit15, 2016-09-29:
I installed from source using the commands from http://moveit.ros.org/install/source/dependencies/. 
I ran `rosdep update`. I didn't use `rosdep install`

davetcoleman, 2016-09-29:
Ah, those are not official build instructions but rather experimental. Please specify that next time. Let us know if you find improvements in those instructions.

arpit15, 2016-09-30:
Thank you @davetcoleman. I was able to successful build moveit with fcl-0.5 for kinetic on 16.04.

closed 2016-09-30 by v4hn

davetcoleman, 2016-09-30:
Did you find any mistakes in my FCL instructions that could be fixed?

arpit15, 2016-09-30:
@davetcoleman I used `git checkout fcl-0.5` instead of your command `git checkout -b tags/0.5.0` and everything build perfectly.

davetcoleman, 2016-09-30:
i just confirmed your fix (i wrote those instructions after the fact of setting up FCL locally, sorry).

do you mind making a quick PR to the moveit website to fix the instructions? 
https://github.com/ros-planning/moveit.ros.org

arpit15, 2016-09-30:
[@88](https://github.com/ros-planning/moveit.ros.org/pull/88) created PR with corrected instructions