moveit/moveit #270: Removing catch(...) instances?

bmagyar, 2016-10-04:
Hi guys,

I'm working on my own planner and obviously screwed up something in it. I discovered that there are some instances of `catch(...)` basically hiding crashes & possible informative error messages.

Wouldn't it be better to eliminate the use of  `catch(...)` overall in MoveIt? 

Adding some sense to my point:
- Users are given an error message they can use to report their issue, more info for maintainers to address issues, `std::runtime_error`s are handled & printed with some information, `catch(...)` with no information printed only "Planning pipeline threw an exception" is useless.
- Planner issues are raised to the top level, an `std::terminate` after all is pretty harsh, more chance it will get reported
- "Keep going" is probably not a good philosophy when a planner has unhandled exceptions propagate up

https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/move_group/src/default_capabilities/move_action_capability.cpp#L157

https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/move_group/src/default_capabilities/plan_service_capability.cpp#L69

v4hn, 2016-10-04:
I mostly agree,

these clauses do not really belong there.
`std::runtime_error` should be handled there, but other exceptions should not.
Removing these clauses, however, means that *your entire move_group node* dies if your planner pipeline throws an exception. This is most likely the reason why these clauses are there at all. If your planning request fails due to some internal error, you might still expect to be able to do cartesian path planning (which doesn't use the planner).

I would argue though that all exceptions that are not due to architecture design and misused API, should be `runtime_error`s anyway.

Could you provide a pull-request that removes the cases you noticed in the `kinetic-devel` branch?

davetcoleman, 2016-10-04:
This seems like a debate between how much MoveIt! should be focused on helping developers and research, versus being "hardened" to continue working even after a crash occurs in the planner, for improved reliability.

As a researcher developing his own planner, I am obviously for removing the ``catch`` statements. +1

bmagyar, 2016-10-04:
Yeah I've been a user of moveit for a long time, first at PAL Robotics now as a PhD student. Error messages matter in both cases. `catch(...)` hides too much. I'm up for other solutions instead of bare removal, if anyone has a short snippet of a gentle coredump-style or some template "print any exception" solution, that's also good for me. It's more about not letting bugs live under rocks like `catch(...)`.

de-vri-es, 2016-10-04:
I'm not aware of portable methods to print exception details apart from the message of classes inheriting from `std::exception`. It is possible to make it runtime configured (by optionally re-throwing the caught exception), but I'm not convinced that is worth the effort.

I'm favour of not blindly catching all exceptions. I think safety is always the first concern, and in that light I think that crashing a process is preferable over continuing with a possibly invalid or unexpected state. It depends on the type of error of course how dangerous this is (and arguably it \*should\* never be dangerous), but I'd stick on the safe side.

If we do want to catch all exceptions then there should atleast be a specific `catch (std::exception const & e)` statement that prints the `e.what()`.

bmagyar, 2016-11-18:
I've read the code a bit and this is my proposed solution:

A recurring pattern is this (referring to the action_name is not everywhere though)
```   
try
{
...
}
catch (std::runtime_error &ex)
{
    logError("Exception caught while processing action '%s': %s", action_name.c_str(), ex.what());
}
catch (...)
{
     logError("Exception caught while processing action '%s'", action_name.c_str());
}
```

I feel that we agree on removing all instances of `catch(...)` and the discussion shifted towards how to handle the rest in the best way.

My proposed solution for the pattern above is this:
```   
try
{
...
}
catch (std::exception &ex) // since std::runtime_exception < std::exception
{
    logError("Exception caught while processing action '%s': %s", action_name.c_str(), ex.what());
}
```
This provides reasonable prints of things that can be printed and fails (hopefully with a stacktrace) on those that cannot.

v4hn, 2016-11-20:
On Fri, Nov 18, 2016 at 04:11:55AM -0800, Bence Magyar wrote:
> My proposed solution for the pattern above is this:
> ```
> try
> {
> ...
> }
> catch (std::exception &ex) // since std::runtime_exception < std::exception
> {
>     logError("Exception caught while processing [...]

Looks good to me. Feel free to open the respective pull-request.

v4hn, 2017-01-13:
ping @bmagyar ?

bmagyar, 2017-01-13:
I think I have these changes on a branch locally, will try to get a PR
together next week. Sorry for disappearing on this one.

bmagyar, 2017-03-01:
Done!

closed 2017-03-01 by bmagyar