moveit/moveit #272: Indigo-devel branch compilation errors in Trusty

lakehanne, 2016-10-04:
Is this branch stable in Ubuntu 14.04 yet? 

I have it cloned in my workspace src folder but I get errors such as:

```bash
/home/robotec/catkin_ws/src/moveit/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:315:63: error: ‘struct controller_manager_msgs::ControllerState_<std::allocator<void> >’ has no member named ‘resources’
         for (std::vector<std::string>::iterator r = c->second.resources.begin(); r != c->second.resources.end(); ++r)
                                                               ^
/home/robotec/catkin_ws/src/moveit/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:315:97: error: ‘struct controller_manager_msgs::ControllerState_<std::allocator<void> >’ has no member named ‘resources’
         for (std::vector<std::string>::iterator r = c->second.resources.begin(); r != c->second.resources.end(); ++r)
```

lakehanne, 2016-10-05:
I changed to the 0.8.3 tag and it seems to be breaking when it's looking for `warehouse_ros/database_connection.h` . The full message:

```bash
In file included from /home/robotec/catkin_ws/src/moveit/moveit_ros/warehouse/warehouse/include/moveit/warehouse/constraints_storage.h:40:0,
                 from /home/robotec/catkin_ws/src/moveit/moveit_ros/warehouse/warehouse/src/constraints_storage.cpp:37:
/home/robotec/catkin_ws/src/moveit/moveit_ros/warehouse/warehouse/include/moveit/warehouse/moveit_message_storage.h:40:47: fatal error: warehouse_ros/database_connection.h: No such file or directory
 #include <warehouse_ros/database_connection.h>
```

I searched this repo and the `database_connection.h` file seem to be missing.

Do you have any idea where this is?

lakehanne, 2016-10-05:
Okay, progress here. I cloned  the jade-devel branch of [warehouse_ros](https://github.com/ros-planning/warehouse_ros/tree/jade-devel) and re-catkin'd my src folder. I cloned the jade-devel branch since indigo-devel does not have the necessary include file. And it seemed to work. Another thing to mote, using the jade-devel branch of moveit_msgs also helps with an error free compilation.

 Don't know how stable my cooked up workspace would be though but I thought to leave comments here that might help anyone later on.

closed 2016-10-05 by lakehanne

davetcoleman, 2016-10-05:
Did you follow these instructions: http://moveit.ros.org/install/source/ ? They use wstool to download all dependent repos.

lakehanne, 2016-10-05:
I obviously missed that. Would have saved myself a lot of trouble.