moveit/moveit #274: OMPL segfaults for RRTstar

rbbg, 2016-10-09:
Not sure if this is a MoveIt! or OMPL issue but I figured I'd try here first.

I was working on exposing some new OMPL planners and noticed that RRT* is segfaulting on kinetic-devel. Some newer RRT related optimization based planners (BIT* and InformedRRT*) suffered the same issue. To reproduce:

- launch moveit_resources demo.launch
- select RRTstarkConfigDefault
- plan

a gdb backtrace points to the segfault coming from RRTstar::setup(), but I can't find any obvious faults or differences with the setup() for any other planner.

I'd be happy to investigate/fix but perhaps someone has an idea what's going on. With the recent changes to std::shared_ptr it's a little tricky rolling back to find out when it stopped working, so I haven't done so. Could this be related to that change? Any ideas on what's going on?

v4hn, 2016-10-10:
On Sun, Oct 09, 2016 at 08:26:21AM -0700, Ruben Burger wrote:
> I'd be happy to investigate/fix

I noticed that segfault on my system too, but did not get around to analyze it.
Please go ahead!

> With the recent changes to std::shared_ptr it's a little tricky rolling back to find out when it stopped working, so I haven't done so.
> Could this be related to that change?

I believe I noticed the problem in the indigo branch already, so it should not be related to the shared_ptr changes.

rbbg, 2016-10-10:
I tried running RRT* in the indigo-source docker image and I didn't encounter it, so I figured it was kinetic specific.

davetcoleman, 2016-10-11:
gdb backtrace?

rbbg, 2016-10-11:
complete bt:
```
[ INFO] [1476199029.209254304]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[New Thread 0x7fffb5aa7700 (LWP 1948)]
[ INFO] [1476199029.218402631]: No optimization objective specified, defaulting to PathLengthOptimizationObjective
[ INFO] [1476199029.218623957]: Planner configuration 'manipulator[RRTstarkConfigDefault]' will use planner 'geometric::RRTstar'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1476199029.219212841]: manipulator[RRTstarkConfigDefault]: problem definition is not set, deferring setup completion...
[Thread 0x7fffb5aa7700 (LWP 1948) exited]
Thread 13 "move_group" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb77fe700 (LWP 779)]
0x00007fffc9c490fa in ompl::geometric::RRTstar::setup() ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
```
```
(gdb) bt
#0  0x00007fffc9c490fa in ompl::geometric::RRTstar::setup() ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#1  0x00007fffca148998 in std::shared_ptr<ompl::base::Planner> (anonymous namespace)::allocatePlanner<ompl::geometric::RRTstar>(std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&) () from /root/ws_moveit/install/lib/libmoveit_ompl_interface.so
#2  0x00007fffca149326 in boost::detail::function::function_obj_invoker3<boost::_bi::bind_t<std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::Planner> (*)(std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&), boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::arg<3> > >, std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&>::invoke(boost::detail::function::function_buffer&, std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&) ()
   from /root/ws_moveit/install/lib/libmoveit_ompl_interface.so
#3  0x00007fffca1700c0 in std::_Function_handler<std::shared_ptr<ompl::base::Planner> (std::shared_ptr<ompl::base::SpaceInformation> const&), boost::_bi::bind_t<boost::_bi::unspecified, boost::function<std::shared_ptr<ompl::base::Planner> (std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&)>, boost::_bi::list3<boost::arg<1>, boost::_bi::value<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, boost::_bi::value<o---Type <return> to continue, or q <return> to quit---
mpl_interface::ModelBasedPlanningContextSpecification> > > >::_M_invoke(std::_Any_data const&, std::shared_ptr<ompl::base::SpaceInformation> const&) ()
   from /root/ws_moveit/install/lib/libmoveit_ompl_interface.so
#4  0x00007fffc9d41378 in ompl::geometric::SimpleSetup::setup() ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#5  0x00007fffca16b749 in ompl_interface::ModelBasedPlanningContext::configure() ()
   from /root/ws_moveit/install/lib/libmoveit_ompl_interface.so
#6  0x00007fffca14826b in ompl_interface::PlanningContextManager::getPlanningContext(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&) const ()
   from /root/ws_moveit/install/lib/libmoveit_ompl_interface.so
#7  0x00007fffca137b38 in ompl_interface::OMPLInterface::getPlanningContext(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&) const ()
   from /root/ws_moveit/install/lib/libmoveit_ompl_interface.so
#8  0x00007fffca3ccc47 in ompl_interface::OMPLPlannerManager::getPlanningContext(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&) const ()
#10 0x00007fffc902af25 in default_planner_request_adapters::FixStartStatePathConstraints::adaptAndPlan(boost::function<bool (std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&)> const&, std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::vector<unsigned long, std::allocator<unsigned long> >&) const ()
   from /root/ws_moveit/install/lib/libmoveit_default_planning_request_adapter_plugins.so
#11 0x00007ffff078592f in planning_request_adapter::PlanningRequestAdapter::adaptAndPlan(std::shared_ptr<planning_interface::PlannerManager> const&, std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::vector<unsigned long, std:: [... cut ...]
   from /root/ws_moveit/install/lib/libmoveit_move_group_default_capabilities.so
#26 0x00007fffc8b3af59 in actionlib::SimpleActionServer<moveit_msgs::MoveGroupAction_<std::allocator<void> > >::executeLoop() ()
   from /root/ws_moveit/install/lib/libmoveit_move_group_default_capabilities.so
#27 0x00007ffff231f5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#28 0x00007ffff621b6fa in start_thread (arg=0x7fffb77fe700) at pthread_create.c:333
#29 0x00007ffff57b4b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

davetcoleman, 2016-10-11:
The backtrace would be more helpful if you compiled your catkin workspace in debug mode

rbbg, 2016-10-12:
I already did so earlier but couldn't really find out more and unfortunately didn't store the backtrace... Anyhow, I just did it again and the complete (huge) backtrace is below in case I missed something. I think the RTTI symbol errors are caused by gdb?

I should probably compile OMPL with debug symbols as well? Haven't gotten round to it.

```
[ INFO] [1476251170.115777553]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[New Thread 0x7fffb2ffd700 (LWP 10313)]
[ INFO] [1476251170.132889348]: No optimization objective specified, defaulting to PathLengthOptimizationObjective
[ INFO] [1476251170.133589272]: Planner configuration 'manipulator[RRTstarkConfigDefault]' will use planner 'geometric::RRTstar'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1476251170.135600744]: manipulator[RRTstarkConfigDefault]: problem definition is not set, deferring setup completion...
[Thread 0x7fffb2ffd700 (LWP 10313) exited]

Thread 13 "move_group" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb9668700 (LWP 10029)]
0x00007fffbbb710fa in ompl::geometric::RRTstar::setup() ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
(gdb) bt
#0  0x00007fffbbb710fa in ompl::geometric::RRTstar::setup() ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#1  0x00007fffd8742cca in (anonymous namespace)::allocatePlanner<ompl::geometric::RRTstar> (si=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<ompl::base::SpaceInformation*, (__gnu_cxx::_Lock_policy)2>'
warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<ompl::base::SpaceInformation*, (__gnu_cxx::_Lock_policy)2>'

std::shared_ptr (count 7, weak 1) 0x7fffa8005720, new_name="manipulator[RRTstarkConfigDefault]", 
    spec=...)
    at /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp:126
#2  0x00007fffd8758e74 in boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::arg<3> >::operator()<std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::Planner> (*)(std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&), boost::_bi::list3<std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&> > (this=0x7fffa800e540, 
    f=@0x7fffa800e538: 0x7fffd8742c01 <(anonymous namespace)::allocatePlanner<ompl::geometric::RRTstar>(ompl::base::SpaceInformationPtr const&, std::__cxx11::string const&, ompl_interface::ModelBasedPlanningContextSpecification const&)>, a=...) at /usr/include/boost/bind/bind.hpp:382
#3  0x00007fffd875650b in boost::_bi::bind_t<std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::Planner> (*)(std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&), boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::arg<3> > >::operator()<std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecifica---Type <return> to continue, or q <return> to quit---
tion const&> (this=0x7fffa800e538, a1=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<ompl::base::SpaceInformation*, (__gnu_cxx::_Lock_policy)2>'
warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<ompl::base::SpaceInformation*, (__gnu_cxx::_Lock_policy)2>'
std::shared_ptr (count 7, weak 1) 0x7fffa8005720, 
    a2="manipulator[RRTstarkConfigDefault]", a3=...) at /usr/include/boost/bind/bind.hpp:934
#4  0x00007fffd87536ba in boost::detail::function::function_obj_invoker3<boost::_bi::bind_t<std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::Planner> (*)(std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&), boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::arg<3> > >, std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&>::invoke (
    function_obj_ptr=..., a0=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<ompl::base::SpaceInformation*, (__gnu_cxx::_Lock_policy)2>'
warning: RTTI symbol not found for class 'std::_Sp_counted_ptr<ompl::base::SpaceInformation*, (__gnu_cxx::_Lock_policy)2>'
std::shared_ptr (count 7, weak 1) 0x7fffa8005720, 
    a1="manipulator[RRTstarkConfigDefault]", a2=...)
    at /usr/include/boost/function/function_template.hpp:138
#5  0x00007fffd87a88c8 in boost::function3<std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&>::operator() (
[... cut ...]
#64 0x00007fffba90e728 in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf0<void, actionlib::SimpleActionServer<moveit_msgs::MoveGroupAction_<std::allocator<void> > > >, boost::_bi::list1<boost::_bi::value<actionlib::SimpleActionServer<moveit_msgs::MoveGroupAction_<std::allocator<void> > >*> > > >::run (this=0xe4aeb0) at /usr/include/boost/thread/detail/thread.hpp:116
#65 0x00007ffff14005d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#66 0x00007ffff5edc6fa in start_thread (arg=0x7fffb9668700) at pthread_create.c:333
#67 0x00007ffff5475b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

```

rbbg, 2016-10-12:
Hm. Perhaps it is more OMPL than MoveIt! related: with the latest ompl source, RRT* works again. From what I can tell, the latest binary ompl release is not that long ago ([10 days?](https://github.com/ros-gbp/ompl-release/tree/debian/kinetic/xenial/ompl)). But there have been some changes in RRTstar since then ([this commit](https://github.com/ompl/ompl/commit/75db3f1f031461cea940896f5fb05786bbae9cc9)) so that might be related.

rbbg, 2016-11-01:
I have just tried to use RRT* with the latest kinetic-release docker image and it is still segfaulting, @mamoll perhaps you have an idea on what's going on?

davetcoleman, 2017-09-28:
@SalvoVirga reports having found the fix for this!

anirudh-shekhawat, 2017-10-09:
@davetcoleman ,@SalvoVirga. Can you update on the current state and share the fix?

v4hn, 2017-10-09:
@davetcoleman: @SalvoVirga didn't report "the fix".
He casually mentioned he has a patch for ompl that makes the problem go away for him. 😃 

@SalvoVirga: Could you please pull-request your patch as-is at https://github.com/ompl/ompl .
If it breaks anything @davetcoleman , @mamoll, me, and whoever else is interested in this can look into a cleaner solution/fixup.

davetcoleman, 2017-10-18:
@v4hn: @SalvoVirga he did say he found the fix, but he has not shared it, correct ;-)

jonbinney, 2017-10-18:
I was able to reproduce this bug using the released version of ompl in kinetic just now, but not able to reproduce it using ompl from source directly from the ompl repo. Investigating further now.

SalvoVirga, 2017-10-18:
I was and I’m still overwhelmed by other duties guys, I will be able to share everything probably in a couple of weeks. If someone feels like working on it, I remember using valgrind helped a lot.

jonbinney, 2017-10-18:
Just confirmed that I get this segfault when compiling the ros version (1.2.1) from source, but not when compiling from the current ompl source. So it seems like this may have been fixed already in OMPL.

jonbinney, 2017-10-18:
I just confirmed that if I take the version released in ROS (1.2.1) and then cherry pick just this commit https://github.com/ompl/ompl/commit/5efd0fa2e6dc0674f3bf0c1a6787920c546919a2 it fixes the bug. @davetcoleman @mamoll if we can release just that into kinetic it should fix this.

jonbinney, 2017-10-18:
I made a branch starting at the 1.2.1 tag that (the one released into ros) and then cherry-picking the fix commit for this issue: https://github.com/jonbinney/ompl/tree/jb-cherry-pick-rrt-star-fix

@davetcoleman can you force-push that to be the kinetic-devel branch for ompl? That can then be released into kinetic.

davetcoleman, 2017-10-18:
Thanks @jonbinney - we now have a custom [OMPL kinetic branch](https://github.com/ompl/ompl/commits/kinetic-devel). 

@mamoll can you re-release ompl for ROS Kinetic (and ROS Lunar) using the ``kinetic-devel`` branch via bloom? You will need to edit the bloom config for OMPL to switch branches from master.

mamoll, 2017-10-19:
The git->hg sync didn't quite work, so I had to redo it. There is now a kinetic-devel branch on bitbucket and GitHub. I don't know how to re-release ompl for kinetic. I don't understand why this also has to be done for lunar, since ompl for lunar already contains this fix.

davetcoleman, 2017-10-19:
> since ompl for lunar already contains this fix.

I hadn't realized that. We are using the same code for kinetic as lunar and I just assumed this had been done for OMPL also. In this case no, it does not need to be released for lunar.

> I don't know how to re-release ompl for kinetic

@wjwwood will actually take care of this while at the same time addressing https://github.com/ros-planning/moveit/issues/169

wjwwood, 2017-10-19:
I still need @mamoll (or someone) to make a new release based on the `kinetic-devel` branch.

Since the branch you linked to @davetcoleman is 1.2.1 with one extra commit, I guess it would be 1.2.2.

davetcoleman, 2017-10-20:
@mamoll you just need to follow [these steps](http://wiki.ros.org/bloom/Tutorials/ReleaseCatkinPackage) but for the bloom step add the ``--edit`` flag so that you can change the branch to ``kinetic-devel`` from ``master``.

wjwwood, 2017-10-20:
@davetcoleman those directions are for a catkin package, I don't believe all of it will apply for ompl... Changing the bloom branch is not the only thing that needs to happen (as I said before there needs to be a new 1.2.2 release based on the `kinetic-devel` branch).

v4hn, 2017-10-20:
*sigh*

So what are the next steps to do and who has the permissions to do them?
It seems everyone is pointing somewhere else at the moment.

wjwwood, 2017-10-20:
I don’t know how to be any more clear, and I don’t think anyone is pointing fingers — I’m happy to help with bloom and ros build farm part I just can’t make new ompl releases.

So I need someone with permission to make ompl releases to create a new version based on the kinetic development branch that @davetcoleman created, the natural next version would be 1.2.2, since kinetic currently has 1.2.1 and lunar has 1.3.x. 

After that I will make a new bloom release and add any patches required. If @mamoll decides to disable the symlinks in 1.2.2 release changes what I do in those patches but I don’t mind either way if it goes in there or in via bloom. 

If @mamoll would like to handle the bloom release as well I’ll gladly give any instruction or help needed but the tutorial link @davetcoleman gave contains a lot of stuff specific to catkin packages that won’t apply to ompl, especially the stuff about generating changelogs and using `catkin_prepare_release`.

mamoll, 2017-10-20:
I have tagged the 1.2.2 release of ompl on bitbucket and github. It includes only the fix for RRT\*. Disabling the symlinks will be done on the master branch and will be included in future releases of ompl. @wjwwood, it's probably sufficient to just comment out this line:
https://github.com/ompl/ompl/blob/c2ffc537dcc79bce389743c46abb155cce914432/CMakeLists.txt#L184

mamoll, 2017-10-20:
@wjwwood, I'd be happy to do the bloom releases in the future. I have kept some emails that helped me create the 1.3 release for lunar. It sounds like creating this kinetic release is non-standard (or at least not as easy), so I'd very much appreciate it if you could do this release,

wjwwood, 2017-10-20:
@mamoll thanks for making the release.

I'll take care of this release, future releases will be (hopefully) less eventful. 👍

nwotero, 2017-12-15:
Is there any estimate for when ompl updated with this fix will be available through debs?

enrico-g, 2018-05-17:
I encountered the same issue (segfaults for RRT_star). My OMPL version turned out to be old, therefore I solved the problem by simply removing and reinstalling it (NOT from source!).

zkingston, 2018-05-21:
As @enrico-g suggests, the latest version of OMPL for kinetic (version `1.2.3`) from the package manager should have the fix that was pointed out by @jonbinney (https://github.com/ompl/ompl/commit/5efd0fa2e6dc0674f3bf0c1a6787920c546919a2, you can see the commit in the release here: https://github.com/ros-gbp/ompl-release/commit/479ec3e89e588c7198edaca1825c1dc83965b3a8#diff-57f4512e132fa498410cefd0c08d5545). It should be enough to just upgrade your OMPL (`apt-get update ros-kinetic-ompl`).

closed 2019-06-20 by BryceStevenWilley