moveit/moveit #275: Build error building Kinetic from source

zplizzi, 2016-10-10:
I've been trying to build the Kinetic beta from source, as in the instructions on the source install page. I encountered a variety of missing package errors, which have been resolved by grabbing various packages and installing them, but am stuck on this error:

```
Scanning dependencies of target moveit_ros_control_interface_trajectory_plugin
[ 25%] Building CXX object CMakeFiles/moveit_ros_control_interface_trajectory_plugin.dir/src/joint_trajectory_controller_plugin.cpp.o
Scanning dependencies of target moveit_ros_control_interface_plugin
[ 50%] Building CXX object CMakeFiles/moveit_ros_control_interface_plugin.dir/src/controller_manager_plugin.cpp.o
/home/zplizzi/projects/bio-robot/ros_ws/src/moveit/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp: In member function ‘void moveit_ros_control_interface::MoveItControllerManager::allocate(const string&, const ControllerState&)’:
/home/zplizzi/projects/bio-robot/ros_ws/src/moveit/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:155:68: error: ‘class pluginlib::ClassLoader<moveit_ros_control_interface::ControllerHandleAllocator>’ has no member named ‘createUniqueInstance’
         alloc_it = allocators_.insert(std::make_pair(type, loader_.createUniqueInstance(type))).first;
                                                                    ^
CMakeFiles/moveit_ros_control_interface_plugin.dir/build.make:62: recipe for target 'CMakeFiles/moveit_ros_control_interface_plugin.dir/src/controller_manager_plugin.cpp.o' failed
CMakeFiles/Makefile2:808: recipe for target 'CMakeFiles/moveit_ros_control_interface_plugin.dir/all' failed
make[2]: *** [CMakeFiles/moveit_ros_control_interface_plugin.dir/src/controller_manager_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/moveit_ros_control_interface_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 75%] Linking CXX shared library /home/zplizzi/projects/bio-robot/ros_ws/devel/.private/moveit_ros_control_interface/lib/libmoveit_ros_control_interface_trajectory_plugin.so
[ 75%] Built target moveit_ros_control_interface_trajectory_plugin
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

I assume this is due to something I did wrong in the setup/config steps, but can't figure out what. The version of moveit is the current head of `kinetic-devel` as of a couple hours ago. 

Very much looking forward to this being released officially for Kinetic - was disappointed to find out that, on my Ubuntu 16.04 computer, there is no configuration of ROS and MoveIt versions that is supported without resorting to Docker or beta builds from source.

de-vri-es, 2016-10-10:
Until the relevant packages are synced, you will also need to install some dependencies from source. This specific error is because moveit for Kinetic requires pluginlib 1.10.4 and class_loader 0.3.5 which are released but not synced yet.

You can find full instructions for compiling from source for kinetic here:
http://moveit.ros.org/install/source/#kinetic

rbbg, 2016-10-10:
Hi zplizzi,

There have been a couple of things that require some changes from MoveIt! dependencies. Check [this](https://groups.google.com/forum/#!topic/moveit-users/yMHt1QMAwuM) announcement on the MoveIt! users group for more info. You can get the required updates from ros-shadow-fixed or build them from source (I think you will need pluginlib, actionlib and geometric_shapes). The error you are currently seeing comes from pluginlib.

v4hn, 2016-10-10:
Hey Zack,

On Sun, Oct 09, 2016 at 08:14:40PM -0700, Zack Polizzi wrote:
> /home/zplizzi/projects/bio-robot/ros_ws/src/moveit/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:155:68: error: ‘class [...]

The current `kinetic-devel` branch depends on the pluginlib and class_loader versions currently in shadow-fixed because of this feature: https://github.com/ros/pluginlib/pull/42
Either you use shadow-fixed or you build them yourself.

> Very much looking forward to this being released officially for Kinetic - was disappointed to find out that, on my Ubuntu 16.04 computer, there is no configuration of ROS and MoveIt versions that is supported without resorting to Docker or beta builds from source.

Working on it.
I suppose, the one thing you would find more disappointing than finding no official release is a broken release. :)

zplizzi, 2016-10-11:
Awesome, it's building properly now after installing pluginlib and class_loader from shadow-fixed. Thanks for the quick help!

closed 2016-10-11 by zplizzi

davetcoleman, 2016-10-11:
@zplizzi I've put a lot of effort into making those source install instructions complete...

>  I encountered a variety of missing package errors, which have been resolved by grabbing various packages and installing them

This should not be the case if you used ``rosdep install`` as documented in the instructions

> error: ‘class pluginlib::ClassLoader`

I think what needs to be added is to run apt-get update and apt-get dist-upgrade after switching to shadow fixed (I had just assumed this step)... was anything else missing?

davetcoleman, 2016-10-11:
Actually the sync already happened for pluginlib and shadow-fixed is no longer required, I think you just needed to download the latest version through apt-get. Updated instructions: https://github.com/ros-planning/moveit.ros.org/pull/89

zplizzi, 2016-10-11:
@davetcoleman I had attempted to follow your instructions, but I think I must have simply failed at copy-pasting the commands into a terminal (awkward) - since, in checking the file `/etc/apt/sources.list.d/ros-latest.list` after the fact, it didn't contain a reference to the shadow repository at all (which is clearly a step in your instructions). 

I think you can just chalk this one up to a careless user, sorry! Again, many thanks for the support.