moveit/moveit #276: Planning with full joint range

achim-k, 2016-10-10:
I'm using MoveIt with a UR10 and I have trouble with the UR's joint limits.

From <https://github.com/ros-industrial/universal_robot>:
> NOTE:
As MoveIt! seems to have difficulties with finding plans for the UR with full joint limits [-2pi, 2pi], there is a joint_limited version using joint limits restricted to [-pi,pi]. In order to use this joint limited version, simply use the launch file arguments 'limited'

First of all, why is that? I haven't found an explanation anywhere.

But I can confirm, when using unlimited joint values, MoveIt has problems planning to pose goals:
```
[ INFO] [/move_group]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [/move_group]: Solution found in 0.062463 seconds
[ INFO] [/move_group]: SimpleSetup: Path simplification took 0.013770 seconds and changed from 4 to 17 states
[ERROR] [/move_group]: Computed path is not valid. Invalid states at index locations: [ 34 35 ] out of 50. Explanations follow in command line. Contacts are published on /move_group/display_contacts
[ INFO] [/move_group]: Found a contact between 'shoulder_link' (type 'Robot link') and 'wrist_3_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
...
```
However, when the robot's end effector is already close to the goal, it works fine. Also joint-space goals seem to work without problems. So it seems to be an issue related to the IK solution?

For my specific application I have troubles with the limited joint limits: When for instance the robot's shoulder joint (*shoulder_pan_joint*) is close to PI and I want it to move e.g. +20deg more, it can't move over the upper joint limit (PI) and instead turns the whole shoulder by almost a full turn to roughly -PI+20deg.

What I did now to circumvent this issue was to write a custom move group plugin that offers a service to set joint limits dynamically (overwriting the values stored in the RobotModel instance). However, I also had to patch the IK plugin (KDL) and the moveit_planners/ompl package with the same service to let MoveIt generate paths within the previously set joint limits.

This approach is rather hacky but works quite well for my case. When my shoulder joint is close to PI and should move over the upper limit, I set the new joint limits for this joint to [0, 2*PI]. The main disadvantage is that the correct joint limits have to be set depending on which site the robot is going to operate.

To summarize:
- Why does MoveIt have problems with the full joint range? I guess this is not only the case for the UR?
- Is there a better way than my hacky SetJointLimit service to make the robot able to utilize its whole joint range?

v4hn, 2016-10-10:
We noticed that problem over here too and it's on my long-term todo list to look into it.
At the moment I have no idea why this poses a problem, but sadly it's not too easy to debug.
It is a really hacky workaround to restrict the joints to [0;2pi]. Your personal workaround is quite nifty
and it's interesting to hear it actually works.

I am sure many users would appreciate a fix for this bug or a detailed analysis!

scottpaulin, 2016-10-11:
Hi @achim-k,

>>Why does MoveIt have problems with the full joint range? I guess this is not only the case for the UR?

This is (probably) because of an unavoidable self-collision. I have noticed this with the UR5. See the discussion [here](https://github.com/ros-industrial/universal_robot/issues/265)

>>Is there a better way than my hacky SetJointLimit service to make the robot able to utilize its whole joint range?

Yes, I am currently writing a short paper addressing this from the motion planning side. It should be on ArXiv soon. If you want a solution now email me (scott[underscore]paulin[at]hotmail.com) and I should be able to get you going if you know how to use OMPL.

achim-k, 2016-10-11:
Hi @scottpaulin, 

> Yes, I am currently writing a short paper addressing this from the motion planning side. It should be on ArXiv soon. If you want a solution now email me (scott[underscore]paulin[at]hotmail.com) and I should be able to get you going if you know how to use OMPL.

It's not that urgent, I can wait for the paper :) Thanks a lot for figuring that out!

davetcoleman, 2016-10-11:
+1 to tracking down the offending components in moveit and proposing a PR to fix the limited joint range bug - i've also run into this and never looked deeply into it. i wouldn't be surprised if it was a kinematics limitation - there are bits of code that "wrap" joints around preventing leaving the range of -pi,pi

davetcoleman, 2016-10-11:
This would also be a great subject to hear from @isucan

v4hn, 2016-10-12:
@davetcoleman According to scott's analysis, this is pretty much a bug in
the UR's urdf: https://github.com/ros-industrial/universal_robot/issues/265

MoveIt cannot cope with disjoint configuration spaces out of the box and there's no easy general fix for this within the framework.
We *could* add checks for disjoint configuration spaces to the setup assistant and reduce the limits accordingly, but that requires quite a bit of insight and workload.
I'm interested in what scott has in mind with this paper. :-)

gavanderhoorn, 2016-10-12:
Pedantic maybe, but can we please stop calling the 2pi range for the elbow a *bug*? The urdf is made based on the specs released by the mfg. Whether those specs make sense is something else, but if the urdf's task is to accurately model the robot, I don't see how having +-2pi for the elbow could be considered a bug.

v4hn, 2016-10-12:
@gavanderhoorn Ok, it's not a bug. It's an unfortunate combination of a missing feature (disjoint configuration spaces) in the Inverse-Kinematics plugins (although it stretches the definition of IK quite a lot to say you need a *reachable* valid joint configuration..) and the attempt to model the robot accurately, even if the details are not actually helpful. :)

I'd still say this is most easy to fix in the urdf description, although support for detecting disjoint configuration spaces would be nice to have in MoveIt.

scottpaulin, 2016-10-13:
> Pedantic maybe, but can we please stop calling the 2pi range for the elbow a bug? The urdf is made based on the specs released by the mfg. Whether those specs make sense is something else, but if [...]

Sorry @gavanderhoorn, I am guilty of using the word 'bug' loosely. You're right, the problem is not with this package, it just happens that the easiest workaround is probably to change something in this package. The cause of the issue is possibly on Universal Robot's end for not communicating that the elbow joint can't actually use the full [-2pi, 2pi) range e.g. in their specs as you mentioned. If it is, then it was probably just an honest mistake by somebody along the line.  There is also a small chance that they are not aware that the full range of the elbow joint can't be used. If the upper arm link was a bit longer the elbow joint would be able to operate in the full range.

I don't have a physical UR5 near by at the moment (I am off campus) but I can see the self-collisions on the simulator they provide, so I don't think that this could be an issue with the collision meshes.

The default value of 0.05 for the 'longest valid segment fraction' that @davetcoleman mentioned works well when the joint limited version of the robot is being used (because it has a much smaller state space). It was probably set with the limited version of the robot in mind since most people seem to have been using that.

v4hn, 2019-08-28:
The problem was circumvented by changing the joint limit of the elbow joints for the UR5.

In general, the problem of having disjoint viable subspaces in the specified planning space still exists and MoveIt does not provide a mechanism to detect them.
Online detection of this condition does not seem tractable and if you know you are looking for such a problem (via special offline analysis of the space), you likely already know what you need to change to solve it too.

Closing as no-fix until someone can provide an efficient way to detect it.

closed 2019-08-28 by v4hn