moveit/moveit #277: how do I use compute_cartesian_path function?

haixuanwo, 2016-10-11:
when I use compute_cartesian_path , I often get erorr.
the value of fraction is 89%,32%,50% etc.
I try to give long time and more tempt to  IK . It is not good, I will get strange trajectory.
I get waypoints in RVIZ using commander of "current". I doubt that I make mistake in here.

I want know how can I get correct waypoints?and how do I use compute_cartesian_path function?

haixuanwo, 2016-10-11:
I have other question whether the compute_cartesian_path is used in two arms.

v4hn, 2016-10-11:
This belongs on ROS answers and is not a direct issue of the framework.
You cross-posted at http://answers.ros.org/question/245476/how-do-i-use-compute_cartesian_path-function/

closed 2016-10-11 by v4hn