moveit/moveit #278: Meet problem when using catkin build to build moveit_ros_planning package

cangjiaxaun, 2016-10-11:
Hi, all
I'm using the kinetic ros ditribution on Ubuntu 16.04.1 and want to install moveit!, but I meet the following problem(erlilyu is my user name and ros_ws is my workspace name)
```
Errors     << moveit_ros_control_interface:make /home/erlilyu/ros_ws/logs_cb/moveit_ros_control_interface/build.make.005.log
/home/erlilyu/ros_ws/src/moveit/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp: In member function ‘void moveit_ros_control_interface::MoveItControllerManager::allocate(const string&, const ControllerState&)’:
/home/erlilyu/ros_ws/src/moveit/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp:155:68: error: ‘class pluginlib::ClassLoader<moveit_ros_control_interface::ControllerHandleAllocator>’ has no member named ‘createUniqueInstance’
```

I notice that in the recent update pluginlib has add Add "ClassLoader::createUniqueInstance", and I clone that package from github and use catkin build to build it. But I still have this error message. I think this error may caused by using the binary installed package instead of the catkin installed one.
But I have this error message as well 

```
Errors     << moveit_ros_planning:make /home/erlilyu/ros_ws/logs_cb/moveit_ros_planning/build.make.004.log
In file included from /home/erlilyu/ros_ws/src/moveit/moveit_ros/planning/planning_request_adapter_plugins/src/list.cpp:37:0:
/home/erlilyu/ros_ws/src/pluginlib/include/pluginlib/class_loader.h:51:48: error: ‘UniquePtr’ in ‘class class_loader::ClassLoader’ does not name a template type
   using UniquePtr = class_loader::ClassLoader::UniquePtr<T>;
```

This seems to indicate that the catkin system knows where to find the source installed package, so why it have the earlier error?

Thanks in advance.

v4hn, 2016-10-11:
You just need to clone the `class_loader` package too...

davetcoleman, 2016-10-11:
Or run apt-get update / apt-get dist-upgrade

I've just clarified this on the website: https://github.com/ros-planning/moveit.ros.org/pull/89

closed 2016-10-11 by davetcoleman