moveit/moveit #282: kinetic without manipulation_msgs dependency

v4hn, 2016-10-12:
@davetcoleman [asked](https://github.com/ros-planning/moveit/issues/18#issuecomment-253003530) for a new issue to discuss this matter.
In https://github.com/ros-interactive-manipulation/manipulation_msgs/issues/9 everyone agreed to move away from depending on `manipulation_msgs`. At the moment, MoveIt! depends only on `manipulation_msgs/GraspPlanning.srv` in [this function](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp#L417).

Because we recently merged #264 and thus have a replacement for calling an external grasp planning service, the old code could either be removed entirely (I'd prefer that) or could be changed to [the new `moveit_msgs/GraspPlanning.srv`](https://github.com/ros-planning/moveit_msgs/pull/32).

I suppose it would be a good idea to remove the outdated dependency right from the first kinetic release.

Cross-reference: https://github.com/ros-planning/moveit/issues/18#issuecomment-252477757 and following

davetcoleman, 2016-10-12:
+1 to removing the code entirely in kinetic, but not backporting. i like the idea of reducing the number of msgs dependencies.

rhaschke, 2016-10-14:
+1 Makes sense to me.

130s, 2016-10-16:
+1

I assume we want this to be done by Kinetic initial release #18?

davetcoleman, 2016-10-16:
We worked on getting manipulation_msgs released for kinetic this past summer, so there is no problem if we release the first version still depending on it and later remove it. Depends how fast @v4hn wants to implement this change, but we must avoid adding more blockers.

v4hn, 2016-11-10:
fixed by #328

closed 2016-11-10 by v4hn