moveit/moveit #288: Allow setting the color of individual collision objects

simonschmeisser, 2016-10-17:
We have several classes of collision objects, ie moveable boxes that contain smaller pieces that come from object detection and will possibly be grasped by the robot. It would be nice to give these different colors. Also we want to highlight (by changing color) one of the objects if it is selected in a list in our UI

I/we will prepare some pull-request at some time soonish, just wanted to write this down and know if maybe somebody has already implemented it.

davetcoleman, 2016-10-17:
I use collision objects with many different colors already, you can see how I do so in [moveit_visual_tools](https://github.com/davetcoleman/moveit_visual_tools/blob/kinetic-devel/include/moveit_visual_tools/moveit_visual_tools.h#L316)

I highly recommend this package - it has lots of tools for visualizing stuff in moveit, as well as adding collision objects!

v4hn, 2016-10-17:
@davetcoleman I feel like you use hell of a lot of third-party utilities with MoveIt of which I have never heard.
How about a "Other Helpful Packages For MoveIt!" section on the website?

simonschmeisser, 2016-10-17:
We're using  parts of it already (I think more of RvizVisualTools actually) :)

Looking at the code I found  the  PlanningScene::setObjectColor(id, color) and similar functions which have somehow escaped my attention so far. Thanks!

Having a ros message based interface would probably still be nice. Or does that exist as well?

closed 2016-10-17 by simonschmeisser

davetcoleman, 2016-10-17:
Yea I don't think that exists - sounds like a good addition via PR

@v4hn I'll put it on my queue

CesMak, 2020-02-24:
Hm I am still not sure how to use this exactly. I tried it like this:
```

	shape_msgs::Mesh mesh;
	shapes::ShapeMsg mesh_msg;

	// for different colors:
	moveit_visual_tools::MoveItVisualToolsPtr visual_tools_;

	std::string pack_path = ros::package::getPath("moveit_tutorials");
	for(int i=0; i<part_positions.size(); i++)
	{
		collision_objects[i].id = part_positions[i].name;
		collision_objects[i].header.frame_id = "panda_link0";
		obj_pose.position.x = part_positions[i].x;
		obj_pose.position.y = part_positions[i].y;
		obj_pose.position.z = part_positions[i].z;
		orientation.setRPY( part_positions[i].rx*M_PI / 180, part_positions[i].ry*M_PI / 180, part_positions[i].rz*M_PI / 180);
		obj_pose.orientation = tf2::toMsg(orientation);

		// Add the mesh to the Collision object message
		float factor = part_positions[i].scaling_part;
		Eigen::Vector3d vectorScale(factor, factor, factor);
		shapes::Mesh* m = shapes::createMeshFromResource("file://"+pack_path+"/parts/"+part_positions[i].file_name, vectorScale);

		shapes::constructMsgFromShape(m, mesh_msg);
		mesh = boost::get<shape_msgs::Mesh>(mesh_msg);
		collision_objects[i].meshes.push_back(mesh);
		collision_objects[i].mesh_poses.push_back(obj_pose);

		//TODO for different colors
		visual_tools_->processCollisionObjectMsg(collision_objects[i], rviz_visual_tools::LIME_GREEN);

		collision_objects[i].operation = collision_objects[i].ADD;
	}
```

This does not work however!