moveit/moveit #289: forward c++-11 settings to downstream projects

rhaschke, 2016-10-17:
I frequently run into trouble with projects using MoveIt, because now c++-11 is required by MoveIt headers. IMHO, MoveIt should also export the c++11 definition to downstream projects, instead of requiring developers to add that option manually to their projects.
What do you think?

v4hn, 2016-10-17:
This was discussed at some length when we progressed to `std::shared_ptr`:

https://github.com/ros-planning/geometric_shapes/pull/50#issuecomment-238381018

Eventually, @de-vri-es won the discussion because (1.) There is no way to specify "at least c++11" with gcc and other compilers and this makes it hard to compile modules that depend on MoveIt with more modern standards as c++14. And (2.) current gcc already defaults to c++11, it's just that Ubuntu still ships a version that does not yet do that.

davetcoleman, 2016-10-17:
We had a lot of discussion on doing this kind of thing for geometric_shapes, correct? http://discourse.ros.org/t/std-shared-ptr-in-the-geometric-shapes-public-api/395/10

If I recall, the conclusion was that we would not want to export it.

davetcoleman, 2016-10-17:
^ right. Perhaps we can solve this through documentation somewhere?

rhaschke, 2016-10-17:
OK. I missed that discussion. Making things explicit is nice. However, @de-vri-es actually uses an implicit method to set his flags too (in an extra package). I think we will get a lot of complaints from users when we not export the required flags to compile against MoveIt correctly.

closed 2016-10-17 by rhaschke

de-vri-es, 2016-10-18:
True, but it's not a general purpose library so I do think there is a difference. The only purpose of that package is to set default compiler options at work.

rhaschke, 2016-10-18:
I think, if those compiler options are needed by your package, the package should define them itself. Otherwise the package wouldn't be self-contained. If the options are not required (which I suppose), it's probably an easy solution. However, it obscures the fact in the very same fashion as you complained about for MoveIt ;-)

We just had the issue, that we had to branch a package because of this `std=c++11` option, just because MoveIt requires it now. Otherwise the indigo and kinetic source would be completely identical for this package. I consider this still somewhat nasty.

ipa-mdl, 2016-12-30:
If you don't want to forward the settings, it would be an option to provide a cmake macro (e.g. `moveit_setup_cxx11`) that takes care of the settings in dependent packages with respect to the actual build environment..
Something like https://github.com/ros/robot_model/pull/145/commits/4f9b7f023e80fde2d145dd9127cab97296d43097
Additional CMake magic could test for complilers which use c++14 per default.

v4hn, 2016-12-30:
I don't think adding additional complexity here is a good idea and will result in unforeseen breakage that is *not* as easy to debug as defining the compile standard explicitly.

-1 from my side

ipa-mdl, 2016-12-31:
>  don't think adding additional complexity here is a good idea

IMHO additional complexity was already introduced by switching to c++11 and by not exporting the flags (no judgement here).
Now the downstream user has to figure out that c++11 is required (no safe-guards!) and how to set it for the different cases and release targets.

A macro can be branched for the different ROS distributions and can be deprecated if it is not needed anymore. Documentation (still missing?) is harder to maintain for different versions, especially for migrating existing code.

v4hn, 2016-12-31:
> IMHO additional complexity was already introduced by switching to c++11 and by not exporting the flags (no judgement here).

Yes. My personal opinion is that we should not add *even more* on top of this to shroud it.

> user has to figure out that c++11 is required (no safe-guards!)

Contributions and concrete pointers to improve the current state are always welcome!
If you have proposals, please report them!
imho it is not a reasonable option to safe-guard *every moveit header* and check for C++11 support.

> A macro can be branched for the different ROS distributions and can be deprecated if it is not needed anymore.

Feel free to pull-request a patch. At the moment I do not fully understand your proposed solution
and how it would simplify the user's experience in kinetic.

ipa-mdl, 2016-12-31:
> Yes. My personal opinion is that we should not add even more on top of this to shroud it.

Agreed.

> At the moment I do not fully understand your proposed solution
and how it would simplify the user's experience in kinetic.

My intention was to provide a clean way for setting c++11 support in downstream packages.
There are many ways to do so depending on compiler and CMake version.
c++11 support might be even provided by installing gcc6.
This is no MoveIt-specific issue, but many MoveIt users might be affected.
Perhaps a documentation entry is sufficient.

> Feel free to pull-request a patch.

I can prepare a macro etc. next week, but I do not know the clean way for activating c++11 support yet ;)

davetcoleman, 2016-12-31:
In a different thread I think we settled on [this](https://github.com/davetcoleman/bolt/blob/kinetic-devel/bolt_core/CMakeLists.txt#L5) being the clean way to activate C++11 support. It would be great to mention that in [this tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html) if you don't mind adding that.

ipa-mdl, 2017-01-08:
I have now proposed compile options support to catkin: https://github.com/ros/catkin/pull/844
Usage example: https://github.com/ipa-mdl/catkin/blob/compile-options/test/mock_resources/src/compile_options/set_compile_options/CMakeLists.txt

ipa-mdl, 2017-01-08:
Macro package with optional auto-forward: https://github.com/ipa-mdl/enable_cxx11_support
I plan to release it to indigo, jade and kinetic.

de-vri-es, 2017-01-09:
I'm still opposed to the idea of exporting a C++ standard to packages depending on us. I'd prefer not to rehash the same discussion though.

ipa-mdl, 2017-01-09:
> I'm still opposed to the idea of exporting a C++ standard to packages depending on us.

1. That's why it's optional in the new package, in addition it will not set `-std=c++11` if c++14 is enabled per default
2. Dependent packages have to set the new standard anyway..

> I'd prefer not to rehash the same discussion though.

That's okay, I think I have read most of the discussions..

de-vri-es, 2017-01-09:
Fair enough, though then I wonder if the extra dependency is worth it. I do like the fact that you're testing for the flags rather than just adding them as we currently do.

I'm not sure about not downgrading to C++11 when the compiler already defaults to something compatible. MoveIt has to stick with C++11 for now though, so it seems best to me if all developers use it by default. Otherwise some developers would have to wait for CI to fail where the local compiler could have errored out on accidental use of C++14 features already.

ipa-mdl, 2017-01-09:
>  though then I wonder if the extra dependency is worth it

An extra package allows reuse..
In addition it can be tested explicitly for compiler support (for now gcc4.7 to 6 and clang 3.3 to 4.0)
GCC6 was tricky in particular as threading needs to be set-up properly. (at least without setting `-std`)

> I'm not sure about not downgrading to C++11 

Neither am I ;)
Perhaps it would make sense use this as a fallback if `-std=c++11` is not supported for some weird reasons.