moveit/moveit #311: Flaky Travis Test

davetcoleman, 2016-10-22:
Travis sometimes fails with this error message during testing:

```
Warnings << moveit_ros_planning_interface:make /root/ws_moveit/logs/moveit_ros_planning_interface/build.make.002.log
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/rostest/cmake/../../../bin/rostest", line 36, in <module>
    rostestmain()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/__init__.py", line 274, in rostestmain
    _main()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_main.py", line 167, in rostestmain
    results_file = xmlResultsFile(pkg, outname, is_rostest, env=env)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosunit/core.py", line 102, in xml_results_file
    raise IOError("cannot create test results directory [%s]. Please check permissions."%(test_dir))
IOError: cannot create test results directory [/root/ws_moveit/build/moveit_ros_planning_interface/test_results/moveit_ros_planning_interface]. Please check permissions.
Cannot find results, writing failure results to '/root/ws_moveit/build/moveit_ros_planning_interface/test_results/moveit_ros_planning_interface/MISSING-rostest-test_python_move_group.xml'
cd /root/ws_moveit/build/moveit_ros_planning_interface; catkin build --get-env moveit_ros_planning_interface | catkin env -si  /usr/bin/make run_tests --jobserver-fds=6,7 -j; cd -
```

@rhaschke I think this is one of the tests you've been developing

davetcoleman, 2016-10-31:
ping this is still causing random PRs to fail, such as https://github.com/ros-planning/moveit/pull/329 with the [travis log](https://travis-ci.org/ros-planning/moveit/jobs/172036984)

davetcoleman, 2016-11-29:
I spent ~30 minutes working on fixing this tonight because I'm tired of our Travis tests randomly failing. I uncovered another failing test https://github.com/ros-planning/moveit/issues/374 and was unable to understand how to fix this. 

The test passes fine locally, only failing sometimes on Travis. It appears to be a permission issue writing the log output file. Unless its author @rhaschke is going to look into it further, I am going to propose a PR commenting out the test.

rhaschke, 2017-01-10:
I have no idea how to fix this issue. It's not clear why rostest fails to create the results directory. Go ahead and disabled this test.

davetcoleman, 2017-02-03:
@v4hn after the next Kinetic release do you think it will be easier to debug this problem due to your [fix](https://github.com/ros/ros/issues/134)? should we keep this topic open and the test in place?

v4hn, 2017-02-03:
On Fri, Feb 03, 2017 at 12:52:25AM -0800, Dave Coleman wrote:
> after the next Kinetic release do you think it will be easier to debug this problem due to your [fix](https://github.com/ros/ros/issues/134)?

I hope so. I would keep the test in place for the moment, so we can look at the error message once the updated rosunit code arrived there.
If it still does not make any sense then, we can always just disable it..