moveit/moveit #323: Random poses for collision objects if TF lookup fails

NikolasE, 2016-10-28:
"[/move_group ERROR 1477640169.928505838]: Unable to transform from frame 'foo' to frame 'bar'. Returning identity."

in transforms.cpp:
```

const Eigen::Affine3d &moveit::core::Transforms::getTransform(const std::string &from_frame) const
{
  if (!from_frame.empty())
  {
    FixedTransformsMap::const_iterator it =
        (from_frame[0] == '/' ? transforms_.find(from_frame) : transforms_.find('/' + from_frame));
    if (it != transforms_.end())
      return it->second;
  }

  logError("Unable to transform from frame '%s' to frame '%s'. Returning identity.", from_frame.c_str(),
           target_frame_.c_str());

  // return identity
  static const Eigen::Affine3d identity = Eigen::Affine3d::Identity();
  return identity;
}
```

In case, no transformation is found, the identity is returned (instead of an error). This (imho horrible) design leads to collision objects sometimes placed (in my case) to the base of the robot instead of where they belong. I have not yet figured out why my transformation (published on tf_static) does not work, but I think this function has to be fixed nonetheless.

v4hn, 2016-10-28:
Well, technically it *does* return an error.
This is the original design of MoveIt with its push-on-no-matter-the-error behavior.
Changing this function to raise an exception is a rather fundamental change to the system and it should be accounted for in all the places where this is used.
You probably wouldn't want your `move_group` to die because you entered an invalid/non-existing frame in your motion plan request.

That being said, I would welcome a pull-request to change it as you propose (even if it does not yet accounts for the change in all MoveIt code using this function.

NikolasE, 2016-10-28:
"Well, technically it does return an error."

No it does not. It returns a completely valid result. Functions like that should rather be implemented as service or action so that I can see if the collision object was published. 

"You probably wouldn't want your move_group to die" 
I'd strongly prefer it to explode instead of planning a path without all obstacles.

davetcoleman, 2016-10-28:
> I'd strongly prefer it to explode instead of planning a path without all obstacles.

LOL

[image]

I ran into this same problem yesterday in old code, are you on the kinetic branch? I suspect something might have changed. To fix this, I first ran ``updateFrameTransforms()`` [ref](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h#L283)

NikolasE, 2016-10-29:
I'm currently still on Indigo. Is there at least a way to get the current obstacles from the move_group-node?

v4hn, 2016-10-31:
> > Well, technically it does return an error.
> 
> No it does not. It returns a completely valid result

Ok, that was phrased the wrong way... It logs an error but does not return it.
I totally agree that is not good enough.

> Functions like that should rather be implemented as service or action so that I can see if the collision object was published.

Getting a simple transform should be a *ROS* service/action? That's not how TF is intended to be used.
On the other hand, raising an exception in this function would make a lot of sense to me.

> "You probably wouldn't want your move_group to die"
> I'd strongly prefer it to explode instead of planning a path without all obstacles.

Yes. I never said it should plan without the obstacles!
Nevertheless you can imagine people would complain if their main move_group node aborts due to an exception just because they made a typo in a `frame_id` field (and probably did not realize yet).

> Is there at least a way to get the current obstacles from the move_group-node?

You might want to have a look at the (rather new) [ApplyPlanningScene service](http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/planning_scene_ros_api_tutorial.html#interlude-synchronous-vs-asynchronous-updates).
This allows you to add obstacles synchronously and also provides a "bool success" in the service response.
If you provide a patch to raise an exception in `getTransform` and try-catch it in the code that applies the object to the `PlanningScene`, you could return `false` there.
(Notice that you have to load the `move_group/ApplyPlanningSceneService` capability in your `move_group.launch` for that to work)

You can request collision objects in the planning scene of the `move_group` node by using the [`PlanningSceneInterface`](https://github.com/ros-planning/moveit/blob/indigo-devel/moveit_ros/planning_interface/planning_scene_interface/include/moveit/planning_scene_interface/planning_scene_interface.h).
The `getObjects` method in there does not seem to be released in indigo yet, but is part of the `indigo-devel` branch.
For your purposes `getObjectPoses` might be enough though.

NikolasE, 2016-10-31:
Thank you for your hints! 

> Getting a simple transform should be a ROS service/action? That's not how TF is intended to be used.
I meant that setting an obstacle should be a service (exactly the ApplyPlanningScene service I think). And at least TF2 is offering an action to get transformations so that a node does not have to subscribe and buffer the full /tf and /tf_static.

v4hn, 2016-10-31:
yw :sunglasses: 

> And at least TF2 is offering an action to get transformations so that a node does not have to subscribe and buffer the full /tf and /tf_static.

Does it? It has been quite a while since I last looked at TF2. Who provides that action?
I didn't think they would centralize TF2 in one node...

NikolasE, 2016-10-31:
Just run a "rosrun tf2_ros buffer_server". If a node just wants a transformation, it can use an action without having to subscribe to /tf and /tf_static (which can be a lot of traffic) so even an arduino-node can use /tf

v4hn, 2016-11-01:
Interesting. thanks :)

Anyway, it would be great to have a pull-request for the behavior we discussed in this issue.

closed 2020-12-08 by NikolasE