moveit/moveit #331: Moveit Commander throwing terminate error with boost

pamico, 2016-10-31:
### Description

I am trying to create a program that uses moveit commander and I get the following error when I try to run it.

terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)

My code is only the python shebang and imports as follows:

#!/usr/bin/env python
import rospy
import geometry_msgs.msg
import moveit_commander

### Your environment
* ROS Distro: Indigo
* OS Version: e.g. Ubuntu 14.04
* Source or Binary build?   Latest binary

MPJansen, 2016-11-02:
Hi @pamico,

I can confirm that the same happens for me, running the same environment while trying the moveit_group tutorial.

I found these two possibly related questions:
http://answers.ros.org/question/241108/cannot-import-moveit_commander/
http://answers.ros.org/question/238081/python-process-died-exit-code-11/

Has not helped me yet though.

MPJansen, 2016-11-02:
Hi @pamico 
Just found this discussion: 
https://github.com/ros-planning/moveit_commander/issues/15

Make sure your file is executable (chmod +x file.py)
Run the code using rosrun [package] file.py
The core dump still happens but only when your script exits, so you should be able to do what you want.

davetcoleman, 2016-11-02:
@MPJansen do you mind updating the python tutorial to mention these steps?

https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/pr2_tutorials/planning/scripts/doc/move_group_python_interface_tutorial.rst

MPJansen, 2016-11-03:
@davetcoleman , I've never done that before, but I'll give it a shot today or tomorrow.

Should I mention the core dump (and the github discussion) explicitly?

MPJansen, 2016-11-03:
@davetcoleman Pull request is here ros-planning/moveit_tutorials#47

davetcoleman, 2016-11-03:
Great, thanks - I left some feedback

pamico, 2016-11-03:
@MPJansen @davetcoleman 
So my script is exicutable and I am running it with rosrun and it will work as you say except for when I actually want to use a moveit commander command in which my script just kills saying Segmentation Faule(Core Dumped) and nothing else

davetcoleman, 2016-11-03:
@pamico there is a different issue in ROS Kinetic: https://github.com/ros-planning/moveit/issues/86

perhaps that is it? otherwise you'll need to share a backtrace compiled in debug mode

pamico, 2016-11-03:
@davetcoleman Using gdb my output is as follows:

```
Starting program: /usr/bin/python /home/peter/ros_ws/src/arm_control/scripts/baxter_moveit_command.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdf076700 (LWP 11968)]
[New Thread 0x7fffd6875700 (LWP 11969)]
[New Thread 0x7fffde875700 (LWP 11970)]
[New Thread 0x7fffde074700 (LWP 11971)]
[New Thread 0x7fffdd873700 (LWP 11972)]
[New Thread 0x7fffdd072700 (LWP 11973)]
[New Thread 0x7fffdc871700 (LWP 11974)]
[New Thread 0x7fffd7fff700 (LWP 11975)]
robot_description
[ERROR] [1478197139.233569751]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1478197139.233735135]: Loading robot model 'baxter'...
[ INFO] [1478197139.233807490]: No root joint specified. Assuming fixed joint
[New Thread 0x7fffd71ce700 (LWP 11976)]
[New Thread 0x7fffd6074700 (LWP 11977)]
[Thread 0x7fffd6074700 (LWP 11977) exited]
[New Thread 0x7fffd6074700 (LWP 11978)]
Current state:
[ WARN] [1478197141.474487748]: Joint values for monitored state are requested but the full state is not known
joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: /base
  name: ['head_pan', 'left_s0', 'left_s1', 'left_e0', 'left_e1', 'left_w0', 'left_w1', 'left_w2', 'l_gripper_l_finger_joint', 'l_gripper_r_finger_joint', 'right_s0', 'right_s1', 'right_e0', 'right_e1', 'right_w0', 'right_w1', 'right_w2', 'r_gripper_l_finger_joint', 'r_gripper_r_finger_joint']
  position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
  velocity: []
  effort: []
multi_dof_joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: /base
  joint_names: []
  transforms: []
  twist: []
  wrench: []
attached_collision_objects: []
is_diff: False
Traceback (most recent call last):
  File "/home/peter/ros_ws/src/arm_control/scripts/baxter_moveit_command.py", line 19, in <module>
    a = robot.right_arm
  File "/opt/ros/indigo/lib/python2.7/dist-packages/moveit_commander/robot.py", line 278, in __getattr__
    return object.__getattribute__(self, name)
AttributeError: 'RobotCommander' object has no attribute 'right_arm'
[Thread 0x7fffd6074700 (LWP 11978) exited]
[Thread 0x7fffdd072700 (LWP 11973) exited]
[Thread 0x7fffdc871700 (LWP 11974) exited]
[Thread 0x7fffd71ce700 (LWP 11976) exited]
[Thread 0x7fffde074700 (LWP 11971) exited]
[Thread 0x7fffdf076700 (LWP 11968) exited]
[Thread 0x7fffd6875700 (LWP 11969) exited]
[Thread 0x7fffdd873700 (LWP 11972) exited]
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument

Program received signal SIGABRT, Aborted.
0x00007ffff782dc37 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
```

Also, as it should be obvious from the above I am using the Baxter robot with their moveit packages

MPJansen, 2016-11-05:
@pamico, 
your trace shows python throwing an attribute error on line 19 of baxter_moveit_command.
Without knowing the details of the script itself, it seems as if you request an attribute of robot (the moveit robot interface?) that is not there. 
What are you trying to achieve?

davetcoleman, 2016-11-05:
yes, it looks like ``AttributeError: 'RobotCommander' object has no attribute 'right_arm'`` is the key problem

pamico, 2016-11-07:
I apologize for the confusion I have been attempting many things to remedy this issue and I'm not all that programming savvy. The above output from GDB was for an example script someone wrote to use the baxter robot with moveit commander using the import line below which may have had some issue reading the baxter info. 

`from moveit_commander import RobotCommander, roscpp_initialize, roscpp_shutdown`

The main problem I am continuing to have is that every time I import moveit_commander it get the error below

`terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument`

or sometimes more simply

`Segmentation fault (core dumped)`

while I may have issues in the future with reading the baxter info from the urdf or wherever it does not change the issue that even if I could read it correctly I get an error every time I import moveit commander or any of its commands

MPJansen, 2016-11-08:
Hi @pamico ,

Try adding
`print 'See if import finishes' 
rospy.spin()
print 'See what happens after termination'`

To your python code. Hit Ctr-c after you see the first print line.
What I am trying to get at is that the fault you are seeing happens during the end of your script (while de-allocating the moveit components I think). So if you see the two print lines appear in the console, you know that your code is actually executed.

pamico, 2016-11-08:
@MPJansen

The code that I ran is as follows:

`#!/usr/bin/env python
import sys
import rospy
import moveit_commander 

rospy.init_node('baxter_moveit_commander')

print 'See if import finishes'
rospy.spin()
print 'See what happens after termination'`

The output from terminal is as follows

`See if import finishes
^CSee what happens after termination
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)`

rhaschke, 2016-11-09:
As @MPJansen suspected, the boost error originates from a bad destruction order in ros core. This should be fixed in ROS Kinetic (https://github.com/ros/ros_comm/pull/871). Which ROS version you are using?

pamico, 2016-11-09:
I am using Indigo

pamico, 2016-11-09:
When I actually try and use the moveit commander it gives me a segmentation fault (core dump) error and wont run any of my program though

rhaschke, 2016-11-09:
https://github.com/ros-planning/moveit/issues/331#issuecomment-259284957 proved that your program runs to the end and only then crashed during cleanup. If you have any other program, that terminates beforehand, please send an appropriate example.

pamico, 2016-11-20:
It turns out something else was terminating my program. Thanks for everyones help feel free to close

closed 2016-11-20 by davetcoleman

pnambiar, 2018-10-19:
Hes this been resolved? I am getting the same error as well. 
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)