moveit/moveit #336: Need default OMPL optimization objective

davetcoleman, 2016-10-31:
Since optimization objects were added to our moveit_planners_ompl package in https://github.com/ros-planning/moveit_planners/pull/75 we now have a warning from MoveIt!

> No optimization objective specified, defaulting to PathLengthOptimizationObjective

@rbbg could you also specify a default so this warning is not output? The default should be path length. Thanks!

rbbg, 2016-11-01:
Right now, the optimization objective is set separately for each planner, do you mean you want to be able to set a different default for _all_ optimizing planners? (much like the projection evaluator?) The path length is already selected in absence of a configuration setting so perhaps removing or changing the message to a debug message would also be sufficient?

v4hn, 2016-11-01:
+1 to change it to debug.

davetcoleman, 2016-11-01:
changing it to debug sounds good to me too, thanks!

closed 2016-11-01 by davetcoleman