moveit/moveit #343: unable to sample any valid states for goal tree

shixinlishixinli, 2016-11-10:
hi,i am using indigo and my own urdf , and i use setPoseTarget().sometimes it can plan and execute ,sometimes it can not .my arm is 4 dof arm but i am sure my moveit is update.
[image]
[image]

v4hn, 2016-11-10:
This is far too unspecific to help you.

- Does plan&execute work in rviz?
- Are you sure your PoseTarget is reachable?
- What exactly does your "test_custom_node" do
- What's up with these warnings about orientation constraints?

shixinlishixinli, 2016-11-16:
plan and execute is work in rviz and pose target is reachable ,my test_custom is publish a poestion and oritation

shixinlishixinli, 2016-11-16:
i am not sure is there any reason about my arm is 4 dof

v4hn, 2017-01-13:
Sorry for the long delay. Is this still relevant? Otherwise I would close this for now...

Planning works in RViz because the rviz plugin already performs inverse kinematics(IK) for your target pose and you simply cannot press the planning button if you don't see the goal joint configuration (in yellow/orange) in the visualization.

There are a number of problems with IK for arms that do not have exactly 6 or 7 DoF
and KDL (I assume this is your selected IK plugin as you didn't mention you use a different one) often fails to find solutions here.
If you can do that, it might be worth a try to generate an IK-fast plugin for your specific robot.

closed 2017-03-20 by v4hn