moveit/moveit #345: MoveIt crashes on some processor architectures due to problem with libfcl.so.0.5

Nic-G, 2016-11-10:
### Description

moveit_ros_move_group node crashes on start-up.  This doesn't seem to be a problem with MoveIt! code, it's the libfcl library, but I'm posting the issue so people are aware what is happening.

Error is 'Illegal Instruction'

Processor (we think) has to be one which accepts the 'avx' flag

More details at http://answers.ros.org/question/247652/crash-running-moveit-move_group-node/

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build > happens with both, not a MoveIt code issue per se.

davetcoleman, 2016-11-10:
Thanks for reporting. I've responded on ROS answers, as have others

gavanderhoorn, 2016-11-11:
Related issue: flexible-collision-library/fcl#187.

gavanderhoorn, 2016-12-28:
It looks like the new packages built by @j-rivero haven't been imported / synced yet: a default install of Kinetic (on a non avx cpu) still `SIGILL`s.

gavanderhoorn, 2017-02-16:
Update: this should be resolved after the next sync for public and in `shadow-fixed` already.

davetcoleman, 2017-03-11:
kinetic was synced March 5th, closing this issue

https://discourse.ros.org/t/new-packages-for-kinetic-2017-03-05/1430

closed 2017-03-11 by davetcoleman