moveit/moveit #348: Move moveit_visual_tools into ros_planning

davetcoleman, 2016-11-11:
I would like to propose, at the [suggestion](https://github.com/ros-planning/moveit_tutorials/pull/45#discussion_r85935284) of @v4hn, that my package [moveit_visual_tools](https://github.com/davetcoleman/moveit_visual_tools/) that I've been developing for the past 5 years be moved to a repo inside of ``ros-planning``. I've been releasing this package regularly since ROS Hydro and it is already released in Kinetic (pending the next sync). Eventually, depending on adoption and feedback, we could even merge it into this main ``moveit`` repo. I already have extensive documentation and demos for both this package and the dependent [rviz_visual_tools](https://github.com/davetcoleman/rviz_visual_tools)

The main motivation was that I was revamping the [move_group_interface](https://github.com/ros-planning/moveit_tutorials/pull/45) tutorial and I really wanted features available in my package, that I use everyday in my personal development, to help new users learn MoveIt!. It is my view point that the more visuals you have, the easier it is to understand a complex system and debug geometric concepts - you can see what I did in this [youtube video](https://www.youtube.com/watch?v=4FSmZRQh37Q). 

I'd like to now share these powerful tools with everyone more publicly. Note that currently my repo has 20 stars, 18 forks, 7 contributors, 21 releases, and a branch for the past 4 ROS distros - it is a fairly popular repo.

mikeferguson, 2016-11-13:
+1 to moving under ros-planning organization -- just make sure to use the "transfer ownership" and then re-fork it back into your account so that all existing forks re-parent to the ros-planning repo and links to your repo aren't broken.

v4hn, 2016-11-14:
+1 Let's do that.

130s, 2016-11-14:
+1 I was wondering why you wouldn't do that ;)

rbbg, 2016-11-14:
As an occasional user of the package: +1

davetcoleman, 2016-11-14:
Done, and documented on [website](https://github.com/ros-planning/moveit.ros.org/pull/105) and [README](https://github.com/ros-planning/moveit/pull/356)

@130s do you agree future releases should be included with the rest of MoveIt!? I should also move https://github.com/davetcoleman/moveit_visual_tools-release to https://github.com/ros-gbp/moveit-visual_tools-release right?

I'll create a tutorial on MVT in the future...

closed 2016-11-14 by davetcoleman

130s, 2016-11-15:
Why not :)?

Actually I noticed a few things that you could guide us about:

- Package version of it is far more advanced (3.x) than `MoveIt` (0.x for all current distros).
- package.xml is version 2 while I believe almost/all current moveit packages are still behind v2. I'm just not sure if they get along well when released together (I'd be a bit surprised if this is going to cause a problem).
- Move `graph_msgs` to ros-planning too?
- Being released within moveit means that the release timing may be possibly slowed down in the future as you worried https://github.com/ros-planning/moveit_tutorials/pull/45#discussion_r85935284. In return you'll be freed (to some extent at least) from the release maintenance, which is a benefit to you and the community too from a greater perspective.

davetcoleman, 2016-11-15:
> Package version of it is far more advanced (3.x) than MoveIt (0.x for all current distros).

Yes, when I've broken API over the years I've bumped the major number, like you're suppose to in software. This won't be a problem unless we wanted to merge it into the main moveit repo.

>  I'm just not sure if they get along well when released together (I'd be a bit surprised if this is going to cause a problem).

Doubt it, and if there are problems we should upgrade MoveIt

> Move graph_msgs to ros-planning too?

I definitly don't think this is necessary, just like all the other ROS messages are not within ros-planning

>  In return you'll be freed (to some extent at least) from the release maintenance

Sure, though I put a lot of time into release tasks as well :)