moveit/moveit #357: urdfdom_compatibility.h breaking moveit_core in rosbuildfarm?

davetcoleman, 2016-11-14:
It looks like the [kinetic release](https://github.com/ros-planning/moveit/issues/18) is [still failing](http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__moveit_core__ubuntu_xenial_amd64__binary/5/console) on the build farm even after a [re-release](https://github.com/ros/rosdistro/pull/13195#event-857429421) a few days ago because of [this error](http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__moveit_core__ubuntu_xenial_amd64__binary/5/console):

```
01:54:21 CMake Error: Could not open file for write in copy operation /include/moveit/urdfdom_compatibility.h.tmp
01:54:21 CMake Error: : System Error: No such file or directory
01:54:21 CMake Error at CMakeLists.txt:42 (configure_file):
```

@rhaschke can you take a look at that error since you created that compatibility layer? maybe merging [your PR](https://github.com/ros-planning/moveit/pull/319) will fix this? @130s thoughts? thanks!

rhaschke, 2016-11-15:
I think you should simply merge #319 as suggested. The order of calling `catkin_package()` and `configure()` is wrong in the MoveIt version of this compatibility code.

davetcoleman, 2016-11-15:
Done

closed 2016-11-25 by v4hn