moveit/moveit #358: No optimization objective is specified in T-RRT

Ry0, 2016-11-15:
### Description
I don't know how to set the optimization objective for T-RRT ｉｎ `model_based_planning_context.cpp`(`moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp`).

### My environment
* ROS Distro: [Indigo|Kinetic]
* OS Version: Ubuntu 14.04(indigo-devel), 16.04(kinetic-devel)
* Source build
* The source is latest version

### Problem
I want to set the optimization objective for T-RRT.
In indigo-devel, T-RRT crashes, because the evaluation function is not well defined, 

In [this issue](https://bitbucket.org/ompl/ompl/issues/117/unchecked-null-pointer-in-trrtcpp-led-to), core dump problem was solved, but it may be not a fundamental solution.

```cpp
 93 void ompl::geometric::TRRT::setup()
 94 {
 95     Planner::setup();
 96     tools::SelfConfig selfConfig(si_, getName());
 97 
 98     if (!pdef_ || !pdef_->hasOptimizationObjective())
 99     {
100         OMPL_INFORM("%s: No optimization objective specified.  Defaulting to mechanical work minimization.", getName().c_str());
101         opt_.reset(new base::MechanicalWorkOptimizationObjective(si_));
102     }
```

### Results of trial with OMPL 1.2.1(kinetic-devel)
I installed OMPL 1.2.1 through `sudo apt-get install ros-kinetic-ompl` .

```bash
[ INFO] [1479125611.408491759]: No optimization objective specified, defaulting to PathLengthOptimizationObjective
[ INFO] [1479125611.408520612]: Planner configuration 'left_arm[TRRTkConfigDefault]' will use planner 'geometric::TRRT'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1479125611.408623095]: left_arm[TRRTkConfigDefault]: No optimization objective specified.  Defaulting to mechanical work minimization.
```

`PathLengthOptimizationObjective` is set in the first line.
> https://github.com/ros-planning/moveit/blob/indigo-devel/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L243

After `PathLengthOptimizationObjective` was set, it must be this objective on OMPL.  
But it seems that `mechanical work minimization` is set.
> https://github.com/ompl/ompl/blob/master/src/ompl/geometric/planners/rrt/src/TRRT.cpp#L101

I only understand `setOptimizationObjective ()` is the function that sets optimization objective option.
> https://github.com/ros-planning/moveit/blob/indigo-devel/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L276

Thank you in advance.

v4hn, 2016-12-17:
I'm not sure what kind of feedback would resolve this issue.

I didn't look into TRRT specifically. Can't you specify it in `ompl_planning.yaml`?
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ompl_interface_tutorial.html?highlight=optimization#ompl-optimization-objectives

Ry0, 2016-12-23:
@v4hn
Thank you for your reply. And I'm sorry that my reply was delayed.
I have not solved this problem yet.
Of course, optimization_objective was defined in ompl_planning.yaml.
The problem is that its definition is not passed to OMPL.
After writing the optimization_objective item in yaml, the following command was executed.
I tested in indigo devel.

As an experiment, I added these descriptions.
```diff
  RRTstarkConfigDefault:
    type: geometric::RRTstar
+   optimization_objective: MaximizeMinClearanceObjective
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    goal_bias: 0.05  # When close to goal select goal, with this probability? default: 0.05
    delay_collision_checking: 1  # Stop collision checking as soon as C-free parent found. default 1
  TRRTkConfigDefault:
    type: geometric::TRRT
+   optimization_objective: MaximizeMinClearanceObjective
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    goal_bias: 0.05  # When close to goal select goal, with this probability? default: 0.05
    max_states_failed: 10  # when to start increasing temp. default: 10
    temp_change_factor: 2.0  # how much to increase or decrease temp. default: 2.0
    min_temperature: 10e-10  # lower limit of temp change. default: 10e-10
    init_temperature: 10e-6  # initial temperature. default: 10e-6
    frountier_threshold: 0.0  # dist new state to nearest neighbor to disqualify as frontier. default: 0.0 set in setup()
    frountierNodeRatio: 0.1  # 1/10, or 1 nonfrontier for every 10 frontier. default: 0.1
    k_constant: 0.0  # value used to normalize expresssion. default: 0.0 set in setup()
```

I used `moveit_resources` package.
https://github.com/ros-planning/moveit_resources

```bash
roslaunch moveit_resources demo.launch
```

RRTstar worked fine, but T-RRT failed.
It may be a problem with ompl rather than moveit.

Thank you.

v4hn, 2016-12-23:
Thanks for the explanation! This sounds much more easy to pinpoint!