moveit/moveit #368: Trajectory display non-default robot_description crashes rviz

willcbaker, 2016-11-18:
moveit_ros_visualization : Trajectory display crashes rviz when switching to a non-default  robot description, i.e., `robot_description` changed to `pr2/robot_description`

ROS Distro: [Indigo]
Ubuntu 14.04
Source (indigo-devel) [19000104454f2ca6eaf4efc42993ecb7e1bec6cf]
    and debian [0.7.2-0trusty-20161018-183008-0700]

### Steps to Reproduce:
load a robot and robot_description parameter under a namespace
```
export ROS_NAMESPACE=test
rosparam load <URDF_FILE>
```
open rviz, add a moveit_ros_visualization display, change the "Robot Description" to test/robot_description"

## Expected behavior
the robot model is loaded and if a trajectory is published on the topic, the display shows the trajectory

### Actual behaviour
rviz crashes

davetcoleman, 2016-11-19:
Is this the same issue I fixed here? https://github.com/ros-planning/moveit_ros/pull/635

willcbaker, 2016-11-21:
yes, sorry; hadn't found that issue/fix

davetcoleman, 2016-11-21:
I've been using ``baxter/robot_description`` in the Trajectory display for many months, though I think only on Jade and now Kinetic. 

Does the same robot_description you are loading work fine without the namespace?
Does ``rosparam list`` output the parameter you are expecting? I wasn't aware of the ``export ROS_NAMESPACE=test`` capability, normally I set namespaces within roslaunch files

A pull request for a fix is most welcomed!

willcbaker, 2016-11-21:
Yes it works without a namespace; and it even works if I manually hardcode the default robot_description to 'test/robot_descritpion' by doing the following in the TrajectoryDisplay::TrajectoryDisplay() : Display() function:
```
  robot_description_property_ = new rviz::StringProperty(
      "Robot Description", "test/robot_description", "The name of the ROS parameter where the URDF for the robot is loaded",
      this, SLOT(changedRobotDescription()), this);
```

I haven't located a fix as of yet; I can't quite tell where the segfault is occurring; I put in a significant amount of debug output and it appears all methods in this file complete successfully