moveit/moveit #373: Moveit Commander no option to set set_max_acceleration_scaling_factor

mmoerdijk, 2016-11-29:
### Description
There is no way to call the [setMaxAccelerationScalingFactor](http://docs.ros.org/jade/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html#ac82f6937c4a43690bc9fef08c42cee7e) from Python, this would be nice to have

### Your environment
* ROS Distro: [~~Indigo~~|Jade|~~Kinetic~~]
* OS Version: Ubuntu 14.04
* Source or Binary build? **Source**
* If binary, which release version? **Jade**

davetcoleman, 2016-11-29:
> I'm happy to do a pull request for this. Can I just add it in a similar way as the set_max_velocity_scaling?

Yep - the acceleration factor was recently added to moveit by almost exactly copying the velocity factor as you can see in this [PR](https://github.com/ros-planning/moveit_ros/pull/634/files)

v4hn, 2017-01-13:
Fixed in #377

closed 2017-01-13 by v4hn