moveit/moveit #386: [Indigo] 0.7.3 relase build failure on ROS buildfarm

130s, 2016-12-22:
Related to #100

Looks like the build farm is failing to build `moveit` 0.7.3-0 particularly `moveit_kinematics` e.g. [this job](http://build.ros.org/view/Ibin_uT64/job/Isrc_uT__moveit_kinematics__ubuntu_trusty__source/5/console):

```
12:12:14 Traceback (most recent call last):
12:12:14   File "/tmp/ros_buildfarm/scripts/release/get_sources.py", line 48, in <module>
12:12:14     sys.exit(main())
12:12:14   File "/tmp/ros_buildfarm/scripts/release/get_sources.py", line 44, in main
12:12:14     args.os_name, args.os_code_name, args.source_dir)
12:12:14   File "/tmp/ros_buildfarm/ros_buildfarm/sourcedeb_job.py", line 59, in get_sources
12:12:14     (source_version, pkg_version))
12:12:14 RuntimeError: The cloned package version from the GBP (0.9.3-0trusty) does not match the expected package version from the distribution file (0.7.3-0)
12:12:14 Build step 'Execute shell' marked build as failure
```
Indeed [I had to change the version of the package](https://github.com/ros-planning/moveit/commit/746dbf9dcc663751e8b2eb3d68b09c080101dcd8#diff-2606211cb8b44b1187b384e29859dd8dL4) (i.e. downgrade from 0.9.3 to 0.7.2. This package was maintained with different version number and moved into this repo recently) in order for it to align with others in the same repo (if I remember correctly this was needed by `catkin_prepare_release` or `bloom`, or both?). 

I'm not sure what to do to fix this case. @tfoote @dirk-thomas I'd appreciate if you could help.

dirk-thomas, 2016-12-22:
You changelog contains a higher version (0.9.3) which ends up in the generate Debian changelog. You either have to strip every bit of information for those higher version numbers or avoid downgrading in the first place and release 0.9.4.

130s, 2016-12-22:
Thanks a lot Dirk. I would have never figured that out by myself.

Would it cause any similar issue if some of the package with smaller version (e.g. 0.5.7) than the one targeted at the release (0.7.3) are missing the `Forthcoming` section in their changelogs?
(Ideally I wanted to add changelogs for those packages once the release this time goes well https://github.com/ros-planning/moveit/issues/100#issuecomment-268446976. But it's already messed up so adding them manually can be an option this time).

dirk-thomas, 2016-12-22:
Looking at #387 it seems you have figured it out .

130s, 2016-12-25:
Yes, this particular issue was resolved at #387.

commit 347fd6cf (2016-12-22): Merge pull request #387 from 130s/i/fix/386

[indigo] Changelog for 0.7.4 (preparing to fix #386)
closed 2016-12-25 by 130s