moveit/moveit #392: Make moveit_ros_planning depend on moveit_kinematics (indigo-only)

ipa-mdl, 2016-12-29:
The old indigo-release contained all kinematics plugins in moveit_ros_planning, but the new release moved them to moveit_kinematics.
In the result many kinematics packages fail to build, need to be fixed and released again.
(https://discourse.ros.org/t/preparing-for-indigo-sync-2016-12-21/1004 etc)

IMHO these problems could be resolved in a single place if moveit_ros_planning would expose the kinematics plugins again (reexport from moveit_kinematics):
* depends on moveit_kinematics
* pass library and includes

130s, 2016-12-30:
Though I haven't explored disadvantages, +1 for the idea.
As a quick fix to the issues referred to at [discourse](https://discourse.ros.org/t/preparing-for-indigo-sync-2016-12-21/1004/2?u=130s), I had to open a couple of PRs https://github.com/ros-industrial/universal_robot/pull/275 and https://github.com/auboliuxin/aubo_robot/pull/6, and we don't know how many other non-released downstream packages are there.

ipa-mdl, 2016-12-31:
> and we don't know how many other non-released downstream packages are there.

Luckily build errors like this are easy to identify, but `exec_depend` for using the kinematic plugin has to be fixed as well.
([36](http://repositories.ros.org/status_page/ros_indigo_default.html?q=moveit_config) `*_moveit_config` packages have been released for indigo)

v4hn, 2016-12-31:
This is a big problem indeed.
We moved the modules to the new package to consolidate the various kinematics plugins.
To ease development across the ROS branches we also introduced this change in indigo.
I didn't expect there would be header dependencies across the plugins and failed to check this before merging @davetcoleman's request. I'm sorry about that.

The solution you proposed @ipa-mdl does not work because it introduces a circular dependency.
The plugins rely on rdf_loader, which is part of moveit_ros_planning.
You're one of the listed authors there. :)

I'm unsure about how to fix this in a reasonable way. Does anyone see a nice way out?
Undoing the change is probably not an option anymore.

ipa-mdl, 2016-12-31:
> it introduces a circular dependency.

I see :-/

> Does anyone see a nice way out?

Not really nice, but some ideas:
1. keep a (symlinked) copy of `rdf_loader` class in `moveit_kinematics`
2. copy/symlink all of `moveit_kinematics` into `moveit_ros_planning`
3. make rdf_loader a separate package

davetcoleman, 2016-12-31:
When I created ``moveit_kinematics`` I [added](https://github.com/ros-planning/moveit/commit/0f0ca09941e57031f581ef6147a0bf249d30ca18#diff-53fe61d3a98493cab13a2bc7b35ca822R29) the ``moveit_kinematics`` dependency to the ``moveit_ros_move_group`` package, which I thought would transfer the dependency smoothly. It does for the:

> 36 *_moveit_config packages have been released for indigo

because ``_moveit_config`` packages depend on ``moveit_ros_move_group``. So I don't think this is as big a problem as it first appeared.

The only problematic packages seem to be custom kinematics packages that used parts of MoveIt!'s kinematics.

ipa-mdl, 2017-01-02:
> I added the moveit_kinematics dependency to the moveit_ros_move_group package, which I thought would transfer the dependency smoothly.
[...]
So I don't think this is as big a problem as it first appeared.

👍 

> The only problematic packages seem to be custom kinematics packages that used parts of MoveIt!'s kinematics.

These build failures are easy to detect, most/all released packages have already been fixed.

The `*_moveit_config` should work and the circular dependency is hard to address.
I will close this issue.

closed 2017-01-02 by ipa-mdl