moveit/moveit #393: New Jade Release

davetcoleman, 2016-12-31:
Now that Indigo is [released](https://github.com/ros-planning/moveit/issues/100#issuecomment-269811232) from the new consolidated MoveIt! repo I think its time to re-release Jade.

As shown on the MoveIt! README on the [main page](https://github.com/ros-planning/moveit) we now have a confusing state where we have removed ``moveit_ikfast`` from Indigo and Kinetic, but we have not yet done so for Jade because the [PR](https://github.com/ros-planning/moveit/pull/263) came after the last Jade release.

130s, 2017-01-04:
Ok, I'll work on this. A few things:

- I like to wait for #396 to be merged in.
- It'll be absolutely more convenient for release person (me currently) if we make a new release for each ros distro at the same time, mostly for the sake of managing changelogs (time taking to keep track of changelogs per distro each of which slightly lagged) and making a blog post. 

So unless there's an objection, I'll make a new release for all IJK in late January.

davetcoleman, 2017-01-05:
> if we make a new release for each ros distro at the same time

Makes perfect sense

130s, 2017-01-05:
>  announce your efforts a week in advance. This gives everyone some time for testing and getting ready and we didn't have a proper soak-time in recent releases.. Especially for indigo, I want to do ABI regression tests before the release.

I see.
Sorry for not allowing soak time at all for the last Indigo release - public sync happened so quick even to me.

v4hn, 2017-01-05:
Actually, I meant MoveIt-internal soak time, but having both would be a good idea too. :)

130s, 2017-01-19:
Maintainers, if there are open work/PRs that are close to be merged let us know.
Otherwise I'll start making a release in a few days (so that we can save the soak-time both for MoveIt! internal and ROS buildfirm).

v4hn, 2017-01-23:
I would like to see #421 merged before the release, if a maintainer has a moment to look at it.

Apart from that, I'm looking forward to the release. :)

130s, 2017-02-06:
- K https://github.com/ros/rosdistro/pull/13809
- I https://github.com/ros/rosdistro/pull/13811
  - There's a risk that some moved metapackages might become unavailable since I couldn't think of any better solution than to remove them from rosdistro https://github.com/ros/rosdistro/pull/13810. We'll see how long the downtime will be...it can be zero, but can be days until the next public sync. OSRF should have the answer to that but I haven't asked.
- J is stuck at https://github.com/ros-infrastructure/catkin_pkg/issues/158, which can be worked around manually if I'm ok spending tons of additional time in editing changelog, or if we ignore changelog this time.

130s, 2017-02-07:
Jade is done https://github.com/ros/rosdistro/pull/13813

closed 2017-02-07 by 130s