moveit/moveit #406: Set color of collision objects via PlanningSceneInterface

davetcoleman, 2017-01-06:
@corot created a PR for this already but [it was removed](https://github.com/ros-planning/moveit_ros/pull/641/commits/4f671931c182c65535a9403a270efe6f2e41b724). I think we should add it back because:

> A) is much compact than publishing a separated PlanningScene.msg B) much more important: sometimes it doesn't work! (and so the RViz plugin shows all with the default green)

corot, 2017-01-08:
Here it is. However, back at that time I removed this change following @v4hn reasoning (I quote him):
```
However, you mixed these additions with changes to addCollisionObjects and I really don't like to see this function changed. This is for two reasons:

(1.) adding the optional parameter changes the ABI and people using the function will have to recompile their code or they get segfaults/ an undefined symbol. I know some other devs don't care about ABI changes, but I do not like to see them in indigo at least.

(2.) This function is meant to provide a simple way to add objects to the planning scene.
If you wish to add colors or allow collisions, why not publish a PlanningScene.msg yourself?
After all addCollisionObjects is just a simple wrapper...
```
Still, I think the change makes PSI interface more convenient (and I use it like this locally)

v4hn, 2017-01-09:
Hey @corot,

The first point related only to indigo (and especially to indigo without sonames - i.e. *before* the latest release). From that side I see no problem with adding it to kinetic (as you propose in your new request).

The latter point still stands from my point of view: I don't think the object color is import enough to warrant this parameter here. But @davetcoleman agrees with you, so I give in. :)

v4hn, 2017-01-23:
merged in #409

closed 2017-01-23 by v4hn