moveit/moveit #408: Colors should not reset upon receiving a new diff-update of the PlanningScene

v4hn, 2017-01-09:
Looking a bit into the code relevant for #407,
I noticed [all colors of all objects are reset whenever a new set of colors is received in a diff-PlanningScene](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/src/planning_scene.cpp#L1217).

I believe this is broken design and probably breaks adding colored objects incrementally in multiple PlanningScene updates.

Did you notice this @corot?

corot, 2017-01-09:
mmm...  nop.  Because I always provide colors for all the objects I add...  I'll take a look, but I don't know the PS code, just use the interface as a black box, so I don't think I can help a lot here

corot, 2017-01-09:
Seems to me that the guilty is this line: https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/src/planning_scene.cpp#L1220
When removed, problem get solved, apparently. At least at first glance, without entering in detail in the code, makes sense, as you just want to add new color assignments or change existing ones, letting the rest untouched.

v4hn, 2017-01-09:
@corot I agree, I created a patch for this in #410.

closed 2017-02-03 by davetcoleman