moveit/moveit #416: Moveit! trajectory velocity has weird behavior on kinetic

renarded, 2017-01-16:
### Description

Hi everyone,

I encountered a problem using Moveit! with ros kinetic. I first developped under indigo and switched to kinetic after first kinetic Moveit! release in december.

After testing some movements with my custom robot (6 axis arm), I noticed that sometimes (like 30-50%), the trajectory calculated by Moveit! does not have a correct velocity. I’m not talking about how smooth the velocity is, it’s not the problem here. Sometimes the velocity just goes up or down.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build : binary
* If binary, which release version? last release version on January, 16th 2017

### Steps to reproduce
For the test I made I asked the robot to do a very basic movement. On indigo, every time, the velocity (of each joint) increases, then reaches the max velocity (or not, depending on the planning time and max acceleration put in the config file), and finally decreases. That is, I think, the correct behavior for the robot. 

However, on kinetic, with the exact same URDF and SRDF, the trajectory sometimes has a “weird” velocity : the velocity increases, and sometimes just decreases, then increases, 2 or 3 times in one trajectory. 

Here’s my configuration for the test I made :

URDF (I did not put any visual or collision for simplicity, the problem I have is not related to that)
```
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from test.urdf.xacro                | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="test_robot" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <link name="ground_link"/>
  <link name="base_link"/>
  <link name="shoulder_link"/>
  <link name="arm_link"/>
  <link name="elbow_link"/>
  <link name="forearm_link"/>
  <link name="wrist_link"/>
  <link name="hand_link"/>
  <link name="gripper_link"/>
  <joint name="ground_joint" type="fixed">
    <parent link="ground_link"/>
    <child link="base_link"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  <joint name="joint_1" type="revolute">
    <parent link="base_link"/>
    <child link="shoulder_link"/>
    <origin rpy="0 0 0" xyz="0 0 0.085"/>
    <axis xyz="0 0 1"/>
    <limit effort="1" lower="-3.14159" upper="3.14159" velocity="1.0"/>
  </joint>
  <joint name="joint_2" type="revolute">
    <parent link="shoulder_link"/>
    <child link="arm_link"/>
    <origin rpy="1.570795 -1.570795 0" xyz="0 0 0.077"/>
    <limit effort="1" lower="-2.07" upper="1.35" velocity="1.0"/>
    <axis xyz="0 0 1"/>
  </joint>
  <joint name="joint_3" type="revolute">
    <parent link="arm_link"/>
    <child link="elbow_link"/>
    <origin rpy="0 0 0" xyz="0.18475 0.0045 0"/>
    <limit effort="1" lower="-2.9" upper="-0.65" velocity="1.0"/>
    <axis xyz="0 0 1"/>
  </joint>
  <joint name="joint_4" type="revolute">
    <parent link="elbow_link"/>
    <child link="forearm_link"/>
    <origin rpy="0 1.570795 0" xyz="0.045 0.02901 0"/>
    <limit effort="1" lower="-2.270795" upper="2.270795" velocity="1.0"/>
    <axis xyz="0 0 1"/>
[... 494 lines cut ...]
      time_from_start: 
        secs: 3
        nsecs: 487602828
multi_dof_joint_trajectory: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  joint_names: []
  points: []

```

----
I have really no idea where the problem lies. Sometimes the velocity is ok, sometimes not. The problem comes before Moveit! calls the controller with the trajectory, but I don't know where it can be.

It seems that my problem is very similar to https://github.com/ros-planning/moveit_core/issues/167 but this one has already been fixed.

I hope it could be an error I made on the configuration, but in this case, why is it working well on indigo, and not well at all on kinetic ? 

Let me know if you and if you have any clue of how to fix this.
Any help would be appreciated
Thanks

rbbg, 2017-01-16:
Hi @renarded,

I would not be surprised if this is caused by the planner rather than the time parameterization. Can you please check if Indigo and Kinetic are indeed using the same planner?

Some changes has been made in the default planner selection, and I have seen some planners do the back-and-forth thing that seems to be happening (judging from your plots).

renarded, 2017-01-16:
Hi @rbbg, 

Thanks for your very fast reply.

The default planner I used on Indigo was LBKPIECE1 if I remember well. I also tried with RRTConnect on Indigo and everything worked perfectly.

With RRTConnect on kinetic I have this problem. I also tried all the 15 or 20 other available planners, all give the same problem.

rbbg, 2017-01-16:
The velocities you posted, even for the 'bad' trajectories, look pretty typical for the default IPTP time parameterization, so I would still suspect something to go wrong during planning.

What robot are you planning with? And are there any obstacles?

renarded, 2017-01-16:
Yes I think IPTP has nothing to do with that, the problems comes from before, during the planning.

There are no obstacles in the planning scene, and  my custom robot has 6 revolute joints, with nothing inside the links. The URDF I've put here is very basic. Also when I use the same planner on Indigo, or on Kinetic, it's working fine on Indigo but not on Kinetic...

I've been trying to search a solution for days, and I really don't have any clue now.

renarded, 2017-01-16:
Oh, so if I understand well : 1. Moveit calculates the joints from whatever is asked (position, pose, etc), and 2. Moveit uses IPTP to add velocities and acceleration (with collision constraints and optional cartesian path constraints) ? 3. And nothing else ?
 
If this statement is correct it seems the problem I have is coming from IPTP (finally coming back to it !). Or it could come from a bad configuration made by myself. I would prefer the second option, but I can't find any way to solve this problem with my current URDF & SRDF, and the strange thing is : everything works perfectly with Ros Indigo. 

Is there something I could do with IPTP to make my trajectories better ? Is there something in IPTP that could cause those weird trajectories, even with no constraints and no collision ?

Also let me know if you encountered some similar problems on other robotic arms.

(ps : thanks for all the fast replies :))

rbbg, 2017-01-17:
@v4hn The reason I suspect planning is twofold. For one, I do not think there is a difference in IPTP between kinetic and indigo, but I could be mistaken. 

In addition, I've done have inspected IPTP (and other parameterization algorithms) a number of times (often on an identical trajectory) and usually when I saw IPTP doing something like above, there were some unnecessary changes of direction or slight jumps in the path (on one of the axis, but they are all dependent). Unfortunately I don't really have any data to support my suspicion, but I thought I'd just share where it came from.

renarded, 2017-01-17:
For this test I tried with the cubic interpolator I found from the Universal robot [package](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_driver/src/ur_driver/driver.py#L560) which is correct and stable.

Here I've plotted the position, velocity and acceleration of the trajectory I logged on my first comment ([INFO] [1484570660.822931]: joint_trajectory:) : 

[image]

This is the data from the Moveit! trajectory (directly received by the controller), no interpolation has been done. You can clearly see that the acceleration is going crazy, which makes the velocity not correct (or the velocity makes the acceleration not correct, I don't really know which way Moveit! is handling this).

I hope this can help ! I don't really have any idea of where, how and why this is happening, but one thing is sure : the trajectory I get from Moveit! is not correct.

@rbbg When I have time I will try to run this test on docker and post the result here. However I've already had similar issues on different systems : amd64 Ubuntu 16.04, and armhf Ubuntu mate 16.04

gavanderhoorn, 2017-01-17:
@renarded wrote:
> For this test I tried with the cubic interpolator I found from the Universal robot [package](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur_driver/src/ur_driver/driver.py#L560) which is correct and stable.

I wouldn't necessarily assume that to be true. See ros-industrial/universal_robot#207 fi.

renarded, 2017-01-18:
@gavanderhoorn ok thanks, that's a useful info ! I haven't seen this issue before, I might have spoken too hasty.

However the last plot with the "raw data" from Moveit! trajectory hasn't been interpolated, so the problem doesn't come from the controller side.

rbbg, 2017-01-19:
It seems that there are a number of problems intertwined here. 

The semantics of the jointTrajectory Message seem to be unclear so perhaps it's a good idea to clarify that first. The documentation is sparse, but I found the best hint in the [pr2_controllers tutorial](http://wiki.ros.org/pr2_controllers/Tutorials/Moving%20the%20arm%20using%20the%20Joint%20Trajectory%20Action#The_trajectory_message):

> For each point in the trajectory, put the desired joint state into positions and velocities. The controller will interpolate between these points using cubic splines. You may optionally fill in accelerations if you wish for the controller to use quintic splines. The controller will attempt to follow whatever trajectory you specify, so it is your responsibility to ensure that the desired trajectory is smooth. 

Consequently, MoveIt! is responsible for ensuring that the trajectory that is being planned will satisfy the specfied constraints _including_ the velocities that will be attained between waypoints (using cubic / quintic interpolation). If this is satisfied by both MoveIt! and the controller, the controller is not required to be aware of the limits that are set by the user. This also means that if the waypoints are set in a faulty manner, the controller will be able to violate the user-set constraints. Note that when accelerations are specified `ur_modern_driver` does not keep his end of the deal by only doing cubic instead of quintic interpolation.

**first problem:**

- The current MoveIt! time parameterization approach (IPTP) does **not** satisfy these constraints _between waypoints_

This is what issue #160 is all about. The violations are mostly on the acceleration limits, but velocity limits can be violated as well. On average though, the velocity profiles of IPTP look okay. 

On the plots @renarded posted, there seems to be something going on with the velocities as well, which leads me to believe that this _might_ be another problem, especially since IPTP hasn't changed between kinetic and indigo. 

**second problem**:

- MoveIt! seems to generate jerkier paths on Kinetic than on Indigo.

I have also heard at least one other report of someone who was quite familiar with MoveIt! on indigo, reporting that he felt the trajectories were not as smooth in kinetic. What might be causing this I currently have no clue but I feel _this_ is the bigger problem here.

I might be completely off the mark though :)

jettan, 2017-01-20:
Hi, I'm glad somebody made an issue for this.
I can also confirm the second problem @rbbg mentioned. For the same moveit_configs that used to generate smooth trajectories on indigo, moveit in kinetic seems to scholastically output smooth/jerky trajectories.

I'd like to investigate further to find out the real problem, but if anybody has pointers where to start that'd be very welcome.

jettan, 2017-01-20:
Alright I've tracked the problem down to OMPL after inspecting the diff between latest kinetic-devel and indigo-devel and seeing almost all of it were just code cleaning.
Since the current ros-kinetic-ompl is running OMPL v1.2.1 and ros-indigo-ompl is on 1.0.3094 I started suspecting the ompl version updates as it was really hard to find a changelog (I still don't know what exactly changed between the two versions except there has been more planners added in).

I then compiled ompl-1.0.3094 from source and applied some fixed for new dependencies of flann to lz4.
Recompile moveit and did some planning... all of the trajectories are 'normal' again like you'd expect them to be -- no jerky trajectories.

So I think we should move the discussion into the ompl repository instead.

rbbg, 2017-01-20:
I discussed earlier with @gavanderhoorn that (as it is reported to affect all planners) it might be related to the PathSimplifier. Judging from the [ompl release notes](http://ompl.kavrakilab.org/core/releaseNotes.html) there have been at least _some_ changes to the PathSimplifier between these versions.

jettan, 2017-01-20:
I created an issue in the ompl bitbucket  [repo](https://bitbucket.org/ompl/ompl/issues/309/performance-regression-going-from-ompl)

mamoll, 2017-01-23:
Since the issue seems to be in OMPL itself, let's continue the discussion there. I have added a comment there with ideas on what might be happening.

gavanderhoorn, 2018-04-25:
@ros-planning/moveit-developers: should this issue receive more attention? Unfortunately I don't have the bandwidth right now to do something about it, but some of the driver components under `ros-industrial` (cross-refs above this comment) are getting complaints that would seem to be at least related to the OMPL issue.

v4hn, 2018-04-25:
> should this issue receive more attention?

Yes.

> I don't have the bandwidth right now to do something about it

Same here.

rhaschke, 2018-04-25:
@gavanderhoorn This issue was [handed over to OMPL](https://bitbucket.org/ompl/ompl/issues/309/performance-regression-going-from-ompl) and you already asked there for the current status. What can we do more?

gavanderhoorn, 2018-04-25:
As developers of a tool with a large userbase and strong dependency on / ties with OMPL we could maybe try to emphasise the fact that this is a serious issue for MoveIt and its (perceived) utility in Kinetic and newer versions of ROS.

Commenting on the issue on BitBucket and / or discussing this with @mamoll in the next maintainer meeting would also be good steps, I believe.

mamoll, 2018-04-25:
In the Melodic release of OMPL I will disable the code that finds a better goal if we cannot get this resolved.

v4hn, 2018-04-26:
> In the Melodic release of OMPL I will disable the code that finds a better goal if we cannot get this resolved.

If this really solves the problem (I do not know whether it does),
could you maybe just add an option to disable this step in melodic release and a backport to kinetic?
I guess ROS-I has more interest in kinetic than in melodic right now...

jettan, 2018-04-26:
As far as I know (and I have frozen OMPL ever since 1+ year ago), I have not encountered this weird behavior after applying the steps commented here: https://bitbucket.org/ompl/ompl/issues/309/performance-regression-going-from-ompl#comment-33827336

gavanderhoorn, 2018-04-26:
@v4hn wrote:
> could you maybe just add an option to disable this step in melodic release and a backport to kinetic?

getting this into Kinetic would certainly be good, yes.

> I guess ROS-I has more interest in kinetic than in melodic right now...

This issue affects many more users of MoveIt, also those outside ROS-Industrial. I would actually assume there are more non-ros-i users of MoveIt than the other way around.

mamoll, 2018-04-26:
Good news: the bug appears to have been fixed, thanks to @BryceStevenWilley and @zkingston! In a nutshell, the order of reducing vertices along the path and getting better goal states was wrong. This resulted in paths that would have many more states than the original path. The gory details can be found here: https://bitbucket.org/ompl/ompl/commits/d4e24d22b75418f0f76676f5527dd6eb23312e25#chg-src/ompl/geometric/src/PathSimplifier.cpp. I will backport this fix for Kinetic.

renarded, 2018-04-26:
Great news !

>  This resulted in paths that would have many more states than the original path

This seems to be exactly what was the problem when I first posted this issue.

Thanks for the backport on Kinetic, when this is ready I'll make some tests with the same config as my initial post, and I will post the result here.

hangkaiyu, 2018-04-26:
I am having the same issue here with a Barrett WAM running MoveIt! with ROS Kinetic on Ubuntu 16.04. 2 (wrist_pitch and wrist_yaw) of the 7 joints on WAM are randomly jerking(90% of time) when executing a planned trajectory, especially during the second half of the trajectories.

I have implemented the ROS hardware interface by myself and am using the `velocity_controllers/JointTrajectoryController` to interface with MoveIt! 

I am so glad to see that this bug has been found and a fix for it will come out. Please leave a message here when the backport fix will be available, I can't wait to fix it in my system to make WAM move like a normal robot! Thanks.

asmorgan24, 2018-05-01:
I am glad to see that this issue is being looked into. I tried to find a work-around over a year ago and had to scrap that approach all together. I see there are some people currently interested in the work-around, do you happen to have an ETA when a fix is available for Kinetic? @mamoll

BryceStevenWilley, 2018-05-02:
@asmorgan24  As of the 1.2.3 tag on the kinetic-devel branch of OMPL (https://bitbucket.org/ompl/ompl/commits/tag/1.2.3), these changes are available from OMPL source, and the correlated fix in MoveIt was merged with #869. If you can, you should be able to build OMPL and MoveIt from source (instructions for OMPL [here](http://moveit.ros.org/install/source/dependencies/#open-motion-planning-library-ompl), use the Bitbucket repo instead of the github for the moment, and for MoveIt [here](http://moveit.ros.org/install/source/#kinetic)), otherwise you'll have to wait for the next Kinetic Release of OMPL.

hangkaiyu, 2018-05-02:
@BryceStevenWilley Do you know approximately when the next Kinetic Release will be available?

hangkaiyu, 2018-05-02:
@mamoll Great, looking forward to it!

mamoll, 2018-05-04:
Done: https://github.com/ros/rosdistro/pull/17747
Let me know if this fixes things.

hangkaiyu, 2018-05-11:
I have tried the version 1.2.3-0 and the problem still exists. Here is what I did:

1. Remove old ompl installation by: 
`sudo apt remove ros-kinetic-ompl`
2. Get the source code of ompl from github repo ros-gbp/ompl-replease:
`git clone https://github.com/ros-gbp/ompl-release.git`
3. Checked out the branch **release/kinetic/ompl**
4. Followed the README to build and install it
5. Followed [http://moveit.ros.org/install/source/](url) to install MoveIt! from source
6. Run my test

When I use the motion planner in rviz to plan and execute, the trajectory is still very jerky for joint_5, and looks like it happens only to joint_5. Below is snapshot of the joint velocity command, it's been going up and down around 0 and never ends until I kill the program. 

[image]

FYI, I am using a hardware_interface with `velocity_controllers/JointTrajectoryController` implemented by myself.

BryceStevenWilley, 2018-05-11:
> 6. Run my test

@hangkaiyu can you explain this step further? You mention above that you are using the Barrett WAM, but what trajectory are you executing and what sort of environment? 

>  Below is snapshot of the joint velocity command, it's been going up and down around 0 and never ends until I kill the program.

This fix changed the trajectories that MoveIt was creating to execute, not necessarily the controllers themselves. Your comment here suggests that the issues lies somewhere in your velocity controller, as if the robot is not currently executing a trajectory, then the MoveIt/OMPL planning is not executed.  Could you give a graph of what the trajectory looks like for each joint position? Something similar to [this image](https://bitbucket-assetroot.s3.amazonaws.com/repository/9pAaox/1599343012-check_and_repair.png?Signature=BBCMuOIK28H86ZUblF5Uk5zqEWY%3D&Expires=1526082048&AWSAccessKeyId=AKIAIQWXW6WLXMB5QZAQ&versionId=N.gqJ.kGGeXrm6jDhf_0p.CgJ9hYTPUW).

hangkaiyu, 2018-05-12:
@BryceStevenWilley 

> what trajectory are you executing and what sort of environment?

It was pretty simple -- I just set random goals for WAM to move around using MoveIt! in an empty environment.

> Your comment here suggests that the issues lies somewhere in your velocity controller

As you suggested, I just did some tests on the velocity controller of our WAM. Yes you are right, the velocity controller of WAM has some problems for joint 4 and 5 -- as long as I send random velocity commands (between 0.4 and 0.6 rad/s) to the controller, those two joints will start jerking and the velocities will bounce between positive and negative values. 

In my test, I tried to send random velocity control commands at between 5Hz and 500Hz, and it was always the same problem, and it won't happen if I send constant velocity commands. so it is obvious that the WAM controller cannot handle sudden velocity changes for those two joints.

Thanks for your help! I'll contact Barrett to see how this can be solved.

v4hn, 2018-05-14:
It would be nice if somebody who experienced the problem could verify that the current MoveIt `kinetic-devel` together with OMPL `1.2.3-0` resolves this problem.

jettan, 2018-05-15:
Hi, I will test this today.

jettan, 2018-05-15:
Unfortunately we are at ROS Lunar and there is no release yet as of now. There seems to be compile problems with the latest moveit on kinetic-devel in combination with the latest OMPL as found from github (EuclideanProjection has been deprecated). So I cannot really confirm the fix at this moment.

zkingston, 2018-05-15:
@jettan The latest OMPL on github is the candidate for OMPL `1.4.0`. You should try OMPL `1.2.3-0`, which you can get from here: https://github.com/ros-gbp/ompl-release/tree/release/kinetic/ompl.

jettan, 2018-05-15:
Thanks, I’ll try again tomorrow.

jettan, 2018-05-16:
I tested today and have verified that using the current MoveIt kinetic-devel together with OMPL 1.2.3-0 resolves this problem.

v4hn, 2018-05-18:
Thanks a lot for the patches and the support from the OMPL crew! :+1: 
Closing.

closed 2018-05-18 by v4hn

hangkaiyu, 2018-05-18:
Thanks so much for the OMPL team's great effort! Really appreciate your brilliant developments and quick responses!

renarded, 2018-05-23:
Thanks a lot !

I've also tested (I know, a little bit late) with the configuration shown in the first message of this thread, and the explained problem is definitively gone.